facebookresearch/habitat-sim #430: Errors when building navmesh on the fly for replica models.

bigbike, 2020-01-14:
## 🐛 Bug

## Command
`# command that you run`

## To Reproduce

Steps to reproduce the behavior:

1.
2.
3.

## Expected behavior

## Additional context

## System Info

Please copy and paste the output from the
[environment collection script](https://github.com/facebookresearch/habitat-sim/tree/master/habitat_sim/utils/collect_env.py)
(or fill out the checklist below manually).

You can run the script with:
```
# For security purposes, please check the contents of collect_env.py before running it.
python habitat_sim/utils/collect_env.py
```

 - OS (e.g., Linux): Fedora 
 - How you installed PyTorch (`conda`, `pip`, source):
 - Build command you used (if compiling from source):
 - Python version:
 - GPU models and configuration:
 - CUDA version:
 - PIP/conda dependencies packages versions
 - Any other relevant information:

aclegg3, 2020-01-15:
Related to this review comment: https://github.com/facebookresearch/habitat-sim/pull/400/files/e9e13ecbd1cee404e8ef3e036157aba904bf59d0#r364340322

Currently, the navmesh is recomputed by joining all geometry from `BaseMesh::collisionMeshData_`. This data-structure is not populated in PTex load currently, so we can't recompute navmesh (no geometry to find) and physics won't work in these scenes either. 

Once PR #390 is merged we can unify the data-structure for asset geometry references and ensure it is available for all asset types.

bigbike, 2020-01-22:
A short term solution is introduced by #434. Will put the long term solution in our TODO list. Close it for now.

closed 2020-01-22 by bigbike