facebookresearch/habitat-sim #431: Build --with--cuda doesn't actually work neither shows an error

erick84mm, 2020-01-15:
## 🐛 Bug

building habitat-sim --with--cuda option doesn't work and it doesn't show an error if it doesn't.

## Command

```
## Starting from a fresh location I run

git clone --recursive https://github.com/facebookresearch/habitat-sim.git
git clone --recursive https://github.com/erick84mm/habitat-api.git

cd habitat-sim
conda create -n habitat python=3.6.9 cmake=3.14.0 -y
conda activate habitat
pip install -r requirements.txt
python setup.py build_ext --parallel 30 install --headless --with-cuda

cd ../habitat-api
pip install -r requirements.txt
python setup.py develop --all
python -c "import habitat_sim; print(habitat_sim.cuda_enabled)" 

```

## To Reproduce

Steps to reproduce the behavior:

1. Create Environment
2. Download
3. Build 
4. Build Succeeds 
5. Test

## Expected behavior

## Additional context

```
(habitat) erick@musa:~/Research/vln/libs/habitat/habitat-sim$ python setup.py build_ext --parallel 30 install --headless --with-cuda
running build_ext
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- The CUDA compiler identification is NVIDIA 10.1.243
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working CUDA compiler: /srv/data/apps/cuda/10.1.243/bin/nvcc
-- Check for working CUDA compiler: /srv/data/apps/cuda/10.1.243/bin/nvcc -- works
-- Detecting CUDA compiler ABI info
-- Detecting CUDA compiler ABI info - done
-- LIB_SUFFIX variable is not defined. It will be autodetected now.
-- You can set it manually with -DLIB_SUFFIX=<value> (64 for example)
-- LIB_SUFFIX autodetected as '', libraries will be installed into /usr/local/lib
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
[... 1458 lines cut ...]
| 27%   31C    P8     2W / 250W |     10MiB / 11019MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   6  GeForce RTX 208...  Off  | 00000000:3E:00.0 Off |                  N/A |
| 27%   35C    P8    16W / 250W |     10MiB / 11019MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   7  GeForce RTX 208...  Off  | 00000000:3F:00.0 Off |                  N/A |
| 27%   35C    P8    22W / 250W |     10MiB / 11019MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   8  GeForce RTX 208...  Off  | 00000000:40:00.0 Off |                  N/A |
| 27%   31C    P8    15W / 250W |     10MiB / 11019MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   9  GeForce RTX 208...  Off  | 00000000:41:00.0 Off |                  N/A |
| 27%   32C    P8     1W / 250W |     10MiB / 11019MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
Pip packages versions:

Conda packages versions:
attrs                     19.3.0                   pypi_0    pypi  19.1.0
numba                     0.47.0                   pypi_0    pypi
numpy                     1.18.1                   pypi_0    pypi
numpy-quaternion          2019.12.11.22.25.52          pypi_0    pypi
pillow                    7.0.0                    pypi_0    pypi
scipy                     1.4.1                    pypi_0    pypi  1.3.0
tqdm                      4.41.1                   pypi_0    pypi
```

erikwijmans, 2020-01-15:
What does `python -c "import habitat_sim; print(habitat_sim.__file__)"` show?

erick84mm, 2020-01-15:
@erikwijmans It shows /home/erick/.local/lib/python3.6/site-packages/habitat_sim-0.1.3-py3.6-linux-x86_64.egg/habitat_sim/__init__.py

erikwijmans, 2020-01-15:
That's the issue.  You are picking up an old build, not the new one in the conda env.  I am not sure why that one is being picked up.  Maybe you set your `PYTHONPATH` env variable?  If all else fails and you don't need it, you can just remove it.

erick84mm, 2020-01-15:
Thank you so much!  That solved my issue, I forgot to check if there were other packages!

closed 2020-01-15 by erick84mm