facebookresearch/habitat-sim #455: Error in building habitat-sim

shivanshpatel35, 2020-01-28:
Hi, 
I am having trouble building habitat-sim on a headless server. I get the following error on running `python setup.py build_ext --parallel 1 install --headless`
```
running build_ext
-- Shared libraries disabled
-- Looking for ZLIB...
-- Checking for module 'zzip-zlib-config'
--   No package 'zzip-zlib-config' found
-- Found ZLIB: optimized;/cvmfs/soft.computecanada.ca/nix/var/nix/profiles/16.09/lib/libz.so;debug;/cvmfs/soft.computecanada.ca/nix/var/nix/profiles/16.09/lib/libz.so
-- Checking for module 'minizip'
--   No package 'minizip' found
-- Enabled formats: AMF 3DS AC ASE ASSBIN ASSXML B3D BVH COLLADA DXF CSM HMP IRRMESH IRR LWO LWS MD2 MD3 MD5 MDC MDL NFF NDO OFF OBJ OGRE OPENGEX PLY MS3D COB BLEND IFC XGL FBX Q3D Q3BSP RAW SIB SMD STL TERRAGEN 3D X X3D GLTF 3MF MMD
-- Disabled formats:
-- Found PythonInterp: /localscratch/shivp.35251576.0/env/bin/python (found suitable version "3.6.3", minimum required is "3.6") 
-- Bindings being generated for python at /localscratch/shivp.35251576.0/env/bin/python
-- Found PythonInterp: /localscratch/shivp.35251576.0/env/bin/python (found version "3.6.3") 
-- pybind11 v2.3.0
-- Found Corrade: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/corrade/src  found components:  Containers rc Utility Main TestSuite 
-- Found Corrade: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/corrade/src  found components:  Containers rc Utility PluginManager 
-- Found Corrade: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/corrade/src  found components:  Containers rc Utility 
-- Found Corrade: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/corrade/src  found components:  Containers rc Utility PluginManager 
-- Found Magnum: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/magnum/src  found components:  Trade AnyImageImporter 
-- Found Magnum: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/magnum/src  found components:  Trade 
-- Found Magnum: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/magnum/src  found components:  Trade AnyImageImporter 
-- Found Corrade: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/corrade/src  found components:  Containers rc Utility 
-- Found Magnum: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/magnum/src   
-- Found Magnum: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/magnum/src  found components:  GL SceneGraph 
-- Found Corrade: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/corrade/src  found components:  Containers rc Utility PluginManager 
-- Could NOT find GLFW (missing: GLFW_LIBRARY GLFW_INCLUDE_DIR) 
-- Could NOT find Magnum (missing: GlfwApplication Sdl2Application WindowlessGlxApplication WindowlessWglApplication) 
Building with ptex support
-- Found Corrade: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/corrade/src  found components:  Containers rc Utility 
-- Found Corrade: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/corrade/src  found components:  Containers rc Utility PluginManager 
-- Found Magnum: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/magnum/src  found components:  Trade GL AnyImageImporter AnySceneImporter MeshTools SceneGraph Shaders AnyImageConverter 
-- Found Magnum: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/magnum/src  found components:  Trade AnyImageImporter 
-- Found Corrade: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/corrade/src  found components:  Containers rc Utility 
-- Found Magnum: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/magnum/src   
-- Found Magnum: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/magnum/src  found components:  GL WindowlessEglApplication 
-- Found Corrade: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/corrade/src  found components:  Containers rc Utility PluginManager 
-- Found Magnum: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/magnum/src  found components:  Trade GL AnyImageImporter AnySceneImporter MeshTools SceneGraph Shaders 
-- Found Magnum: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/magnum/src  found components:  Trade AnyImageImporter 
-- Found MagnumPlugins: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/magnum-plugins/src  found components:  AssimpImporter 
-- Found MagnumPlugins: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/magnum-plugins/src  found components:  StbImageImporter StbImageConverter TinyGltfImporter 
Building Python bindings
-- Found Corrade: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/corrade/src  found components:  Containers rc Utility 
-- Found Magnum: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/magnum/src   
-- Configuring done
-- Generating done
-- Build files have been written to: /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/build
[1/226] Building CXX object deps/magnum/src/Magnum/CMakeFiles/MagnumObjects.dir/PixelStorage.cpp.o
FAILED: deps/magnum/src/Magnum/CMakeFiles/MagnumObjects.dir/PixelStorage.cpp.o 
ccache /cvmfs/soft.computecanada.ca/nix/var/nix/profiles/gcc-5.4.0/bin/c++   -I/home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps -I/home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/magnum/src -Ideps/magnum/src -I/home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/corrade/src -Ideps/corrade/src -isystem /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/cmake/../deps/eigen-git-mirror -isystem /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/cmake/../deps/Sophus -isystem /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/cmake/../deps/rapidjson/include -DVERSION_INFO=\"0.1.3\" -fvisibility=hidden -O2 -g -DNDEBUG -fPIC   -Wall -Wextra -Wold-style-cast -Winit-self -Werror=return-type -Wmissing-declarations -pedantic -fvisibility=hidden -fvisibility-inlines-hidden -Wzero-as-null-pointer-constant -Wdouble-promotion -fno-strict-aliasing -std=gnu++14 -MD -MT deps/magnum/src/Magnum/CMakeFiles/MagnumObjects.dir/PixelStorage.cpp.o -MF deps/magnum/src/Magnum/CMakeFiles/MagnumObjects.dir/PixelStorage.cpp.o.d -o deps/magnum/src/Magnum/CMakeFiles/MagnumObjects.dir/PixelStorage.cpp.o -c /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/magnum/src/Magnum/PixelStorage.cpp
In file included from /cvmfs/soft.computecanada.ca/nix/var/nix/profiles/gcc-7.3.0/include/c++/7.3.0/bits/move.h:54:0,
                 from /cvmfs/soft.computecanada.ca/nix/var/nix/profiles/gcc-7.3.0/include/c++/7.3.0/bits/stl_pair.h:59,
                 from /cvmfs/soft.computecanada.ca/nix/var/nix/profiles/gcc-7.3.0/include/c++/7.3.0/utility:70,
                 from /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/magnum/src/Magnum/PixelStorage.h:33,
                 from /home/shivp/projects/def-msavva/shivp/aihabitat/habitat-sim/src/deps/magnum/src/Magnum/PixelStorage.cpp:26:
/cvmfs/soft.computecanada.ca/nix/var/nix/profiles/gcc-7.3.0/include/c++/7.3.0/type_traits:458:49: error: expected ‘>’ before ‘_GLIBCXX_NOEXCEPT_PARM’
   template<typename _Res, typename... _ArgTypes _GLIBCXX_NOEXCEPT_PARM>
                                                 ^
/cvmfs/soft.computecanada.ca/nix/var/nix/profiles/gcc-7.3.0/include/c++/7.3.0/type_traits:459:65: error: template argument 1 is invalid
     struct is_function<_Res(_ArgTypes...) _GLIBCXX_NOEXCEPT_QUAL>
                                                                 ^
/cvmfs/soft.computecanada.ca/nix/var/nix/profiles/gcc-7.3.0/include/c++/7.3.0/type_traits:462:49: error: expected ‘>’ before ‘_GLIBCXX_NOEXCEPT_PARM’
   template<typename _Res, typename... _ArgTypes _GLIBCXX_NOEXCEPT_PARM>
                                                 ^
/cvmfs/soft.computecanada.ca/nix/var/nix/profiles/gcc-7.3.0/include/c++/7.3.0/type_traits:463:67: error: template argument 1 is invalid
     struct is_function<_Res(_ArgTypes...) & _GLIBCXX_NOEXCEPT_QUAL>
                                                                   ^
/cvmfs/soft.computecanada.ca/nix/var/nix/profiles/gcc-7.3.0/include/c++/7.3.0/type_traits:466:49: error: expected ‘>’ before ‘_GLIBCXX_NOEXCEPT_PARM’
[... cut ...]
  File "setup.py", line 202, in run
    self.build_extension(ext)
  File "setup.py", line 276, in build_extension
    shlex.split("cmake --build {}".format(self.build_temp)) + build_args
  File "/cvmfs/soft.computecanada.ca/easybuild/software/2017/Core/python/3.6.3/lib/python3.6/subprocess.py", line 291, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--', '-j1']' returned non-zero exit status 1.```
Thank you very much.

dhruvbatra, 2020-01-28:
> -- Checking for module 'zzip-zlib-config'
> --   No package 'zzip-zlib-config' found
> -- Checking for module 'minizip'
> --   No package 'minizip' found
> -- Could NOT find GLFW (missing: GLFW_LIBRARY GLFW_INCLUDE_DIR) 

From our readme: 
> See this configuration for a full list of dependencies that our CI installs on a clean Ubuntu VM. If you run into build errors later, this is a good place to check if all dependencies are installed.

msavva, 2020-01-28:
@shivanshpatel35 the above looks like a compiler error.  Check the gcc version -- the build has been tested with gcc 7+ .

shivanshpatel35, 2020-02-09:
It seems like some dependencies were not installed. The problem is solved now. Thanks for the help.

closed 2020-02-09 by shivanshpatel35

M-Pixel, 2021-02-21:
Which dependencies?