facebookresearch/habitat-sim #459: Inserted physics models are invisible

smorad, 2020-01-28:
## ❓ Questions and Help

Running the physics test example I see cheezit boxes floating above the table. I've converted a `ply` mesh to `glb` and placed it along with the other test objects in the data directory. When I place my `glb` files (one convex hull for collisions and one mesh for rendering) and the physics JSON (references the `glb`s), I get no errors but I do not see any objects.

I've used mesh viewers to ensure that the mesh is fine after conversion to glb. Is there anything special I need to do to get my mesh to render?

smorad, 2020-01-29:
Issue was the meshes were too big. After shrinking them, they appear. Unfortunately, it seems like the models fail to load if I apply any materials in blender.

How were the test models created? I figure if I make my mesh the same way as the test models, it should work.

aclegg3, 2020-01-29:
Hey, @smorad.
I'm glad the issue was only a matter of scale. For convenience you can specify scale in the object config file instead of manually scaling your models. 
E.g.
```
...
"scale": [3.0,2.0,1.0],
...
``` 

Regarding materials, it is true that not all Blender materials will render correctly in Habitat-sim at the moment. For flat color shading you can use the `Principled BSDF` Material as with `data/test_assets/objects/sphere.glb`. We will have a Phong illumination model coming soon (#450).  You can also take a look at the material setup for ycb objects provided in the [example physics objects dataset](http://dl.fbaipublicfiles.com/habitat/objects_v0.1.zip) or bake the textures into per-vertex colors as with .obj format.

closed 2020-02-05 by smorad