facebookresearch/habitat-sim #480: Error when run 'setup.py'

liuxinzhu0353150307, 2020-02-09:
I run the command "sudo python setup.py install", but I got the following error: 
  
File "setup.py", line 388
    subprocess.check_call(shlex.split(f"pip install {pymagnum_build_dir}"))
                                                                        ^
SyntaxError: invalid syntax

I want to know how to solve the roblem.

erikwijmans, 2020-02-09:
Habitat-sim requires python>=3.6.  Please follow the build instructions here: https://github.com/facebookresearch/habitat-sim#from-source

closed 2020-02-09 by erikwijmans

mylad13, 2023-02-07:
I am following the build instruction

> Habitat-sim requires python>=3.6. Please follow the build instructions here: https://github.com/facebookresearch/habitat-sim#from-source

I am following these instruction, but I still get the same error when using "python setup.py install". 
[image]

dhruvbatra, 2023-02-08:
Note for future viewers: we no longer support python 3.6