facebookresearch/habitat-sim #487: coordinate correspondence between blender (or other 3d program) and habitat

ronjamoller, 2020-02-11:
Hi, I am working on inserting human meshes into the simulation and i would like to know the exact position of the joints of my human mesh. I can import various meshes from blender into habitat but i cannot get to insert the mesh in such a way that i can reliably compute the joint positions relative to the insertion point. With trial and error i have figured out that when a .glb mesh is imported from blender into habitat the y and z axis are mirrored but there are still translation offsets. 
[image]

Furthermore, sometimes a mesh will look completely normal in blender and when imported into habitat it is suddenly deformed and translated/rotated. 
[image]

[image]

I have also tried importing the sokloster glb into blender to double check that the orientation of my human mesh is correct and it is, however when i import it into habitat-sim the issues persist. I anyone has experience with inserting objects into habitat in a way that allows coordinate correspondences i would be very thankful for any advice.

aclegg3, 2020-02-11:
Translational offsets can occur when the COM of the imported object is not specified in the template. In this case, the object's center (often of its bounding box) is used and the asset is translated such that this point is the origin of the local coordinate system. This translational offset is printed to terminal on load, so you should be able to adjust your points by that offset. Alliteratively, if you do not intend to simulate the object, you can define the COM manually as [0,0,0] to prevent any translation from being applied.

ronjamoller, 2020-02-26:
thank you so much for the tip - i assumed the COM would translate automatically. For anyone else having the issue: just add "COM": [0,0,0], in the phys_properties json , this will make everything match up. 
Thanks again for the help.
Please let me know if you want me to close this issue.

closed 2020-03-19 by aclegg3