facebookresearch/habitat-sim #491: ModuleNotFoundError: No module named 'habitat_sim._ext'

ducha-aiki, 2020-02-13:
I have the same problem as described here 
https://github.com/facebookresearch/habitat-sim/issues/48

1. python setup.py install 
Neither build.sh nor python setup.py install --headless  help. Build is successfull.

2. Download habitat-test-scenes from provided link
3. python examples/example.py --scene /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb

(habitat) old-ufo@oldufo-ubuntupc:~/storage/NEWHABITAT/habitat-sim$ python examples/example.py --scene data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
Traceback (most recent call last):
  File "examples/example.py", line 10, in <module>
    import demo_runner as dr
  File "/home/old-ufo/storage/NEWHABITAT/habitat-sim/examples/demo_runner.py", line 15, in <module>
    from settings import default_sim_settings, make_cfg
  File "/home/old-ufo/storage/NEWHABITAT/habitat-sim/examples/settings.py", line 5, in <module>
    import habitat_sim
  File "/home/old-ufo/storage/NEWHABITAT/habitat-sim/habitat_sim/__init__.py", line 14, in <module>
    from habitat_sim.nav import *
  File "/home/old-ufo/storage/NEWHABITAT/habitat-sim/habitat_sim/nav/__init__.py", line 1, in <module>
    from habitat_sim._ext.habitat_sim_bindings import (
ModuleNotFoundError: No module named 'habitat_sim._ext'

## Expected behavior

Run the example

## Additional context

## System Info

Please copy and paste the output from the
[environment collection script](https://github.com/facebookresearch/habitat-sim/tree/master/habitat_sim/utils/collect_env.py)
(or fill out the checklist below manually).

ENVIRONMENT INFO:
Platform: Linux-4.4.0-142-generic-x86_64-with-debian-stretch-sid
Machine: x86_64
Processor: x86_64
Libc version: glibc 2.9
Mac version: 
Python version: 3.6.7
Architecture: 64bit 
Win version:    
System OS: Linux
Release: 4.4.0-142-generic
Version: #168-Ubuntu SMP Wed Jan 16 21:00:45 UTC 2019
Operational System: linux
GCC version: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
CMAKE version: cmake version 3.14.0
NVIDIA-SMI: Thu Feb 13 14:06:23 2020       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.33.01    Driver Version: 440.33.01    CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX TIT...  Off  | 00000000:01:00.0 Off |                  N/A |
| 22%   36C    P8    14W / 250W |    704MiB / 12209MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
Pip packages versions:

Conda packages versions:
attrs                     19.3.0                   pypi_0    pypi  19.1.0
numba                     0.48.0                   pypi_0    pypi
numpy                     1.18.1                   pypi_0    pypi
numpy-quaternion          2019.12.11.22.25.52          pypi_0    pypi
pillow                    7.0.0                    pypi_0    pypi
scipy                     1.4.1                    pypi_0    pypi  1.3.0
tqdm                      4.42.1                   pypi_0    pypi

erikwijmans, 2020-02-13:
Looks like you have `/path/to/habitat-sim` in your PYTHONPATH.  Please do `unset PYTHONPATH` (or similar) to remove it.

chale96, 2020-02-15:
I am having the same issue and neither if I have the `/path/to/habitat-sim` in my `PYTHONPATH` nor if I `unset PYTHONPATH` works. 

Still the same error.

- **running**:  `python examples/example.py --scene data/scene_datasets/habitat-test-scenes/skokloster-castle.glb`

**with** `export PYTHONPATH=$PYTHONPATH:/path/to/habitat-sim`:

2020-02-15 15:10:42,777 Initializing dataset PointNav-v1
Traceback (most recent call last):
  File "examples/example.py", line 25, in <module>
    example()
  File "examples/example.py", line 11, in example
    env = habitat.Env(config=habitat.get_config("configs/tasks/pointnav.yaml"))
  File "/Users/alessandrocandeloro/Documents/University/in3007_ind_proj/implementation/habitat-api/habitat/core/env.py", line 74, in __init__
    id_dataset=config.DATASET.TYPE, config=config.DATASET
  File "/Users/alessandrocandeloro/Documents/University/in3007_ind_proj/implementation/habitat-api/habitat/datasets/registration.py", line 20, in make_dataset
    return _dataset(**kwargs)
  File "/Users/alessandrocandeloro/Documents/University/in3007_ind_proj/implementation/habitat-api/habitat/datasets/pointnav/__init__.py", line 38, in __init__
    raise pointnav_import_error
  File "/Users/alessandrocandeloro/Documents/University/in3007_ind_proj/implementation/habitat-api/habitat/datasets/pointnav/__init__.py", line 20, in _try_register_pointnavdatasetv1
    from habitat.datasets.pointnav.pointnav_dataset import (
  File "/Users/alessandrocandeloro/Documents/University/in3007_ind_proj/implementation/habitat-api/habitat/datasets/pointnav/pointnav_dataset.py", line 15, in <module>
    from habitat.tasks.nav.nav import (
  File "/Users/alessandrocandeloro/Documents/University/in3007_ind_proj/implementation/habitat-api/habitat/tasks/nav/nav.py", line 30, in <module>
    from habitat.sims.habitat_simulator.actions import HabitatSimActions
  File "/Users/alessandrocandeloro/Documents/University/in3007_ind_proj/implementation/habitat-api/habitat/sims/habitat_simulator/actions.py", line 12, in <module>
    import habitat_sim
  File "/Users/alessandrocandeloro/Documents/University/in3007_ind_proj/implementation/habitat-sim/habitat_sim/__init__.py", line 14, in <module>
    from habitat_sim.nav import *
  File "/Users/alessandrocandeloro/Documents/University/in3007_ind_proj/implementation/habitat-sim/habitat_sim/nav/__init__.py", line 1, in <module>
    from habitat_sim._ext.habitat_sim_bindings import (
ModuleNotFoundError: No module named 'habitat_sim._ext'

- **with** `unset PYTHONPATH`

2020-02-15 15:06:22,369 Initializing dataset PointNav-v1
Traceback (most recent call last):
  File "examples/example.py", line 25, in <module>
    example()
  File "examples/example.py", line 11, in example
    env = habitat.Env(config=habitat.get_config("configs/tasks/pointnav.yaml"))
  File "/usr/local/lib/python3.7/site-packages/habitat/core/env.py", line 74, in __init__
    id_dataset=config.DATASET.TYPE, config=config.DATASET
  File "/usr/local/lib/python3.7/site-packages/habitat/datasets/registration.py", line 20, in make_dataset
    return _dataset(**kwargs)
  File "/usr/local/lib/python3.7/site-packages/habitat/datasets/pointnav/__init__.py", line 38, in __init__
    raise pointnav_import_error
  File "/usr/local/lib/python3.7/site-packages/habitat/datasets/pointnav/__init__.py", line 20, in _try_register_pointnavdatasetv1
    from habitat.datasets.pointnav.pointnav_dataset import (
  File "/usr/local/lib/python3.7/site-packages/habitat/datasets/pointnav/pointnav_dataset.py", line 15, in <module>
    from habitat.tasks.nav.nav import (
  File "/usr/local/lib/python3.7/site-packages/habitat/tasks/nav/nav.py", line 30, in <module>
    from habitat.sims.habitat_simulator.actions import HabitatSimActions
  File "/usr/local/lib/python3.7/site-packages/habitat/sims/habitat_simulator/actions.py", line 12, in <module>
    import habitat_sim
ModuleNotFoundError: No module named 'habitat_sim'

erikwijmans, 2020-02-15:
It seems like you didn't install habitat-sim.  Please follow the instructions here: https://github.com/facebookresearch/habitat-sim#from-source

chale96, 2020-02-15:
> It seems like you didn't install habitat-sim. Please follow the instructions here: https://github.com/facebookresearch/habitat-sim#from-source

I did install it following the instructions.
I tried it installing it both  inside the habitat-api folder and outside in the same folder where habitat-api is located but still having the same issue

erikwijmans, 2020-02-16:
Just to clarify, you are currently running habitat-api's example script, not habitat-sim.  Can you try to run `python examples/example.py` while in the `/path/to/habitat-sim` folder?  That'll help reduce the number of variables.  Also, please ensure you followed the habitat-sim build instructions while in that folder (i.e. not in the folder for habitat-api).  If you installed into a conda/virtual/etc environment, please make sure that environment is activated.

chale96, 2020-02-17:
> Just to clarify, you are currently running habitat-api's example script, not habitat-sim. Can you try to run `python examples/example.py` while in the `/path/to/habitat-sim` folder? That'll help [...]

Thank you,
Running `python examples/example.py` while in the `/path/to/habitat-sim` folder works perfectly.

erikwijmans, 2020-02-17:
Is this with or without `unset PYTHONPATH`?

chale96, 2020-02-17:
> Is this with or without `unset PYTHONPATH`?

My `echo $PYTHONPATH` returns empty since I reopen the terminal, so I guess with.

erikwijmans, 2020-02-17:
When in `/path/to/habitat-sim`, what does `echo $(cd examples; python -c "import habitat_sim; print(habitat_sim.__file__)")` show?

chale96, 2020-02-17:
> When in `/path/to/habitat-sim`, what does `echo $(cd examples; python -c "import habitat_sim; print(habitat_sim.__file__)")` show?

this is what it returns:

/Users/alessandrocandeloro/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.1.4-py3.6-macosx-10.9-x86_64.egg/habitat_sim/__init__.py

erikwijmans, 2020-02-17:
Looks like you installed into a conda env named habitat.  Please ensure that this environment is properly activated when you try to run things with habitat-api and please make sure you installed habitat-api in the same conda environment.  Based on your current outputs, habitat-api was installed globally and you are likely not in this conda environment.

shahhaard47, 2020-04-27:
Hello! I'm in the same boat. I am installing habitat_sim (stable branch) on GCP. I got the `python setup.py build_ext --parallel 2 --headless` installed after a lot of work. It finishes install with the final message about magnum-0.0.0 successfully installed (which is consistent with other places I've installed it). 

However, [like above](https://github.com/facebookresearch/habitat-sim/issues/491#issuecomment-586594191) when I do `pip list | grep habitat`, I don't see it. And when I try to run any examples, I get ```
ModuleNotFoundError: No module named 'habitat_sim'
```
and when I add installed location to PYTHONPATH, I get the 
```
ModuleNotFoundError: No module named 'habitat_sim._ext'
```

Thank you for your help in other issues thus far, it has helped me greatly. Any help here would be greatly appreciated!

erikwijmans, 2020-04-27:
Looks like you forgot install in `python setup.py build_ext --parallel 2 --headless`, please try `python setup.py build_ext --parallel 2 install --headless`

shahhaard47, 2020-04-27:
Thank you so much haha.

closed 2020-04-28 by erikwijmans

BryanBo-Cao, 2020-05-30:
Hi All,

Thanks for all the info! Just to share my experience regarding the

```ModuleNotFoundError: No module named 'habitat_sim'```.

I am installing Habitat in a newly used laptop Alienware 17 R5 and installed CUDA 10.2 manually.

I followed the instructions in https://github.com/facebookresearch/habitat-sim#from-source. Initially, I was not able to build by ```python setup.py install --with-cuda```, but I was able to do so by ```python setup.py install```. However, when I run the test script by ```python examples/example.py --scene /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb```, I got the error ```ModuleNotFoundError: No module named 'habitat_sim'```.

Then I removed, git cloned the repo and followed the same procedure until the setup steps. More specifically I installed by
```
python setup.py build_ext --parallel 2 install --headless
```
as mentioned and now I can run the ```example.py``` without any error.

20481535, 2022-06-08:
from habitat_sim._ext.habitat_sim_bindings import Simulator as SimulatorBackend
ModuleNotFoundError: No module named 'habitat_sim._ext'

I still have the above issues after 
>python setup.py build_ext --parallel 2 install --headless

Mrakas, 2025-03-03:
download package from https://anaconda.org/search?q=habitat-sim , and solved.