facebookresearch/habitat-sim #498: CMake Error when installation in Colab 2

BryanBo-Cao, 2020-02-15:
## 🐛 Bug
Following the #476 

Run
```
!python setup.py install --headless
```
in Colab with the updated setup.py:
```
running install
running bdist_egg
running egg_info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
reading manifest template 'MANIFEST.in'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
CMake Error: The source directory "/content/gdrive/My Drive/habitat-testing/habitat-sim/Drive/habitat-testing/habitat-sim/src" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
Traceback (most recent call last):
  File "setup.py", line 380, in <module>
    include_package_data=True,
  File "/usr/local/lib/python3.6/dist-packages/setuptools/__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/bdist_egg.py", line 172, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/bdist_egg.py", line 158, in call_command
    self.run_command(cmdname)
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/usr/lib/python3.6/distutils/command/install_lib.py", line 109, in build
    self.run_command('build_ext')
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 202, in run
    self.build_extension(ext)
  File "setup.py", line 269, in build_extension
    env=env,
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '-H/content/gdrive/My', 'Drive/habitat-testing/habitat-sim/src', '-Bbuild', '-DBUILD_PYTHON_BINDINGS=ON', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/content/gdrive/My Drive/habitat-testing/habitat-sim/build/lib.linux-x86_64-3.6/habitat_sim/_ext', '-DPYTHON_EXECUTABLE=/usr/bin/python3', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-DBUILD_GUI_VIEWERS=OFF', '-DTARGET_HEADLESS=ON', '-DBUILD_TEST=OFF', '-DBUILD_WITH_BULLET=OFF', '-DBUILD_DATATOOL=OFF', '-DBUILD_WITH_CUDA=OFF']' returned non-zero exit status 1.
```

erikwijmans, 2020-02-16:
Please make sure you are either on https://github.com/facebookresearch/habitat-sim/commit/05db83e02fc1decc7d49ba6b63759d10ce9e1c87 or on a latter commit.  Note that you will need to be on master for this commit, this commit is later than the current `stable` tag.

PS.  In the future, feel free to re-open the existing issue instead of creating a second issue.  Make it easier for everyone :)

BryanBo-Cao, 2020-02-16:
I actually have attempted to reopen #476 and append my commit in it but I was unable to reopen it. That's why I opened a second one.

Skylion007, 2020-02-20:
I was able to build it in a directory that has spaces in the filenames so I am having trouble reproducing it. I also am not getting this error. I think the initial cmake error that the folder does not exist is more telling. Does it work if you open the directory from a folder that is symlinked to have no spaces? Or just mount your GoogleDrive in a location without spaces.

BryanBo-Cao, 2020-02-21:
@Skylion007 Thanks for looking into this issue! It would be great to have greater details about how to build it in Colab.

(1) Previously, I mount google drive by
```
from google.colab import drive
drive.mount('/content/gdrive')
```
and create a folder named "habitat-testing" in My Drive and
```
cd 'gdrive/My Drive/habitat-testing'
```
that's why there's a space in ```My Drive```.

---
(2) Then I tried to install it without mounting google drive. To be more specific at the beginning of my Colab code:
```
!git clone https://github.com/facebookresearch/habitat-sim.git
```
```
cd habitat-sim/
```
```
pip install -r requirements.txt
```
```
!python setup.py install --headless
```

But when it proceeded to around 56%, I got an "textfield"(?) and the process just got stuck.
[image]

BryanBo-Cao, 2020-02-21:
BTW, I have successfully installed and used habitat-sim for a couple of months in a fixed remote server on Ubuntu 18.04.3 LST.

For this issue, I tried to use habitat-sim in Colab with my student account. So the actual backend depends on what Google assigns to me, although I don't think it should matter.

erikwijmans, 2020-02-21:
For collab, you will likely need to do `python setup.py build_ext --parallel 1 install --headless` to reduce the amount of parallelism.

closed 2020-08-09 by dhruvbatra