facebookresearch/habitat-sim #499: Headless build issue

tfy14esa, 2020-02-18:
## ❓ Questions and Help
Hi,
I have successfully been able to build and run habitat without the options "--headless" and "--with-cuda" on my local machine. I have access to a cluster where I also want to run habitat, which is not connected to any display. This means that I should build habitat-sim according to: "python setup.py install --headless --with-cuda". 

I am trying to build the system on my local machine, which has a screen connected to it. I run the nvidia-driver 390.116 on this local machine. The cuda version which I compile habitat-sim with is cuda 9.2 which needs at least nvidia driver 396.37 to work. The cluster which I aim to run the system on does however have a compatible nvidia-driver version to cuda 9.2. I wanted to provide you with the information, but I don't think this should cause any problems in itself for the build.

The build error is at the bottom of this post. My own comments:
The system finds the EGL library and OpenGL according to the lines
```
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so   
-- Found EGL: /usr/lib/x86_64-linux-gnu/libEGL.so 
```
But all major erros seem to be related to the EGL_NO_CONTEXT variable.

The build fail is caused by:
```
[569/635] Linking CXX shared module deps/magnum-bindings/src/python/_magnum.cpython-36m-x86_64-linux-gnu.so
ninja: build stopped: subcommand failed.
```
From what I understand, "_magnum.cpython-36m-x86_64-linux-gnu.so" needs a working version of libEGL.so.

The following libraries are not found:
```
-- Could NOT find GLFW (missing: GLFW_LIBRARY GLFW_INCLUDE_DIR) 
-- Could NOT find SDL2 (missing: SDL2_LIBRARY SDL2_INCLUDE_DIR) 
```
When I run the commands dpkg -l | grep glfw and dpkg -l | grep sdl2 I do, however, get the following outputs:
```
ii  libglfw3:amd64                                                   3.2.1-1                                     amd64        portable library for OpenGL, window and input (x11 libraries)
```
```
ii  libsdl2-2.0-0:amd64                                              2.0.5+dfsg1-2                               amd64        Simple DirectMedia Layer
```

Furthermore, this leads to magnum not finding the following components
```
-- Could NOT find Magnum (missing: GlfwApplication Sdl2Application) 
-- Could NOT find Magnum (missing: WindowlessGlxApplication)
```

I would be very grateful for any help to resolve this.
Cheers,
Erik

Below is the complete build output and error messages:
```
(/scratch_net/nudel/esandstroem/venvs/habitat_w_headless_w_cuda_env) esandstroem@nudel:~/scratch/programs/habitat-sim_w_head_w_cuda$ python setup.py install --headless --with-cuda
running install
running bdist_egg
running egg_info
creating habitat_sim.egg-info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib.linux-x86_64-3.6
creating build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/sim.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/logging.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/sensor.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/physics.py -> build/lib.linux-x86_64-3.6/habitat_sim
[... 653 lines cut ...]
    cmd_obj.run()
  File "/scratch_net/nudel/esandstroem/venvs/habitat_w_headless_w_cuda_env/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 172, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/scratch_net/nudel/esandstroem/venvs/habitat_w_headless_w_cuda_env/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 158, in call_command
    self.run_command(cmdname)
  File "/scratch_net/nudel/esandstroem/venvs/habitat_w_headless_w_cuda_env/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/scratch_net/nudel/esandstroem/venvs/habitat_w_headless_w_cuda_env/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/scratch_net/nudel/esandstroem/venvs/habitat_w_headless_w_cuda_env/lib/python3.6/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/scratch_net/nudel/esandstroem/venvs/habitat_w_headless_w_cuda_env/lib/python3.6/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/scratch_net/nudel/esandstroem/venvs/habitat_w_headless_w_cuda_env/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/scratch_net/nudel/esandstroem/venvs/habitat_w_headless_w_cuda_env/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 202, in run
    self.build_extension(ext)
  File "setup.py", line 278, in build_extension
    shlex.split('cmake --build "{}"'.format(self.build_temp)) + build_args
  File "/scratch_net/nudel/esandstroem/venvs/habitat_w_headless_w_cuda_env/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--']' returned non-zero exit status 1.
```

erikwijmans, 2020-02-18:
This happens with old EGL headers.  Maybe the discussions in these issues will help: https://github.com/facebookresearch/habitat-sim/issues/25 
https://github.com/facebookresearch/habitat-sim/issues/145
https://github.com/facebookresearch/habitat-sim/issues/146

tfy14esa, 2020-02-25:
Thanks for your response. I could not try to update the drivers since I am working on a managed computer and the IT team cannot update the drivers either due to compatibility problems. At the end, we made it work by using a containerized version.

closed 2020-08-09 by dhruvbatra