facebookresearch/habitat-sim #504: error: identifier "__builtin_ia32_monitorx" and "__builtin_ia32_mwaitx" are undefined when --headless --with-cuda

mhaboali, 2020-02-25:
## ❓ Questions and Help
Actually, I don't know if I should post this issue here or on habitat-api, so please if here is not suitable for it, let me know and I'll post it there. 

I ran Habitat-Api headless on a remote machine with 1x RTX 2080 Ti and it worked properly
(with the stable branch for both habitat-api and habitat-sim and with --headless installation of habitat-sim. I ran pointnav PPO training and validation examples and it took around 4~5 hours for training)

My goal is to run a Habitat-baseline(It may be PPO or DDPPO) on multiple GPUs to see how much scaling up will be done. I thought I need to build habitat-sim this time --headless --with-cuda flags, was I correct?

Also, I'm wondering about the difference between `SIMULATOR_GPU_ID `and `TORCH_GPU_ID` usage and which one should I use for training on multiple GPUs?

System Specs: I followed the installation procedure step by step and I tried with both branches(stable and master for this time).
Remote machine with 2x RTX 2080 Ti, Ubuntu 16.04, CUDA Version: 10.2, Cuda compilation tools, release 7.5, V7.5.17, and gcc version: 5.4.0

```
[ 84%] Built target assets
[ 88%] Built target gfx
[ 90%] Built target geo
/usr/lib/gcc/x86_64-linux-gnu/5/include/mwaitxintrin.h(36): error: identifier "__builtin_ia32_monitorx" is undefined

/usr/lib/gcc/x86_64-linux-gnu/5/include/mwaitxintrin.h(42): error: identifier "__builtin_ia32_mwaitx" is undefined

2 errors detected in the compilation of "/tmp/tmpxft_00000ee7_00000000-7_RedwoodNoiseModel.cpp1.ii".
esp/sensor/CMakeFiles/noise_model_kernels.dir/build.make:62: recipe for target 'esp/sensor/CMakeFiles/noise_model_kernels.dir/RedwoodNoiseModel.cu.o' failed
make[2]: *** [esp/sensor/CMakeFiles/noise_model_kernels.dir/RedwoodNoiseModel.cu.o] Error 2
CMakeFiles/Makefile2:4710: recipe for target 'esp/sensor/CMakeFiles/noise_model_kernels.dir/all' failed
make[1]: *** [esp/sensor/CMakeFiles/noise_model_kernels.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
Traceback (most recent call last):
  File "setup.py", line 382, in <module>
    include_package_data=True,
  File "/opt/conda/envs/habitat2/lib/python3.6/site-packages/setuptools/__init__.py", line 144, in setup
    return distutils.core.setup(**attrs)
  File "/opt/conda/envs/habitat2/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/opt/conda/envs/habitat2/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/opt/conda/envs/habitat2/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/opt/conda/envs/habitat2/lib/python3.6/site-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/opt/conda/envs/habitat2/lib/python3.6/site-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/opt/conda/envs/habitat2/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/opt/conda/envs/habitat2/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/opt/conda/envs/habitat2/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 172, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/opt/conda/envs/habitat2/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 158, in call_command
    self.run_command(cmdname)
  File "/opt/conda/envs/habitat2/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/opt/conda/envs/habitat2/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/opt/conda/envs/habitat2/lib/python3.6/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/opt/conda/envs/habitat2/lib/python3.6/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/opt/conda/envs/habitat2/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
 File "/opt/conda/envs/habitat2/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 202, in run
    self.build_extension(ext)
  File "setup.py", line 278, in build_extension
    shlex.split('cmake --build "{}"'.format(self.build_temp)) + build_args
  File "/opt/conda/envs/habitat2/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--', '-j']' returned non-zero exit status 2.
```

erikwijmans, 2020-02-25:
You do not need `--with-cuda` to use multiple GPUs (--headless can already target different GPUs).  Note that the simulator always uses just 1 GPU at a time.  I am not sure what is causing this error, but it seems related to your system setup -- gcc 5.4 may be too old for nvcc 10.2 to play nicely with it (it is technically supported, but in my experience, mvcc does not like older compilers).

> Also, I'm wondering about the difference between SIMULATOR_GPU_ID and TORCH_GPU_ID usage and which one should I use for training on multiple GPUs?

Those are only relevant to the PPO trainer -- `SIMULATOR_GPU_ID` specifies the GPU ID for the simulator to use and `TORCH_GPU_ID` specifies the GPU ID for pytorch to use. 

The DD-PPO trainer ignores those and assigns 1 GPU to each worker.

mhaboali, 2020-02-25:
Thanks for the fastest reply I got ever :)

I got what you mean. So If I'm using a machine with 2x or 8x RTX 2080 Ti, it doesn't boost my training process, am I correct?

If I'm wrong, how can I run the training process on such a machine as I read [here](https://openreview.net/attachment?id=H1gX8C4YPr&name=original_pdf)?

erikwijmans, 2020-02-25:
That is what the DD-PPO trainer is for.  You can use this script: https://github.com/facebookresearch/habitat-api/blob/master/habitat_baselines/rl/ddppo/single_node.sh and set `--nproc_per_node` to 8 and it'll run on 8 GPUs.

mhaboali, 2020-02-28:
I've tried to run this script after downloading the gibson data set under /path/to/habitat-api/data/datasets/gibson/v1

When I ran the script, I got these errors

`WARNING: Logging before InitGoogleLogging() is written to STDERR
E0228 04:58:23.710914  3021 ResourceManager.cpp:95] Cannot load from file data/scene_datasets/gibson/Rosser.glb
E0228 04:58:23.713742  3021 Simulator.cpp:107] cannot load data/scene_datasets/gibson/Rosser.glb
2020-02-28 04:58:27,293 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
E0228 04:58:27.423190  3026 ResourceManager.cpp:95] Cannot load from file data/scene_datasets/gibson/Albertville.glb
E0228 04:58:27.423388  3026 Simulator.cpp:107] cannot load data/scene_datasets/gibson/Albertville.glb
2020-02-28 04:58:27,931 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
E0228 04:58:28.030680  3035 ResourceManager.cpp:95] Cannot load from file data/scene_datasets/gibson/Sanctuary.glb
E0228 04:58:28.030863  3035 Simulator.cpp:107] cannot load data/scene_datasets/gibson/Sanctuary.glb
2020-02-28 04:58:28,112 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
E0228 04:58:28.192147  3025 ResourceManager.cpp:95] Cannot load from file data/scene_datasets/gibson/Hambleton.glb
E0228 04:58:28.192324  3025 Simulator.cpp:107] cannot load data/scene_datasets/gibson/Hambleton.glb
2020-02-28 04:58:29,208 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
E0228 04:58:29.271996  3031 ResourceManager.cpp:95] Cannot load from file data/scene_datasets/gibson/Stanleyville.glb
E0228 04:58:29.272173  3031 Simulator.cpp:107] cannot load data/scene_datasets/gibson/Stanleyville.glb
2020-02-28 04:58:29,615 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
E0228 04:58:29.706712  3033 ResourceManager.cpp:95] Cannot load from file data/scene_datasets/gibson/Ballou.glb
E0228 04:58:29.706907  3033 Simulator.cpp:107] cannot load data/scene_datasets/gibson/Ballou.glb
2020-02-28 04:58:30,193 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
E0228 04:58:30.272650  3027 ResourceManager.cpp:95] Cannot load from file data/scene_datasets/gibson/Maryhill.glb
E0228 04:58:30.272857  3027 Simulator.cpp:107] cannot load data/scene_datasets/gibson/Maryhill.glb
2020-02-28 04:58:30,414 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
E0228 04:58:30.502071  3029 ResourceManager.cpp:95] Cannot load from file data/scene_datasets/gibson/Goffs.glb
E0228 04:58:30.502261  3029 Simulator.cpp:107] cannot load data/scene_datasets/gibson/Goffs.glb
Process ForkServerProcess-1:`

I think I'm missing how to pass these paths to the config file, could you help with this error, please?

erikwijmans, 2020-02-28:
I just symlink things into where habitat-api expects them to be, do I haven't ever used the functionality to put things in different locations.  @mathfac do you know how that works?

mhaboali, 2020-02-29:
Ok, got your point. I read [this ](https://github.com/facebookresearch/habitat-api/issues/63#issuecomment-485025924) but I didn't manage to apply it.

Could you explain how to symlink clearer, please?

erikwijmans, 2020-02-29:
I just make a symlink from where the Gibson scenes actually are to `data/scene_datasets/gibson`, i.e. `ln -s /path/to/gibson data/scene_datasets/gibson` where you replace `/path/to/gibson` with the folder containing the Gibson scenes.

mhaboali, 2020-02-29:
Ok, thanks!

I'll try it soon and I'll keep you posted.

mhaboali, 2020-03-01:
I tried the symlink, but I failed to run the script(single_node.sh) and I got these errors

`2020-03-01 18:29:00,380 Initializing dataset PointNav-v1
2020-03-01 18:29:34,757 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
E0301 18:29:34.913580 14840 ResourceManager.cpp:95] Cannot load from file data/scene_datasets/gibson/Mobridge.glb
E0301 18:29:34.913942 14840 Simulator.cpp:107] cannot load data/scene_datasets/gibson/Mobridge.glb
2020-03-01 18:29:35,118 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0301 18:29:35.160429 14841 WindowlessContext.cpp:117] Check failed: isNvidiaGpuReadable(eglDevId) [EGL] EGL device 1, CUDA device 1 is not readable
*** Check failure stack trace: ***
Traceback (most recent call last):
  File "habitat_baselines/run.py", line 68, in <module>
    main()
  File "habitat_baselines/run.py", line 38, in main
    run_exp(**vars(args))
  File "habitat_baselines/run.py", line 62, in run_exp
    trainer.train()
  File "/habitat-api/habitat_baselines/rl/ddppo/algo/ddppo_trainer.py", line 168, in train
    self.config, get_env_class(self.config.ENV_NAME)
  File "/habitat-api/habitat_baselines/common/env_utils.py", line 95, in construct_envs
    tuple(zip(configs, env_classes, range(num_processes)))
  File "/habitat-api/habitat/core/vector_env.py", line 135, in __init__
    read_fn() for read_fn in self._connection_read_fns
  File "/habitat-api/habitat/core/vector_env.py", line 135, in <listcomp>
    read_fn() for read_fn in self._connection_read_fns
  File "/opt/conda/envs/habitat2/lib/python3.6/multiprocessing/connection.py", line 250, in recv
    buf = self._recv_bytes()
  File "/opt/conda/envs/habitat2/lib/python3.6/multiprocessing/connection.py", line 407, in _recv_bytes
    buf = self._recv(4)
  File "/opt/conda/envs/habitat2/lib/python3.6/multiprocessing/connection.py", line 379, in _recv
    chunk = read(handle, remaining)
ConnectionResetError: [Errno 104] Connection reset by peer
Exception ignored in: <bound method VectorEnv.__del__ of <habitat.core.vector_env.VectorEnv object at 0x7f7ee87335f8>>
Traceback (most recent call last):
  File "/habitat-api/habitat/core/vector_env.py", line 468, in __del__
    self.close()
  File "/habitat-api/habitat/core/vector_env.py", line 350, in close
    write_fn((CLOSE_COMMAND, None))
  File "/opt/conda/envs/habitat2/lib/python3.6/multiprocessing/connection.py", line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File "/opt/conda/envs/habitat2/lib/python3.6/multiprocessing/connection.py", line 404, in _send_bytes
    self._send(header + buf)
  File "/opt/conda/envs/habitat2/lib/python3.6/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
BrokenPipeError: [Errno 32] Broken pipe
2020-03-01 18:29:37,445 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
E0301 18:29:37.528736 14838 ResourceManager.cpp:95] Cannot load from file data/scene_datasets/gibson/Sasakwa.glb
E0301 18:29:37.529157 14838 Simulator.cpp:107] cannot load data/scene_datasets/gibson/Sasakwa.glb
Process ForkServerProcess-3:
Traceback (most recent call last):
  File "/opt/conda/envs/habitat2/lib/python3.6/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
`

erikwijmans, 2020-03-02:
> Check failed: isNvidiaGpuReadable(eglDevId) [EGL] EGL device 1, CUDA device 1 is not readable

That indicates that the GPU device file (`/dev/nvidia1` in this case) isn't readable by the user.  This is an odd state for the GPU to be in, please contact whoever manages that machine.

mhaboali, 2020-03-02:
Ok, I will contact them.

Great thanks!

aclegg3, 2022-08-31:
Seems resolved or out of domain, closing the issue.

closed 2022-08-31 by aclegg3