facebookresearch/habitat-sim #506: Material manipulation

matthewjmay, 2020-02-26:
## 🚀 Feature
Allow new materials to be programatically created, and loaded materials to be progamatically manipulated.

## Motivation

Now that we have phong shading, manipulating materials programatically would allow for tweaking existing models or building scenes from primitives programatically, especially as we currently have some issues with parsing PBR materials into a phong shading model.

An example of where this would be useful is domain randomization (#361), and this also has user requests already (#502).

## Pitch

For materials (right now only phong model supported), users should be able to create materials, and manipulate existing materials. Properties which can be manipulated will be:
* Colors (ambient, diffuse, specular)
* Shininess
* Textures (ambient, diffuse, specular)

Users should also be able to query these properties.

For assets loaded from files, we can allow users to query for material IDs (indexes) used by the physics object template so they can easily be manipulated. Another option would be to have the material key be `{filename}-{index}`.

## Alternatives

PBR shader should be implemented, but this feature is still required for building scenes from primitives and domain randomization

ronjamoller, 2020-03-13:
Hi, i am having a material issue that im guessing is related to the new phong shading. My imported meshes looked fine (the surface setting in blender is principled BSDF) in the last version but now it is incredibly shiny and looks unrealistic. Can someone recommend a surface setting in blender that fixes this (i tried several but none of them work) or if that is not possible the version of habitat i have to roll back to to recover the previous state without phong shading ? I am not too bothered by the shininess per se and im sure this will be worked on soon, but i need to create images for next week and they cannot be shiny.
thanks

aclegg3, 2020-03-13:
Thanks for bringing this up! 👍 

At the moment, we are aware that Blender BSDF materials are not being handled well by our Phong shading model. We do intend to allow manipulation of individual materials after asset loading similar to the current configurable LightSetup options. 

For now, **you can opt into the old flat shading model** by adding the `"requires lighting": false` flag to the physics config file for your object as shown below. Once we have a better idea of Blender material compatibility we'll let you know.

[image]

ronjamoller, 2020-03-13:
Thank you very much for getting back to me so quickly, unfortunately my mesh is still shiny.
[image]
[image]

Could this be an issue with the version ? I updated to the latest stable version last about 1.5 to 2 weeks ago.

aclegg3, 2020-03-13:
You are exactly right. This option was introduced 9 days ago on PR #515.

ronjamoller, 2020-03-13:
Thank you so much ! It is running now with the latest stable version :)

aclegg3, 2020-03-13:
Happy to hear it. 👍 We are working on material support solutions and will continue to update this issue.

yyf17, 2022-01-02:
@ronjamoller How did you add the model to the simulator? What is the version of your Habitat-sim?