facebookresearch/habitat-sim #511: zsh: no such file or directory: build/viewer

chale96, 2020-02-28:
## ❓ Questions and Help

When I am running `build/viewer --enable-physics /path/to/data/scene_datasets/habitat-test-scenes/van-gogh-room.glb` for the testing with Physical interactions (https://github.com/facebookresearch/habitat-sim#testing)

It returns me this error: 
`zsh: no such file or directory: build/viewer`

aclegg3, 2020-02-28:
Thanks for pointing this out. 👍 I've updated the README, it should say:

```
./build/viewer --enable-physics /path/to/data/scene_datasets/habitat-test-scenes/van-gogh-room.glb
```

Also note that you should replace "`/path/to`" with the relative or global path to the `habitat-sim` directory on your system.

chale96, 2020-02-29:
I'm still having the same issue: if from the terminal the autocomplete for the path it finds `build/` but it seems that can't see the `viewer` file. 
Checking in the folder I can see an alias file named `viewer`

erikwijmans, 2020-02-29:
If you built with `--headless`, the viewer won't be built (it requires an attached display).

chale96, 2020-02-29:
> If you built with `--headless`, the viewer won't be built (it requires an attached display).

I followed the instructions for normal building (not `--headless`) + the mac `bullet` instructions

chale96, 2020-03-01:
I am trying to reinstall it and when I do `python setup.py install` it gives me this error:

subprocess.CalledProcessError: Command '['cmake', '-H/Users/alessandrocandeloro/Documents/University/in3007_ind_proj/implementation/habitat-sim/src', '-Bbuild', '-DBUILD_PYTHON_BINDINGS=ON', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/Users/alessandrocandeloro/Documents/University/in3007_ind_proj/implementation/habitat-sim/build/lib.macosx-10.9-x86_64-3.6/habitat_sim/_ext', '-DPYTHON_EXECUTABLE=/Users/alessandrocandeloro/anaconda3/envs/habitat/bin/python', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-GNinja', '-DBUILD_GUI_VIEWERS=ON', '-DBUILD_TEST=OFF', '-DBUILD_WITH_BULLET=OFF', '-DBUILD_DATATOOL=OFF', '-DBUILD_WITH_CUDA=OFF']' returned non-zero exit status 1.

erikwijmans, 2020-03-01:
Can you provide more of the error log?  That is simply telling us that the command failed, not why it failed

chale96, 2020-03-01:
This is the full process it does when I run `python setup.py install`

running install
running bdist_egg
running egg_info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.macosx-10.9-x86_64/egg
running install_lib
running build_py
running build_ext
CMake Error: The current CMakeCache.txt directory /Users/alessandrocandeloro/Documents/University/in3007_ind_proj/implementation/habitat-sim/build/CMakeCache.txt is different than the directory /Users/alessandrocandeloro/Documents/University/in3007_ind_proj/implementation/habitat-api/habitat-sim/build where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
CMake Error: The source "/Users/alessandrocandeloro/Documents/University/in3007_ind_proj/implementation/habitat-sim/src/CMakeLists.txt" does not match the source "/Users/alessandrocandeloro/Documents/University/in3007_ind_proj/implementation/habitat-api/habitat-sim/src/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
Traceback (most recent call last):
  File "setup.py", line 382, in <module>
    include_package_data=True,
  File "/Users/alessandrocandeloro/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/__init__.py", line 144, in setup
    return distutils.core.setup(**attrs)
  File "/Users/alessandrocandeloro/anaconda3/envs/habitat/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/Users/alessandrocandeloro/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/Users/alessandrocandeloro/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/Users/alessandrocandeloro/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/Users/alessandrocandeloro/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/Users/alessandrocandeloro/anaconda3/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/Users/alessandrocandeloro/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/Users/alessandrocandeloro/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 172, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/Users/alessandrocandeloro/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 158, in call_command
    self.run_command(cmdname)
  File "/Users/alessandrocandeloro/anaconda3/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/Users/alessandrocandeloro/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/Users/alessandrocandeloro/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/Users/alessandrocandeloro/anaconda3/envs/habitat/lib/python3.6/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/Users/alessandrocandeloro/anaconda3/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/Users/alessandrocandeloro/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 202, in run
    self.build_extension(ext)
  File "setup.py", line 271, in build_extension
    env=env,
  File "/Users/alessandrocandeloro/anaconda3/envs/habitat/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '-H/Users/alessandrocandeloro/Documents/University/in3007_ind_proj/implementation/habitat-sim/src', '-Bbuild', '-DBUILD_PYTHON_BINDINGS=ON', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/Users/alessandrocandeloro/Documents/University/in3007_ind_proj/implementation/habitat-sim/build/lib.macosx-10.9-x86_64-3.6/habitat_sim/_ext', '-DPYTHON_EXECUTABLE=/Users/alessandrocandeloro/anaconda3/envs/habitat/bin/python', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-GNinja', '-DBUILD_GUI_VIEWERS=ON', '-DBUILD_TEST=OFF', '-DBUILD_WITH_BULLET=OFF', '-DBUILD_DATATOOL=OFF', '-DBUILD_WITH_CUDA=OFF']' returned non-zero exit status 1.

if the failure of those commands are not a problem then the error is somewhere else because when I run `./build/viewer --enable-physics data/scene_datasets/habitat-test-scenes/van-gogh-room.glb` it still giving me zsh: no such file or directory: ./build/viewer as error.

erikwijmans, 2020-03-01:
Yes, the failure of these commands is definitely the problem -- habitat-sim is not building and the viewer binary is not getting built!

Can you try a clean re-build, i.e. `rm -rf build` then `python setup.py install` (+ bullet if you want)?

chale96, 2020-03-01:
It works perfectly,
I guess I made some error while installing and reinstalling the habitat.

Thank you for your support.

closed 2020-03-03 by chale96

KenaHemnani, 2020-11-18:
@chale96 and @erikwijmans , how this error was solved I am facing the same issue.
That command gives following error:
 ./build/viewer: No such file or directory