facebookresearch/habitat-sim #524: How to get world coordinates

bareblackfoot, 2020-03-11:
## ❓ Questions and Help
Hi, I want to know how to get the world coordinates.
I tried an unproject method in the demo code, but it is quite inaccurate.
I believe that there are ground truth 3d points in the scene in the matterport3d dataset.

erikwijmans, 2020-03-11:
Can you provide reproduction code for the inaccuracy -- those should be accurate.  One issue that you may run into is that habitat-sim rotates the MP3D scene meshes to be -Y gravity instead of -Z gravity.  You can correct that with

```python
from habitat_sim.utils.common import quat_from_two_vectors, quat_rotate_vector
from habitat_sim import geo

rotation_mp3d_habitat = quat_from_two_vector(geo.GRAVITY, np.array([0, 0, -1])

pt_habitat = ... # Some point in habitat-sim's coordinate frame
pt_mp3d = quat_rotate_vector(rotation_mp3d_habitat, pt_habitat) # That point in the mp3d scene mesh coordinate frame.
```

bareblackfoot, 2020-03-12:
Thank you for reply!
Since my code has a lot of functions and classes, I just wrote down the key parts.
Image size is 300x300 and the sensor has 90 fov (therefore, I did not use intrinsic camera matrix K, which is identity matrix.)
Here I write down an unprojection code for a single frontal camera, but I used 4 rgb/depth sensors at the front, left, right, back. (to make a custom panoramic image)
If you need more information, please ask me.

#```
Firstly, I randomly select a navigable point. 
position = self._sim.pathfinder.get_random_navigable_point()

# I observed that the simulator randomly generates a rotation of a robot.
scene_file = self._sim_settings["scene"]
if (not os.path.exists(scene_file) and scene_file == default_sim_settings["test_scene"]):
     print("Test scenes not downloaded locally, downloading and extracting now...")
     utils.download_and_unzip(default_sim_settings["test_scene_data_url"], ".")
     print("Downloaded and extracted test scenes data.") 
     self._sim.reset()
     random.seed(self._sim_settings["seed"])
     self._sim.seed(self._sim_settings["seed"])

# initialize the agent at a random start state
state = self.init_agent_state(self._sim_settings["default_agent"])
rotation = state.rotation.components

color_obs_f = observations["front_sensor"]
color_img_f = Image.fromarray(color_obs_f, mode="RGBA").convert("RGB")
depth = observations["depth_front"]
depth = np.clip(depth, 0., 10.)
xs, ys = np.meshgrid(np.linspace(-1, 1, 300), np.linspace(1, -1, 300))
xs = np.expand_dims(xs, 0) 
ys = np.expand_dims(ys, 0)
xyz = np.vstack((xs * depth, ys * depth, -depth, np.ones(depth.shape)))

# Fo left, right, back image, I rotated 3d points around y axis when their center is (0, 0, 0). For example,
angle = 90 (since y axis is gravity direction, rotation to the left has positive sign)
xyz_l = xyz_l.reshape(4, -1)
rot_mtx = np.array([[np.cos(angle * np.pi / 180), 0, np.sin(angle * np.pi / 180), 0], [0, 1, 0, 0], [-np.sin(angle * np.pi / 180), 0, np.cos(angle * np.pi / 180), 0], [0, 0, 0, 1]])
xyz_l = np.matmul(rot_mtx, xyz_l))
xyz_l = out.reshape(xyz_l.shape) 

#After that, I concatenate all of the 3d points from left, front, right, back.
xyz = np.concatenate((xys_l.reshape(4, 300, 300), xys_f.reshape(4, 300, 300), xys_r.reshape(4, 300, 300), xys_b.reshape(4, 300, 300)), -1).reshape(4, -1)

# Translate/rotation points to the world.
rotation = quaternion.as_rotation_matrix(quaternion.as_quat_array(rotation))
T_world_camera = np.eye(4)
T_world_camera[0:3, 0:3] = rotation
T_world_camera[0:3, 3] = translation
xyz = np.matmul(T_world_camera, xyz)

For your information, The depth sensor:
@habitat.registry.register_sensor(name="angle_depth_sensor")
class Angle_DepthSensor(DepthSensor):
    sim_sensor_type: habitat_sim.SensorType
    min_depth_value: float
    max_depth_value: float

    def __init__(self, config):
        self.sim_sensor_type = habitat_sim.SensorType.DEPTH
        self.angle = config.ANGLE
        self.config = config
        if config.NORMALIZE_DEPTH:
            self.min_depth_value = 0
            self.max_depth_value = 1
        else:
            self.min_depth_value = config.MIN_DEPTH
            self.max_depth_value = config.MAX_DEPTH

        super().__init__(config=config)

    # Defines the name of the sensor in the sensor suite dictionary
    def _get_uuid(self, *args: Any, **kwargs: Any):
        return "depth_" + self.angle

    # Defines the type of the sensor
    def _get_sensor_type(self, *args: Any, **kwargs: Any):
        return habitat.SensorTypes.DEPTH

    # This is called whenever reset is called or an action is taken
    def _get_observation_space(self, *args: Any, **kwargs: Any):
        return spaces.Box(
            low=self.min_depth_value,
            high=self.max_depth_value,
            shape=(self.config.HEIGHT, self.config.WIDTH, 1),
            dtype=np.float32,
        )

    def get_observation(self, sim_obs):
        obs = sim_obs.get(self.uuid, None)
        check_sim_obs(obs, self)

        obs = np.clip(obs, self.config.MIN_DEPTH, self.config.MAX_DEPTH)

        obs = np.expand_dims(
            obs, axis=2
        )  # make depth observation a 3D array
        if self.config.NORMALIZE_DEPTH:
            # normalize depth observation to [0, 1]
            obs = (obs - self.config.MIN_DEPTH) / (
                self.config.MAX_DEPTH - self.config.MIN_DEPTH
            )
        return obs

```

closed 2020-03-12 by bareblackfoot

KenaHemnani, 2020-12-22:
Hii I also want to know how to get 3D world coordinate point using 2D depth image? Is there any method which gives this?Or we need to calculate it somehow?

erikwijmans, 2020-12-22:
https://aihabitat.org/docs/habitat-lab/view-transform-warp.html

KenaHemnani, 2020-12-22:
Thank you very much for clearing the doubt.
Hence, according to this code we don't need to multiply by 10 the depth image values . The depth matric  itself gives distance in meters . Is that so? 
`if depth_obs.size != 0:
        depth_img = Image.fromarray((depth_obs / 10 * 255).astype(np.uint8), mode="L")
        arr.append(depth_img)
        titles.append("depth")`
Referred from https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/master/examples/tutorials/colabs/ECCV_2020_Navigation.ipynb .

erikwijmans, 2020-12-22:
That is mapping depth values in [0, 10] to [0, 255] for display as an image.

KenaHemnani, 2020-12-22:
Yeah Thank you very much!!

nikhilaravi, 2021-02-09:
@erikwijmans to clarify, is the world coordinate system in habitat +Y up, +X left, +Z into the screen?  or is it +Y up, +X right, +Z out of the screen?

erikwijmans, 2021-02-09:
The second one

derrick-xwp, 2022-06-28:
mark