facebookresearch/habitat-sim #527: Error in WebGL Habitat-Sim while running the webserver

rutadesai, 2020-03-14:
## 🐛 Bug

I followed the instruction on https://github.com/facebookresearch/habitat-sim#webgl to run the webgl demo. However, in step 5., when I run: `python -m http.server 8000 --bind 127.0.0.1` , I get the following error: 

```
127.0.0.1 - - [14/Mar/2020 14:53:19] "GET /build_js/esp/bindings_js/bindings.html HTTP/1.1" 200 -
127.0.0.1 - - [14/Mar/2020 14:53:20] "GET /build_js/esp/bindings_js/WebApplication.css HTTP/1.1" 200 -
127.0.0.1 - - [14/Mar/2020 14:53:20] "GET /build_js/esp/bindings_js/WindowlessEmscriptenApplication.js HTTP/1.1" 200 -
127.0.0.1 - - [14/Mar/2020 14:53:20] "GET /build_js/esp/bindings_js/bundle.js?4c536547307b8acf0c58 HTTP/1.1" 200 -
127.0.0.1 - - [14/Mar/2020 14:53:20] code 404, message File not found
127.0.0.1 - - [14/Mar/2020 14:53:20] "GET /build_js/esp/bindings_js/hsim_bindings.js HTTP/1.1" 404 -
127.0.0.1 - - [14/Mar/2020 14:53:20] code 404, message File not found
```

Corresponding output in the browser: 

[image]

```
python -m http.server 8000 --bind 127.0.0.1
Serving HTTP on 127.0.0.1 port 8000 (http://127.0.0.1:8000/) ...
127.0.0.1 - - [14/Mar/2020 14:53:19] "GET /build_js/esp/bindings_js/bindings.html HTTP/1.1" 200 -
127.0.0.1 - - [14/Mar/2020 14:53:20] "GET /build_js/esp/bindings_js/WebApplication.css HTTP/1.1" 200 -
127.0.0.1 - - [14/Mar/2020 14:53:20] "GET /build_js/esp/bindings_js/WindowlessEmscriptenApplication.js HTTP/1.1" 200 -
127.0.0.1 - - [14/Mar/2020 14:53:20] "GET /build_js/esp/bindings_js/bundle.js?4c536547307b8acf0c58 HTTP/1.1" 200 -
127.0.0.1 - - [14/Mar/2020 14:53:20] code 404, message File not found
127.0.0.1 - - [14/Mar/2020 14:53:20] "GET /build_js/esp/bindings_js/hsim_bindings.js HTTP/1.1" 404 -
127.0.0.1 - - [14/Mar/2020 14:53:20] code 404, message File not found
127.0.0.1 - - [14/Mar/2020 14:53:20] "GET /build_js/esp/bindings_js/hsim_bindings.js HTTP/1.1" 404 -
```

## System Info

ENVIRONMENT INFO:
Platform: Darwin-18.7.0-x86_64-i386-64bit
Machine: x86_64
Processor: i386
Libc version:
Mac version: 10.14.6
Python version: 3.6.10
Architecture: 64bit
Win version:
System OS: Darwin
Release: 18.7.0
Version: Darwin Kernel Version 18.7.0: Tue Aug 20 16:57:14 PDT 2019; root:xnu-4903.271.2~2/RELEASE_X86_64
Operational System: darwin
GCC version: Apple LLVM version 10.0.1 (clang-1001.0.46.4)
CMAKE version: cmake version 3.14.0
NVIDIA-SMI:
Pip packages versions:

Conda packages versions:
attrs                     19.3.0                     py_0    19.1.0
numba                     0.48.0                   pypi_0    pypi
numpy                     1.18.1                   pypi_0    pypi
numpy-quaternion          2019.12.11.22.25.52          pypi_0    pypi
pillow                    7.0.0            py36h4655f20_0
scipy                     1.4.1            py36h9fa6033_0    1.3.0
tqdm                      4.43.0                   pypi_0    pypi

dhruvbatra, 2020-03-14:
CC: @apsdehal and @msbaines

msbaines, 2020-03-16:
This is regression caused by #334 which changed RUNTIME_OUTPUT_DIRECTORY

You can temporarily workaround via:

```
cp -v build_js/Release/bin/hsim_bindings.* build_js/esp/bindings_js/
```

dhruvbatra, 2020-03-16:
CC: @mosra about the regression.

rutadesai, 2020-03-31:
I tried what @msbaines suggested. I don't get the 404 (file not found) error anymore. No other errors in the terminal too. But the browser still doesn't show anything. I also see `wasm streaming compile failed: TypeError: Failed to execute 'compile' on 'WebAssembly': Incorrect response MIME type. Expected 'application/wasm'.
falling back to ArrayBuffer instantiation` error in the browser. 

Output in the browser:

[image]

vidhiJain, 2020-10-20:
I tried the same steps and copied the RUNTIME_OUTPUT_DIRECTORY bindings but I still get the file not found error.  
[image]
The viewer.html and bindings.html code seems to load 'hsim_bindings.js' which is a huge monolith of javascript. I have not found the way to check the path for the scenes yet. How to proceed in this case?

apsdehal, 2020-10-21:
Hi,
As per the error message, it seems like you are missing the required files in the `/build_js/esp/bindings_js/` folder itself. Can you copy them into that folder? The first point of https://github.com/facebookresearch/habitat-sim#webgl describes where to find the test data.

PR #985 (2020-12-18): [WebGL] Viewer improve CMake  and folder path handling
## Motivation and Context

This solves two issues. One it removes the need for a cp after building the viewer and also allows the viewer to take in arbitrary relative urls (including those in folders or absolute file locations).

Now we can have folders of scenes, configs, and other params on the webserver and they will load properly.

## How Has This Been Tested
* Locally

## Types of changes

- [X] Docs change / refactoring / dependency upgrade
- [X] Bug fix (non-breaking change which fixes an issue)

## Checklist

- [X] My code follows  [... cut]
closed 2020-12-18 by Skylion007