facebookresearch/habitat-sim #534: Build issue

HelloN00bman, 2020-03-18:
## 🐛 Bug

CMake errors out during a subprocess call while building using setupy.py. It looks like it is complaining that it cannot find some assimp object files. 

## Command
`python setup.py install --bullet`

## To Reproduce

Steps to reproduce the behavior:

1. `git clone https://github.com/facebookresearch/habitat-sim.git`
2. `cd habitat-sim`
3. `git checkout tags/stable`
4. `python setup.py install --bullet`

Full build messages/errors: [build_error.txt](https://github.com/facebookresearch/habitat-sim/files/4350374/build_error.txt).

## Expected behavior

habitat-sim builds with no issues. 

## Additional context

I have been able to build this before. Perhaps, this is related to automatic system updates?

## System Info

ENVIRONMENT INFO:
Platform: Darwin-19.3.0-x86_64-i386-64bit
Machine: x86_64
Processor: i386
Libc version:
Mac version: 10.15.3
Python version: 3.6.10
Architecture: 64bit
Win version:
System OS: Darwin
Release: 19.3.0
Version: Darwin Kernel Version 19.3.0: Thu Jan  9 20:58:23 PST 2020; root:xnu-6153.81.5~1/RELEASE_X86_64
Operational System: darwin
GCC version: Apple clang version 11.0.0 (clang-1100.0.33.17)
CMAKE version: cmake version 3.14.0
NVIDIA-SMI:
Pip packages versions:

Conda packages versions:

 - OS (e.g., Linux): macOS Catalina
 - How you installed PyTorch (`conda`, `pip`, source): conda install pytorch torchvision -c pytorch
 - Build command you used (if compiling from source): python setup.py install --bullet
 - GPU models and configuration: None
 - CUDA version: None

erikwijmans, 2020-03-18:
Looking at the errors, this appears to be an issue with your system compiler.
```
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:318:9: error: no member named 'signbit' in the global namespace
using ::signbit;
      ~~^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:319:9: error: no member named 'fpclassify' in the global namespace
using ::fpclassify;
      ~~^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:320:9: error: no member named 'isfinite' in the global namespace; did you mean 'finite'?
using ::isfinite;
```

A quick google search shows that maybe this is a Catalina issue (https://stackoverflow.com/questions/58628377/catalina-c-using-cmath-headers-yield-error-no-member-named-signbit-in-th).

mosra, 2020-03-26:
I remember a similar issue happening after Xcode upgrade and it was solved by doing something like `xcode-select --install` ... *I think*. Can't find the right magic incantation that I used back when this happened to me.

closed 2020-05-10 by dhruvbatra