facebookresearch/habitat-sim #543: The gltf coordinate system and habitat coordinate system

xiangyuanhang, 2020-03-26:
## ❓ Questions and Help
Hi,

I'm trying to select some start/end points manully from Gibson-dataset.
First I opened the glb model using blender.
And then I visualized several points using the coords in the point-nav json file.
However, these points appear not in the house. The picture below shows the model Adrian.glb and several points from pointnav/train/content/Adrian.json
[image]
So I wonder the points in the json file are not in the same coordinate system with the glb model?
Then I followed the instruction that @erikwijmans provided.https://github.com/facebookresearch/habitat-sim/issues/524
But  those points from json file are still not in the house model.
[image]
Could this be the problem of blender? Or there are some other transformations that need to take?

If so, is there any convenient approach to manually select some points from the model and then configure my own episode?

Thank you so much!

aclegg3, 2020-04-11:
Hey @xiangyuanhang, yes this is Blender coordinate system related.

Blender and glTF have differing coordinate systems: https://github.com/KhronosGroup/glTF-Blender-Exporter/blob/master/docs/developer.md

The instructions provided by Erik are correct in that once the model is imported into Habitat an additional 90 degree rotation is applied for glb assets.

To get a Habitat-sim global point you can apply the mapping Blender (x,y,z) -> glTF(x,z,-y) to your point and then apply the inverse of the transformation we add as Erik suggests. 

The resulting transformation is actually just a 180 rotation around the x-axis, so you can directly apply the mapping: Blender(x,y,z) -> Habitat(x,-y,-z) for these glb scenes!

Hope this helps! 👍

xiangyuanhang, 2020-05-18:
Thank you so much. It works perfectly.

closed 2020-11-20 by dhruvbatra