facebookresearch/habitat-sim #552: Queries about using Matterport

MADHAVAN001, 2020-03-30:
## ❓ Questions and Help

Is the mesh provided in matterport habitat tasks, complete?

I downloaded the dataset using `python download_mp.py --task habitat -o path/to/download/`

I am using habitat_sim.utils.data.data_extractor's Image extractor to generate free poses and images and the number of images obtained from 17DRP5sb8fy is only 21.

If I want to load the entire  house1, house2 mesh in habitat, how do I proceed about this?

dhruvbatra, 2020-03-30:
Could you please say more -- what are you trying to achieve?

CC: @mpiseno

MADHAVAN001, 2020-03-30:
I am trying to extract free poses in the environment where the agent can move around. For all such places, I want to extract the image that the agent sees.

```python
from habitat_sim.utils.data.data_extractor import ImageExtractor

scene_filepath = "/path/to/mp3d/17DRP5sb8fy/17DRP5sb8fy.glb"

extractor = ImageExtractor(
    scene_filepath,
    labels=[0.0],
    img_size=(512, 512),
    output=["rgba", "depth", "semantic"],
)

print(len(extractor[:1000]))

```

dhruvbatra, 2020-03-30:
I'm sure @mpiseno can let you know how to increase that number (21). 

However, please keep in mind that the number of places the agent can "stand" is infinite, so you can't get all such images.

MADHAVAN001, 2020-03-30:
I don't want an infinite number of images, but I want to control the number for my experiment.

Right now, I am not so sure about how to do that

mpiseno, 2020-03-30:
The way the image extractor works (by default) is by creating a grid of camera positions, and for each camera position, the camera angle is determined by finding the closest non-navigable point to that position. For some scenes this results in a low number of images (e.g. 21). Creating rules for extracting more images from each scene is an active area of work and is non-trivial, but there are currently 2 ways you can get more images from the extractor:

1. Set the extraction_method parameter to "panorama". This spins the camera around to get more camera poses rather than finding the closest non-navigable point.
2. Instead of 1 scene, you can pass in a list of ".glb" or ".ply" files instead of just 1. Just note that doing this can be very memory intensive because you'll be switching scenes a lot and it might slow down any training pipelines you have. 

Overall the image extractor is still a work in progress so please feel free to give feedback on existing features or features that would be helpful for the future.

MADHAVAN001, 2020-03-31:
This makes sense. 

However, in a reasonably large scene, it is difficult to imagine that there is only a small number of free poses. Theoretically, I must be able to make the grid infinitesimally small to capture more images from the environment with small movement in agent positions. This is why in theory, we should be able to take infinite images. Is there a way I can tweak to this type of behavior?

Also, could you please explain what you mean by the closest non-navigable point? 

Additionally, such an approach to use multiple scenes would not work if the model was scene-specific. There should be flexibility in the simulator to achieve this.

mpiseno, 2020-03-31:
Yes if you look inside pose_extractor.py on line 43, "dist" determines how many pixels each camera position is spaces apart from the point of view of the topdown map. So if the output size is 512 X 512  and dist = 10, the grid would be 50 x 50. Modifying this value to a lower number will make the camera poses closer together. In retrospect, this should probably be a user-defined parameter.

closed 2020-05-10 by dhruvbatra