facebookresearch/habitat-sim #556: Add CI Testing with "heavy" meshes

dhruvbatra, 2020-04-01:
## 🚀 Feature
We should add CI testing with "heavy" meshes like MP3D and Gibson, so that we are certain we aren't suffering a regression. 

## Motivation

We suffered a major unintentional regression via #542. Our current CI testing uses a fairly light testing mesh: 
https://github.com/facebookresearch/habitat-sim/blob/ddf84ad79164bdc30aeff233f8932aac0b757fe2/.circleci/config.yml#L289

Our hypothesis is that if we test with heavier meshes like Replica, MP3D, Gibson, we would have caught the regression as a failed test.

mosra, 2020-04-01:
(Sorry about that regression again.)

Not only a heavy mesh, we also need something that verifies semantic output / meshes with semantic info. Looking at the report, basically [all](https://codecov.io/gh/facebookresearch/habitat-sim/src/master/src/esp/assets/GenericInstanceMeshData.cpp) [code](https://codecov.io/gh/facebookresearch/habitat-sim/src/master/src/esp/gfx/PrimitiveIDShader.cpp)  related to it is completely uncovered. I'm currently removing those and replacing them with builtin Magnum functionality, but having no automated regression testing in place for these makes me wary.

soumith, 2020-05-02:
Added an issue: https://github.com/facebookresearch/habitat-sim/issues/613 which would benefit from this. Basically all ReplicaDataset environments are broken when using Bullet physics.

aclegg3, 2022-03-01:
We are now testing in CI with HM3D and MP3D example scan assets containing semantics. This does not address the Replica physics issue.

closed 2022-03-01 by aclegg3