facebookresearch/habitat-sim #567: Using noise model for DepthSensor

srama2512, 2020-04-04:
## ❓ Questions and Help
I was interested in getting the noisy depth inputs from Habitat-sim for training PointNav agents. For some reason, the depth outputs don't make any sense at all. For example, find some images below that were generated from the `test_sensors.py` script in habitat-api.  The left-hand side is the noisy RGB data with intensity 0.5 and the right-hand side is the noisy depth data with the Redwood depth model. It seemed more relevant to ask the question here because the noise models are primarily in habitat-sim.

[image]
[image]
[image]

srama2512, 2020-04-05:
I was using the stable version of habitat-sim. When I changed to the challenge-2020 tag, things started working well! May be this is worth keeping track of somewhere.

[image] 
[image]
[image]

erikwijmans, 2020-04-05:
Hi, can you provide additional details?  What settings did compile on, what operating system, etc?

I tried to reproduce locally (on MacOS) by adding
```
cv2.imshow("Depth Noise", obs["depth"])
cv2.waitKey(0)
```
to the test you referenced in Habitat-API, but all looks fine.

srama2512, 2020-04-05:
Hi Erik,

Sure. This was compiled on an Ubuntu 18.04.3 LTS with the following:
- gcc 7.4.0
- cmake 3.10.2
- python 3.7
- cuda 10.1
# Testing
I installed habitat-sim as follows:
```
cd $HABITAT_ROOT
git clone --branch stable https://github.com/facebookresearch/habitat-sim.git
cd habitat-sim
pip install -r requirements.txt
python setup.py install --headless --with-cuda
``` 

Habitat-api was a bit different. I had an older version that I made changes to for my project and then merged it with the master version (as of April 3rd, 2020).  In that merged habitat-api repository, I did the following:
```
cd $HABITAT_ROOT/habitat-api
ipython
>>> cd test
>>> from test_sensors import test_noise_models_rgbd
>>> cd ../
>>> test_noise_models_rgbd()
```
The `cd ...` steps were needed to bypass some path issues. I modified `test_noise_models_rgbd()` by setting `DEMO_MODE = True`. The screenshots I sent were from that video. Please let me know if there are any other details that I can add. 

# Fix
```
cd $HABITAT_ROOT
rm -rf habitat-sim
git  clone --branch challenge-2020 https://github.com/facebookresearch/habitat-sim.git
cd habitat-sim
pip install -r requirements.txt
python setup.py install --headless --with-cuda
```

erikwijmans, 2020-04-05:
Hmm, still can't reproduce.  Since this is fixed for you now, I will go ahead and close it.

closed 2020-04-05 by erikwijmans

srama2512, 2020-04-06:
Alright, thanks.