facebookresearch/habitat-sim #576: Unexpected behavior of Semantic Sensor after Object Insertion

shivanshpatel35, 2020-04-09:
## Bug

Hi, I am inserting objects using habitat-api. When I visualize results, RBG and Depth sensors work as expected, but the Semantic Sensor ignores the inserted object. This can be verified [here](https://drive.google.com/open?id=1DFALGg_5jW7v3KrZbh3cmblvRRJjQaWk). 
Also, the inserted object does not appear in `env._sim.semantic_annotations()`. Perhaps some existing functionalities do not completely support inserted objects.
@aclegg3

aclegg3, 2020-04-10:
@shivanshpatel35 thanks for opening this issue. You are correct that semantic rendering of non-scene objects is not fully supported at this time. There is a draft PR (#341) which began to address this awhile back. This is something which we should look into again.

closed 2020-04-15 by shivanshpatel35

shivanshpatel35, 2020-05-10:
Hi @aclegg3 , 
I notice that there are no updates in PR #341 . Will this be fixed anytime soon?

aclegg3, 2020-07-07:
PR #668 addresses this by enabling configurable semantic ids for rendering non-scene objects with `SEMANTIC` sensor.