facebookresearch/habitat-sim #581: Viewer crashing when trying to visualize the navmesh

dexter1691, 2020-04-09:
## 🐛 Bug
The latest master is crashing when trying to visualize the navmesh while running the Viewer. 

## Command
`./build/viewer --enable-physics data/scene_datasets/habitat-test-scenes/skokloster-castle.glb`

## To Reproduce
Steps to reproduce the behavior:
- Run the above command, and then press N to visualize the navmesh

## System Info
Platform: Darwin-19.3.0-x86_64-i386-64bit
Machine: x86_64
Processor: i386
Libc version:
Mac version: 10.15.3
Python version: 3.6.10
Architecture: 64bit
Win version:
System OS: Darwin
Release: 19.3.0
Version: Darwin Kernel Version 19.3.0: Thu Jan  9 20:58:23 PST 2020; root:xnu-6153.81.5~1/RELEASE_X86_64
Operational System: darwin
GCC version: Apple clang version 11.0.0 (clang-1100.0.33.17)
CMAKE version: cmake version 3.14.0
NVIDIA-SMI:
Pip packages versions:

Conda packages versions:

jturner65, 2020-04-09:
Please see if this has been addressed by patch.  Thanks!

dexter1691, 2020-04-10:
Yes. This fixes it! Thanks

closed 2020-04-10 by dexter1691