facebookresearch/habitat-sim #584: Misleading CUDA Installation Guidance

Xiaoming-Zhao, 2020-04-10:
## ❓ Questions and Help

Hi, not sure whether this is a bug so I just open this issue as a question.

The main concern is about whether we need to include `--with-cuda` when install `habitat-sim` using `setup.py`? Actually, If I install without adding `--with-cuda`, I get `habitat_sim.cuda_enabled` as `Fasle`.

Here is some information which supports no need to include `--with-cuda`:
* `README.md` states the following situations where `--headless` should be used:
> For headless systems (i.e. without an attached display, e.g. in a cluster) and multiple GPU systems
* Discussion in issue #288: https://github.com/facebookresearch/habitat-sim/issues/288#issuecomment-540633510

However, from the code in `setup.py`, it seems that CUDA will only be compiled if we explicitly add a flag for that, [L258@setup.py](https://github.com/facebookresearch/habitat-sim/blob/ebb855c2fce29e48acd35c947812f45da32324d3/setup.py#L258).

Maybe we could make this more clear?

erikwijmans, 2020-04-10:
Hi, `--headless` and `--with-cuda` do different things and `--headless` does not automatically enable `--with-cuda`.  This is intended as compiling CUDA can be somewhat tricky (gcc and nvcc don't always play nice), so we don't want to make people compile that if they don't need it.

The discussion you link to is about inner details about how GPU selection is done.  Habitat-Sim will respect `CUDA_VISIBLE_DEVICES` when determining which physical GPU device to use (which is why the terminology CUDA device gets used) as setting `CUDA_VISIBLE_DEVICES` is a common way for resource managers to do allocation on shared systems.

I recently added the following note to the README:

Note1: Build flags stack, *e.g.* to build in headless mode, with CUDA, and bullet, one would use `--headless --with-cuda --bullet`.

Does that clarify things enough?

Xiaoming-Zhao, 2020-04-10:
Hi, I think it is really clear now. Thanks a lot

closed 2020-04-10 by Xiaoming-Zhao