facebookresearch/habitat-sim #586: Treating added objects as obstacles?

naruarjun, 2020-04-11:
## ❓ Questions and Help
When I run `./build/viewer data/habitat-test-scenes/skokloster-castle.glb --enable-physics` and add objects using `o`. The agent does not treat the added object as an obstacle and goes right through it. Is there a way such that the agent can stop when it comes in contact with the added objects? The object that I am adding to the scene is `nested_box.glb` and nested_box.physics_properties.json looks as follows:

```
{
    "render mesh": "nested_box.glb",
    "join collision meshes": true,
    "use bounding box for collision": true
}
```

aclegg3, 2020-04-11:
Hey @naruarjun, there are two ways to do this depending on whether or not you would like to run physics simulation and move the obstacle often.

1. You want a kinematic solution that does not require simulation and the obstacle will remain un-moving somewhere. In this case you can bake the obstacle into the NavMesh by setting its motion type to STATIC and recomputing the NavMesh:
```
object_id = sim.add_object(template_id)
sim.set_object_motion_type( habitat_sim.physics.MotionType::STATIC, object_id)
navmesh_settings = habitat_sim.NavMeshSettings()
navmesh_settings.set_defaults()
sim.recompute_navmesh(sim.pathfinder, navmesh_settings, include_static_objects=True)
```

2. You want the obstacle to block an embodied, fully simulated agent. You can attach the agent to a physical body, then control the agent with velocity or force commands. Note: the NavMesh will not be respected in this case.
```
#attach agent to simulated rigid object
agent_object_id = sim.add_object(body_object_template_id, attachment_node=agent_node)
obstacle_object_id = sim.add_object(obstacle_object_template_id)

#to control the agent with velocity_control, manipulate this struct
vel_control = sim.get_object_velocity_control(agent_object_id)

#or control the agent with force
sim.apply_force(...
sim.apply_torque(...
```

shivanshpatel35, 2020-04-12:
Hi @aclegg3 , I tried your first suggestion on the already provided banana object. When is use `include_static_objects=True` flag, I get segmentation fault. But it does not give any error if `include_static_objects=False`. What might be causing this?
Thanks in advance

aclegg3, 2020-04-13:
@shivanshpatel35 can you provide the code you are running and console output when crashing?

Tests are working with `"data/scene_datasets/habitat-test-scenes/van-gogh-room.glb"` and the banana object on my side.

shivanshpatel35, 2020-04-14:
@aclegg3 Hi, it was my fault. I missed enabling physics in api [here](https://github.com/facebookresearch/habitat-api/blob/master/habitat/config/default.py#L275). Now I am not getting any error.
I am inserting object in habitat-api using the following code snippet
```
object_id = env._sim._sim.add_object(0)
env._sim._sim.set_translation(object_position, object_id)
env._sim._sim.set_object_motion_type(habitat_sim.physics.MotionType.STATIC, object_id)
navmesh_settings = habitat_sim.NavMeshSettings()
navmesh_settings.set_defaults()
env._sim._sim.recompute_navmesh(env._sim._sim.pathfinder, navmesh_settings, include_static_objects=True)
```
After this when I try to check navigability at the `object_position` using `env._sim._sim.pathfinder.is_navigable(object_position)`, it returns `True`. I was expecting it to return `False`

naruarjun, 2020-04-14:
Hey @aclegg3, this has solved my issue. Thank you for the help.

erikwijmans, 2020-04-14:
> After this when I try to check navigability at the object_position using env._sim._sim.pathfinder.is_navigable(object_position), it returns True

Depending on the object and `object_position`, this may be expected as the top of the object can be navigable.  Any flat-ish surface will be navigable, however, the agent won't be able to climb up onto the object from the floor as long as the object is sufficiently tall.

shivanshpatel35, 2020-04-15:
> Depending on the object and `object_position`, this may be expected as the top of the object can be navigable. Any flat-ish surface will be navigable, however, the agent won't be able to climb up onto the object from the floor as long as the object is sufficiently tall.

That clears my doubt, thanks! We can close the issue now.

closed 2020-06-02 by aclegg3

nnsriram97, 2022-05-11:
Is there a way to treat included objects with `kinematic` motion type as obstacles for the agent. I added locobot object body to the agent ([from this comment](https://github.com/facebookresearch/habitat-sim/issues/586#issuecomment-612493562)) but it still goes right through it. I want to control both the object and the agent but wouldn't want them to collide with each other or other objects. Thanks!

aclegg3, 2022-05-11:
@nnsriram97 you may want to open a new issue if my answer here isn't sufficient since your question is somewhat tangential to the original question.

From what I understand:
1. You want to control the object kinematically by setting it's transform explicitly. 
2. You want to control the agent via the navmesh.
3. You want the agent to collide with the object via the navmesh.
4. You want to the object to collide with the agent and other objects (preventing you from setting it's state?)

Sounds like there are a few contradictions here. A kinematically controlled object has no collision response (it has infinite mass and can only be moved by manual state setting), so you would need to manually reject state updates that cause a collision. 

If you are doing that anyway, you may add a collision object to the agent (as discussed in the Habitat-Sim for Interaction tutorial from our [ECCV 2020 tutorial series](https://aihabitat.org/tutorial/2020/)) and then also reject state setting which causes the agent to collide with an object, the stage, or anything else. You could modify the collision filter function (also discussed in that tutorial) to do this.

Alternatively you could consider dynamically controlling the object and the agent to get collision responses without the navmesh (you could still use the navmesh to constrain target locations as in the tutorial).

If the object will not move often (e.g. you are placing a new piece of furniture) you could try a hybrid approach where you set the object to STATIC once it is placed and recompute the navmesh with that placement. You can follow the tutorial to see how you might place the object in the scene in a valid location.