facebookresearch/habitat-sim #587: Color of inserted objects

shivanshpatel35, 2020-04-11:
Hi, 
I followed [this](http://www.open3d.org/docs/release/tutorial/Basic/visualization.html#geometry-primitives) to generate basic geometric objects and saved them as .glb files. When I visualize those in an external viewer, they appear as expected. But when I insert them inside the simulator, they appear white. This can be verified [here](https://drive.google.com/open?id=1jhaWmB2ATJlKdE2W7PwqrfqO0vYm85rC). The link also contains the object's corresponding .gltf file so that it is ASCII readable. What might be causing this?
@aclegg3 
Thanks in advance.

aclegg3, 2020-04-15:
Hey @shivanshpatel35, can you also put the .glb file into this folder so I can test it?

shivanshpatel35, 2020-04-15:
@aclegg3 Done.

aclegg3, 2020-04-15:
@mosra Is this related to our recent importer changes?

mosra, 2020-04-15:
It is -- the importer is now checking glTF validity more thoroughly. The file seems to be exported by Open3D, I'm checking if this type combination is actually allowed by the spec, will get back to you once I know more.

mosra, 2020-04-15:
If you drop the file to https://github.khronos.org/glTF-Validator/, you'll see a bunch of validation errors, mainly due to floats being specified as normalized, which doesn't make sense. The errors are however not critical and could be worked around, but I don't feel great about adding such workarounds :) Where does the file come from? Should I open a bug report on the Open3D glTF exporter?

Besides that, the error handling on Habitat's side is not ideal -- it should be exiting more gracefully, not aborting on an assert. I'll see what can be done.

mosra, 2020-04-15:
Okay, so the bug most probably seems to be in to the revision of `tiny_gltf` that Open3D is using, and was fixed later in https://github.com/syoyo/tinygltf/commit/ae9364902a973cbb72a8ec15ea5e883a64fa4969 -- before that it seems that the `normalized` attribute is left at whatever random value the memory had before, and thus generating invalid files most of the time. If I'm able to reproduce locally without too much effort, I'll open a PR on the Open3D repo fixing this.

@shivanshpatel35 a manual workaround until you can get a fixed build of Open3D is to go through the exported glTF file and remove all lines that say `"normalized": true`. That should make it import properly.

dhruvbatra, 2020-04-15:
>  I'll open a PR on the Open3D repo fixing this.

That's a great idea, thanks @mosra. We know the folks at Open3D and should coordinate efforts.

mosra, 2020-04-15:
PR opened: https://github.com/intel-isl/Open3D/pull/1711

mosra, 2020-04-17:
The Open3D bugfix is now merged. @shivanshpatel35 if you're able to use Open3D master, everything should work properly again.

shivanshpatel35, 2020-04-17:
It works for me. Thanks a lot!

closed 2020-04-17 by shivanshpatel35