facebookresearch/habitat-sim #595: Compilation Error with libEGL.so

Xiaoming-Zhao, 2020-04-21:
## ❓ Questions and Help

Hi, I try to install `habitat-sim` on Ubuntu 18.04.3 LTS with `g++` 7.4.0. The command I used is:
```bash
python setup.py install --headless --with-cuda
```
I encountered an error which stated that
```bash
No rule to make target '/usr/lib/x86_64-linux-gnu/libEGL.so'
```
I have checked that I installed all the dependencies as described [here](https://github.com/facebookresearch/habitat-api/blob/master/.circleci/config.yml#L45) and there exists `/usr/lib/x86_64-linux-gnu/libEGL.so`.

Any guidance on how to solve this error? Thanks!

The following is the detailed log information:
```bash
running install
running bdist_egg
running egg_info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib.linux-x86_64-3.6
creating build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/physics.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/sensor.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/logging.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/registry.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/geo.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/sim.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/simulator.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/scene.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/errors.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/gfx.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim
creating build/lib.linux-x86_64-3.6/habitat_sim/sensors
copying habitat_sim/sensors/sensor_suite.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors
copying habitat_sim/sensors/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors
creating build/lib.linux-x86_64-3.6/habitat_sim/bindings
copying habitat_sim/bindings/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/bindings
creating build/lib.linux-x86_64-3.6/habitat_sim/utils
copying habitat_sim/utils/common.py -> build/lib.linux-x86_64-3.6/habitat_sim/utils
copying habitat_sim/utils/collect_env.py -> build/lib.linux-x86_64-3.6/habitat_sim/utils
copying habitat_sim/utils/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/utils
creating build/lib.linux-x86_64-3.6/habitat_sim/agent
copying habitat_sim/agent/agent.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent
copying habitat_sim/agent/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent
creating build/lib.linux-x86_64-3.6/habitat_sim/nav
copying habitat_sim/nav/greedy_geodesic_follower.py -> build/lib.linux-x86_64-3.6/habitat_sim/nav
copying habitat_sim/nav/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/nav
creating build/lib.linux-x86_64-3.6/habitat_sim/sensors/noise_models
copying habitat_sim/sensors/noise_models/speckle_noise_model.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors/noise_models
copying habitat_sim/sensors/noise_models/poisson_noise_model.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors/noise_models
copying habitat_sim/sensors/noise_models/salt_and_pepper_noise_model.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors/noise_models
copying habitat_sim/sensors/noise_models/gaussian_noise_model.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors/noise_models
copying habitat_sim/sensors/noise_models/redwood_depth_noise_model.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors/noise_models
copying habitat_sim/sensors/noise_models/no_noise_model.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors/noise_models
copying habitat_sim/sensors/noise_models/sensor_noise_model.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors/noise_models
copying habitat_sim/sensors/noise_models/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors/noise_models
creating build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/agent/controls/controls.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/agent/controls/object_controls.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
[... 891 lines cut ...]
    cmd_obj.run()
  File "/data/work/xz23/Python/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 173, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/data/work/xz23/Python/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 159, in call_command
    self.run_command(cmdname)
  File "/data/work/xz23/Python/anaconda3/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/data/work/xz23/Python/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/data/work/xz23/Python/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/data/work/xz23/Python/anaconda3/envs/habitat/lib/python3.6/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/data/work/xz23/Python/anaconda3/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/data/work/xz23/Python/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 202, in run
    self.build_extension(ext)
  File "setup.py", line 276, in build_extension
    shlex.split("cmake --build {}".format(self.build_temp)) + build_args
  File "/data/work/xz23/Python/anaconda3/envs/habitat/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--', '-j']' returned non-zero exit status 2.
```

Xiaoming-Zhao, 2020-04-21:
After some messy checks, it finally runs smoothly. I will list my solution here in case someone else encounters similar issues in the future.

## Check Existence of Shared Library

Please check the existence of following library files or files they are linked to: `libEGL.so`, `libOpenGL.so`, `libGL.so` etc.

In my case, `/usr/lib/x86_64-linux-gnu/libEGL.so` is a soft link and the file it links to is missing. After re-linking it to `/usr/lib/x86_64-linux-gnu/libEGL.so.1`, the compilation succeeds.

Please note, after installing some packages related to OpenGL, the links may be dead again. It is a good idea to have a second check.

## For `import` error

When trying to `import habitat_sim`, I encountered the following error:
```bash
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/data/work/xz23/Python/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.1.4-py3.6-linux-x86_64.egg/habitat_sim/__init__.py", line 14, in <module>
    from habitat_sim.nav import *
  File "/data/work/xz23/Python/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.1.4-py3.6-linux-x86_64.egg/habitat_sim/nav/__init__.py", line 1, in <module>
    from habitat_sim._ext.habitat_sim_bindings import (
ImportError: /usr/lib/x86_64-linux-gnu/libOpenGL.so.0: undefined symbol: _glapi_tls_Current
```
I used the following commands to locate the dependencies I need to install:
```bash
# find symbols from object files
> nm -D /usr/lib/x86_64-linux-gnu/libOpenGL.so.0 | grep tls
U _glapi_tls_Current

# get shared libraries dependencies
> ldd /usr/lib/x86_64-linux-gnu/libOpenGL.so.0
linux-vdso.so.1 (0x00007ffc68b2e000)
libGLdispatch.so.0 (0x00007fafef33f000)
libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fafeef4e000)
libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fafeed4a000)
/lib64/ld-linux-x86-64.so.2 (0x00007fafef823000)
```
Then I found `libGLdispatch.so.0` was another dead link. I need to install the pacakge `libglvnd0` to get `libGLdispatch.so.0.0.0`.
```bash
> nm -D /usr/lib/x86_64-linux-gnu/libOpenGL.so.0.0.0 | grep tls
U _glapi_tls_Current
```

closed 2020-04-21 by Xiaoming-Zhao

KenaHemnani, 2020-12-12:
@Xiaoming-Zhao I am getting exactly same error:
 **from _magnum import *
ImportError: /lib/x86_64-linux-gnu/libOpenGL.so.0: undefined symbol: _glapi_tls_Current**

Can you please explain in detail how to resolve it?
I didn't understood the above description.

msalvaris, 2020-12-20:
The solution was to install some dependencies I found in the docker file https://github.com/facebookresearch/habitat-lab/blob/master/Dockerfile
Mainly
libjpeg-dev \
    libpng-dev \
    libglfw3-dev \
    libglm-dev \
    libx11-dev \
    libomp-dev \
    libegl1-mesa-dev \