facebookresearch/habitat-sim #606: Build issue

AdventureO, 2020-04-28:
**Issue**
CMake errors out during a subprocess call while building using setupy.py. 
```subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--', '-j']' returned non-zero exit status 2```

**Steps performed**
1. ```git clone --branch stable https://github.com/facebookresearch/habitat-sim.git```
2. ```cd habitat-sim```
3. Install all dependencies
4. ```python setup.py install --headless```

I got the following full error [error.txt](https://github.com/facebookresearch/habitat-sim/files/4548248/error.txt)

**Additional context**
I have been able to build this before. Got an error after pulling commits for last 3 weeks. Than clone repository again and make an installation from zero, got the error shown above. Please help!

erikwijmans, 2020-04-28:
Looks like you are using gcc 4.8.  This isn't supported, please update your gcc install.

closed 2020-04-29 by AdventureO

amarthyasasi, 2021-05-27:
Hi @erikwijmans, I am faceing the same issue but my gcc version is 7.5.0. I am trying to install habitat in headless mode in a remote server without root access using conda. The entire error message is the same. Is there any other way to tackle this error?

@AdventureO  Can you please confirm if the problem was resolved by updating gcc or you had to do something else?

erikwijmans, 2021-05-27:
Please open a new issue.  Your issue is due to 

```
No rule to make target '/usr/lib/x86_64-linux-gnu/libEGL.so', 
needed by 'deps/magnum-bindings/src/python/_magnum.cpython-36m-x86_64-linux-gnu.so'. Stop.
```