facebookresearch/habitat-sim #649: web browser failed to retrieve Request URL: https://habitat-resources.s3.amazonaws.com/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb

rabbit721, 2020-06-12:
## ❓ Questions and Help
Hi, I followed the below steps, and inspect the webpage and see that it failed to retrieve the data from s3. 

Request URL: https://habitat-resources.s3.amazonaws.com/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb

I downloaded and extracted the test data locally, so I have /habitat-sim/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb. But I don't know how to force binding_js to look for the local file instead of getting it from s3. 

Thanks!

dhruvbatra, 2020-06-12:
CC: @apsdehal @msbaines

apsdehal, 2020-06-12:
@rabbit721 Hi,

If you running it locally, please use localhost address so that configuration can adjust accordingly. See https://github.com/facebookresearch/habitat-sim/blob/master/src/esp/bindings_js/modules/defaults.js#L33.

rabbit721, 2020-06-12:
I am running it inside google colab. So I used colab's kernel's own proxy port instead of localhost. Is there another way to specify using the local test data?

apsdehal, 2020-06-12:
IIRC, you should be able to specify `scene=<path>` directly in your url.

rabbit721, 2020-06-13:
It works, thanks a lot!

closed 2020-06-13 by rabbit721