facebookresearch/habitat-sim #651: How to load object config in the old version?

lailvlong, 2020-06-14:
Hello! In the ["lighting setup tutorial"](https://aihabitat.org/docs/habitat-sim/lighting-setups.html), it use load_object_config() and add_object() to add an sphere to the scene. However, in the version of "git checkout 9575dcd45fe6f55d2a44043833af08972a7895a9", the Simulator doesn't have the load_object_config() function. Is there an alternative?

erikwijmans, 2020-06-14:
No, that tutorial will never work with that version of the simulator (that version is before we supported lighting :))

lailvlong, 2020-06-15:
> No, that tutorial will never work with that version of the simulator (that version is before we supported lighting :))

Thank you for you replying! I have another question. After i add some new objects to the scene, is there any function used to save the scene as glb file?

erikwijmans, 2020-06-15:
No, that is not something we currently support.

closed 2020-08-09 by dhruvbatra