facebookresearch/habitat-sim #652: Build issue on OSX Catalina (10.15)

guydav, 2020-06-15:
## 🐛 Bug

The build fails, regardless of whether I try building with a dedicated (and fresh) condo environment or in my base environment. It fails with two different error messages.

## Command
```
git clone --branch stable https://github.com/facebookresearch/habitat-sim.git
cd habitat-sim
conda create -n habitat python=3.6 cmake=3.14.0
conda activate habitat
pip install -r requirements.txt
python setup.py install
```
It fails both with and without the `--bullet` flag. 

## To Reproduce

The errors it fails with are different when I'm in my base environment and when I'm in the newly created habitat environment. That might be related to which gcc is used? My base environment uses gcc 5.20 installed through conda, while my dedicated habitat environment uses OSX's gcc (/usr/bin/gcc). As the errors are different, I've attached the full error printouts for both.

[error_base.txt](https://github.com/facebookresearch/habitat-sim/files/4780984/error_base.txt)

[error_habitat.txt](https://github.com/facebookresearch/habitat-sim/files/4780988/error_habitat.txt)

## Expected behavior

Habitat-sim installing successfully.

## Additional context

OSX 10.15.3 (Catalina). 

## System Info

[base_env.txt](https://github.com/facebookresearch/habitat-sim/files/4781005/base_env.txt)

[habitat_env.txt](https://github.com/facebookresearch/habitat-sim/files/4781011/habitat_env.txt)

erikwijmans, 2020-06-15:
In the habitat env, can you try adding `--cmake` to the command?  Looks like cmake needs to be re-run

guydav, 2020-06-16:
That doesn't appear to make much of a difference. It builds more things but fails similarly: 

```
(habitat) guydavidson@Guys-MBP-2 habitat-sim (detached*)*$ conda env list
# conda environments:
#
base                     /Users/guydavidson/opt/anaconda3
ds-ga-1004               /Users/guydavidson/opt/anaconda3/envs/ds-ga-1004
habitat               *  /Users/guydavidson/opt/anaconda3/envs/habitat
minecraft_env            /Users/guydavidson/opt/anaconda3/envs/minecraft_env
```

Then I ran `(habitat) guydavidson@Guys-MBP-2 habitat-sim (detached*)*$ python setup.py install --cmake  &> error_habitat_cmake.txt`. The output ends on a similar failure with a ninja flag:

[error_habitat_cmake.txt](https://github.com/facebookresearch/habitat-sim/files/4785760/error_habitat_cmake.txt)

jturner65, 2020-06-16:
I know we suggest gcc to be 7.4 on ubuntu, not sure if that is an issue here with a mac, at least with the base environment.

erikwijmans, 2020-06-16:
For MacOS, just using clang should be fine (no need to gcc).  This seems to be an issue with ninja (ninja is what is erroring out).  Can you run `conda install ninja` and try again?  I think your base env has ninja in it, but the habitat one doesn't, so cmake picked up ninja from the base, but now setup.py is trying to setup args for make since it can't find ninja.

guydav, 2020-06-16:
It seems that at some point conda started using its own toolchain, or at least, something I did in the base environment did, as the new habitat conda env uses clang. I'm not sure how to point the base env to use OSX's compiler again.
```
(base) guydavidson@Guys-MBP-2 habitat-sim (detached*)*$ which gcc
/Users/guydavidson/opt/anaconda3/bin/gcc
(base) guydavidson@Guys-MBP-2 habitat-sim (detached*)*$ conda activate habitat
(habitat) guydavidson@Guys-MBP-2 habitat-sim (detached*)*$ which gcc
/usr/bin/gcc
```
However, installing ninja in the habitat env didn't help:
```
(habitat) guydavidson@Guys-MBP-2 habitat-sim (detached*)*$ which ninja
/Users/guydavidson/opt/anaconda3/envs/habitat/bin/ninja
(habitat) guydavidson@Guys-MBP-2 habitat-sim (detached*)*$ python setup.py install &> error_habitat_ninja.txt
```
[error_habitat_ninja.txt](https://github.com/facebookresearch/habitat-sim/files/4787646/error_habitat_ninja.txt)

erikwijmans, 2020-06-16:
Can you do `rm -rf build` and then re-run in the habitat env?  Looks like it's still picking up up the old gcc.

guydav, 2020-06-16:
Just did. Still fails, but appears to get a little later? I also verified that at least in the shell, gcc and clang point to the OS versions:
```
(habitat) guydavidson@Guys-MBP-2 habitat-sim (detached*)*$ which gcc
/usr/bin/gcc
(habitat) guydavidson@Guys-MBP-2 habitat-sim (detached*)*$ which clang
/usr/bin/clang
```
[error_post_rm_build.txt](https://github.com/facebookresearch/habitat-sim/files/4787717/error_post_rm_build.txt)

Am I doing something wrong? Thank you for your patience!

erikwijmans, 2020-06-16:
```
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:320:9: error: no member named 'isinf' in the global namespace
using ::isinf;
```

That is some strange issue that pops up with macOS build tools sometimes.  There isn't a one size fits all solution to that, if you google "error: no member named 'isinf' in the global namespace" there will be suggestions of things to try.  The one that has worked for me in the past is `sudo xcode-select --reset`, but I am not sure if that will have unintended side effects on other things.

guydav, 2020-06-17:
I think I figured it out. At some point while I was trying to compile with conda's gcc in the base env, it couldn't find some of the library headers, so I added the following line to my .bashrc:

```
export CPATH="$CPATH:/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include"
```

Once I removed it and opened a new shell, everything compiled in my habitat env just fine. Trying to run the habitat-api examples now.

closed 2020-07-08 by Skylion007