facebookresearch/habitat-sim #657: cmake fail on colab

AveenD, 2020-06-18:
## 🐛 Bug
cmake fails at 49%. The error is attached below

## Command

!python setup.py install --headless

output:
running install
running bdist_egg
running egg_info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
[ -2%] Built target ShaderResources-dependencies
[ -2%] Built target AnyImageImporter-dependencies
[ -2%] Built target AssimpImporter-dependencies
[ -2%] Built target StbImageImporter-dependencies
[ -2%] Built target PrimitiveImporter-dependencies
[ -2%] Built target MagnumImGuiIntegrationShaders_RCS-dependencies
[ -2%] Built target AnySceneImporter-dependencies
[ -2%] Built target AnyImageConverter-dependencies
[ -2%] Built target StanfordImporter-dependencies
[ -2%] Built target TinyGltfImporter-dependencies
[ -2%] Built target MagnumShaders_RCS-dependencies
[ -2%] Built target StbImageConverter-dependencies
[ -2%] Built target BasisImporter-dependencies
[ -2%] Built target IrrXML
[ -2%] Built target MagnumFlextGLObjects
[ -1%] Built target Detour
[  0%] Built target MagnumSceneGraphObjects
[  1%] Built target Recast
[  1%] Built target CorradePluginManagerObjects
[  2%] Built target glog
[  3%] Built target MagnumEglContextObjects
[  3%] Built target MagnumObjects
[  4%] Built target MagnumMeshToolsObjects
[  6%] Built target CorradeUtilityObjects
[  7%] Built target corrade-rc
[  8%] Built target MagnumMathObjects
[  9%] Built target MagnumTradeObjects
[  9%] Linking CXX shared library ../../../../../lib.linux-x86_64-3.6/habitat_sim/_ext/libCorradeUtility.so
[ 10%] Built target MagnumShadersObjects
[ 11%] Built target MagnumDebugToolsObjects
[ 17%] Built target MagnumGLObjects
[ 49%] Built target assimp
CMake Error: cmake_symlink_library: System Error: Operation not supported
CMake Error: cmake_symlink_library: System Error: Operation not supported
deps/corrade/src/Corrade/Utility/CMakeFiles/CorradeUtility.dir/build.make:193: recipe for target 'lib.linux-x86_64-3.6/habitat_sim/_ext/libCorradeUtility.so.2.3' failed
make[2]: *** [lib.linux-x86_64-3.6/habitat_sim/_ext/libCorradeUtility.so.2.3] Error 1
make[2]: *** Deleting file 'lib.linux-x86_64-3.6/habitat_sim/_ext/libCorradeUtility.so.2.3'
CMakeFiles/Makefile2:500: recipe for target 'deps/corrade/src/Corrade/Utility/CMakeFiles/CorradeUtility.dir/all' failed
make[1]: *** [deps/corrade/src/Corrade/Utility/CMakeFiles/CorradeUtility.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
Traceback (most recent call last):
  File "setup.py", line 382, in <module>
    include_package_data=True,
  File "/usr/local/lib/python3.6/site-packages/setuptools/__init__.py", line 129, in setup
    return distutils.core.setup(**attrs)
  File "/usr/local/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/local/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/local/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.6/site-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/usr/local/lib/python3.6/site-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/usr/local/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/local/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 172, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/usr/local/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 158, in call_command
    self.run_command(cmdname)
  File "/usr/local/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/local/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.6/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/usr/local/lib/python3.6/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/usr/local/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/local/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 202, in run
    self.build_extension(ext)
  File "setup.py", line 278, in build_extension
    shlex.split('cmake --build "{}"'.format(self.build_temp)) + build_args
  File "/usr/local/lib/python3.6/subprocess.py", line 291, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--', '-j']' returned non-zero exit status 2.

BryanBo-Cao, 2020-07-01:
Try this?
```
python setup.py build_ext --parallel 1 install --headless
```
https://github.com/facebookresearch/habitat-sim/issues/498#issuecomment-589737511

#476 #498

Skylion007, 2020-07-08:
if you still have the issue, let us know as we do have CMake builds working fine in Colab. We will also be releasing official notebooks soon as part of our ECCV tutorial.

closed 2020-07-08 by Skylion007