facebookresearch/habitat-sim #658: Undefined reference to assimp when building in Ubuntu 20.04

sotpapathe, 2020-06-24:
## 🐛 Bug

Compiling on Ubuntu 20.04 results in a linker error about assimp, although the
`libassimp-dev` package is installed (error log below).

## Command
`python setup.py install` while in the correct conda environment.

## To Reproduce

Steps to reproduce the behavior:

Just followed the README instructions, creating a conda environment, installing
dependencies and compiling.

Output from running `python setup.py install` for the 3rd time to hide any successful compilation steps

```
running install
running bdist_egg
running egg_info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
[  1%] Built target CorradePluginManagerObjects
[  3%] Built target CorradeUtilityObjects
[  4%] Built target glog
[  5%] Built target IrrXML
[  5%] Built target corrade-rc
[  7%] Built target Detour
[  8%] Built target Recast
[  9%] Built target glfw
[  9%] Built target MagnumFlextGLObjects
[ 10%] Built target MagnumMathObjects
[ 10%] Built target MagnumGlxContextObjects
[ 10%] Built target MagnumPlatformObjects
[ 11%] Built target MagnumObjects
[ 12%] Built target MagnumTradeObjects
[ 13%] Built target MagnumDebugToolsObjects
[ 13%] Built target AnyImageImporter-dependencies
[ 13%] Built target MagnumSceneGraphObjects
[ 13%] Built target MagnumShaders_RCS-dependencies
[ 13%] Built target MagnumMeshToolsObjects
[ 13%] Built target AnyImageConverter-dependencies
[ 13%] Built target AnySceneImporter-dependencies
[ 13%] Built target AssimpImporter-dependencies
[ 13%] Built target StanfordImporter-dependencies
[ 13%] Built target StbImageConverter-dependencies
[ 13%] Built target BasisImporter-dependencies
[ 13%] Built target PrimitiveImporter-dependencies
[ 13%] Built target StbImageImporter-dependencies
[ 13%] Built target TinyGltfImporter-dependencies
[ 17%] Built target MagnumGLObjects
[ 17%] Built target MagnumImGuiIntegrationShaders_RCS-dependencies
[ 17%] Built target ShaderResources-dependencies
[ 18%] Built target CorradeUtility
[ 19%] Built target simple
[ 19%] Built target particles
[ 19%] Built target offscreen
[ 19%] Built target heightmap
[ 20%] Built target boing
[ 20%] Built target wave
[... 135 lines cut ...]
Machine: x86_64
Processor: x86_64
Libc version: glibc 2.31
Python version: 3.8.2
Architecture: 64bit ELF
System OS: Linux
Release: 5.4.0-37-generic
Version: #41-Ubuntu SMP Wed Jun 3 18:57:02 UTC 2020
Operational System: linux
GCC version: gcc (Ubuntu 9.3.0-10ubuntu2) 9.3.0
CMAKE version: cmake version 3.14.0
NVIDIA-SMI: Wed Jun 24 13:19:56 2020
```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.64       Driver Version: 440.64       CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  Quadro P1000        Off  | 00000000:01:00.0  On |                  N/A |
| N/A   49C    P8    N/A /  N/A |    255MiB /  4032MiB |      6%      Default |
+-------------------------------+----------------------+----------------------+
```
Pip packages versions: pip 20.1.1
Conda packages versions: conda 4.8.3

erikwijmans, 2020-06-24:
Seems plausible that system assimp and our bundled assimp aren't playing nice.  If you recompile with `--cmake-args="-DUSE_SYSTEM_ASSIMP=ON"` does that work?

sotpapathe, 2020-06-25:
That worked, thanks!

For future reference I ran:
1. `rm -rf build` to remove anything built previously
2. `python setup.py install --cmake-args="-DUSE_SYSTEM_ASSIMP=ON"`

closed 2020-06-25 by sotpapathe