facebookresearch/habitat-sim #660: Build failed - Could NOT find OpenGL during CMake

BryanBo-Cao, 2020-06-25:
## 🐛 Bug

I tried to install ```habitat-sim``` on a new machine, but the system build failed. Thus, the code could not import ```habitat_sim```.

## Command
```
python setup.py build_ext --parallel 2 install --headless
```

## To Reproduce

Steps to reproduce the behavior:

1. git clone --branch stable https://github.com/facebookresearch/habitat-sim.git
2. cd habitat-sim
3. conda create -n habitat python=3.6 cmake=3.14.0
4. \# conda activate habitat # Note that this is commented out
5. pip install -r requirements.txt
6. sudo apt-get update || true
7. sudo apt-get install -y --no-install-recommends \\
     libjpeg-dev libglm-dev libgl1-mesa-glx libegl1-mesa-dev mesa-utils xorg-dev freeglut3-dev
8. python setup.py build_ext --parallel 2 install --headless

Error message:
```
(base) anonymity@white:~/Repos/habitat-sim$ python setup.py build_ext --parallel 2 install --headless
running build_ext
Submodule 'docs/m.css' (https://github.com/mosra/m.css) registered for path 'docs/m.css'
Submodule 'src/deps/Sophus' (https://github.com/strasdat/Sophus) registered for path 'src/deps/Sophus'
Submodule 'src/deps/assimp' (https://github.com/assimp/assimp) registered for path 'src/deps/assimp'
Submodule 'src/deps/corrade' (https://github.com/mosra/corrade.git) registered for path 'src/deps/corrade'
Submodule 'src/deps/eigen-git-mirror' (https://github.com/eigenteam/eigen-git-mirror) registered for path 'src/deps/eigen-git-mirror'
Submodule 'src/deps/glfw' (https://github.com/glfw/glfw) registered for path 'src/deps/glfw'
Submodule 'src/deps/glog' (https://github.com/google/glog.git) registered for path 'src/deps/glog'
Submodule 'src/deps/googletest' (https://github.com/google/googletest) registered for path 'src/deps/googletest'
Submodule 'src/deps/imgui' (https://github.com/ocornut/imgui.git) registered for path 'src/deps/imgui'
Submodule 'src/deps/magnum' (https://github.com/mosra/magnum.git) registered for path 'src/deps/magnum'
Submodule 'src/deps/magnum-bindings' (https://github.com/mosra/magnum-bindings.git) registered for path 'src/deps/magnum-bindings'
Submodule 'src/deps/magnum-integration' (https://github.com/mosra/magnum-integration.git) registered for path 'src/deps/magnum-integration'
Submodule 'src/deps/magnum-plugins' (https://github.com/mosra/magnum-plugins.git) registered for path 'src/deps/magnum-plugins'
Submodule 'src/deps/pybind11' (https://github.com/pybind/pybind11.git) registered for path 'src/deps/pybind11'
Submodule 'src/deps/rapidjson' (https://github.com/Tencent/rapidjson.git) registered for path 'src/deps/rapidjson'
Submodule 'src/deps/recastnavigation' (https://github.com/erikwijmans/recastnavigation.git) registered for path 'src/deps/recastnavigation'
Submodule 'src/deps/tinyobjloader' (https://github.com/syoyo/tinyobjloader.git) registered for path 'src/deps/tinyobjloader'
Cloning into '/home/anonymity/Repos/habitat-sim/docs/m.css'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/Sophus'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/assimp'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/corrade'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/eigen-git-mirror'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/glfw'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/glog'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/googletest'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/imgui'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/magnum'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/magnum-bindings'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/magnum-integration'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/magnum-plugins'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/pybind11'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/rapidjson'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/recastnavigation'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/tinyobjloader'...
Submodule path 'docs/m.css': checked out '01ea5ad52eb8e85b018027a5d540f5822ebfebba'
Submodule path 'src/deps/Sophus': checked out 'b475c0a81a197ff8ea6bdcf8df8a2f5a7b1da879'
Submodule path 'src/deps/assimp': checked out '8f0c6b04b2257a520aaab38421b2e090204b69df'
Submodule path 'src/deps/corrade': checked out '2eac8ab3f1192d92641e0c9f78a08ebb501f32a4'
Submodule 'toolchains' (https://github.com/mosra/toolchains.git) registered for path 'src/deps/corrade/toolchains'
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/corrade/toolchains'...
Submodule path 'src/deps/corrade/toolchains': checked out 'f694eee91faf46cc185616ae3668cdba00a3e028'
Submodule path 'src/deps/eigen-git-mirror': checked out 'd10e1f03d4ef456f78ff6b91d9078b1f13a5975c'
Submodule path 'src/deps/glfw': checked out '0a49ef0a00baa3ab520ddc452f0e3b1e099c5589'
[... 294 lines cut ...]
numba==0.50.0
numpy==1.19.0
numpy-quaternion==2020.5.19.15.27.24
tqdm==4.46.1
Conda packages versions:
attrs                     19.3.0                     py_0    19.1.0
gitpython                 3.1.3                    pypi_0    pypi
matplotlib                3.1.3                    py37_0  
numba                     0.48.0           py37h0573a6f_0  
numpy                     1.18.1           py37h4f9e942_0  
numpy-quaternion          2020.5.19.15.27.24          pypi_0    pypi
pillow                    7.0.0            py37hb39fc2d_0  
scipy                     1.4.1            py37h0b6359f_0    1.3.0
tqdm                      4.42.1                     py_0
```
GPU: NVIDIA GeForce RTX 2080 Super
| OS/Driver/Lib  | Version |
| ------------- | ------------- |
| Ubuntu  | 18.04.4 LTS  |
| GPU Driver  | 450.36.06  |
| CUDA | 11.0 |
| Tensorflow | 2.2.0 |
| Keras | 2.3.1 |

Any help would be appreciated!

BryanBo-Cao, 2020-06-25:
I fixed it by hard resetting it back to a2b5522fb7a6eb20c1c87398cbfcf873bbb0309d and redo the steps in ```Steps to reproduce the behavior``` above.

BryanBo-Cao, 2020-06-25:
### On the machine named ```alienware```

I did the same steps in the ```Steps to reproduce the behavior``` section above,

And got
```
(base) anonymity@alienware:~/Repos/habitat-sim$ python setup.py build_ext --parallel 2 install --headless
running build_ext
Submodule 'docs/m.css' (https://github.com/mosra/m.css) registered for path 'docs/m.css'
Submodule 'src/deps/Sophus' (https://github.com/strasdat/Sophus) registered for path 'src/deps/Sophus'
Submodule 'src/deps/assimp' (https://github.com/assimp/assimp) registered for path 'src/deps/assimp'
Submodule 'src/deps/corrade' (https://github.com/mosra/corrade.git) registered for path 'src/deps/corrade'
Submodule 'src/deps/eigen-git-mirror' (https://github.com/eigenteam/eigen-git-mirror) registered for path 'src/deps/eigen-git-mirror'
Submodule 'src/deps/glfw' (https://github.com/glfw/glfw) registered for path 'src/deps/glfw'
Submodule 'src/deps/glog' (https://github.com/google/glog.git) registered for path 'src/deps/glog'
Submodule 'src/deps/googletest' (https://github.com/google/googletest) registered for path 'src/deps/googletest'
Submodule 'src/deps/imgui' (https://github.com/ocornut/imgui.git) registered for path 'src/deps/imgui'
Submodule 'src/deps/magnum' (https://github.com/mosra/magnum.git) registered for path 'src/deps/magnum'
Submodule 'src/deps/magnum-bindings' (https://github.com/mosra/magnum-bindings.git) registered for path 'src/deps/magnum-bindings'
Submodule 'src/deps/magnum-integration' (https://github.com/mosra/magnum-integration.git) registered for path 'src/deps/magnum-integration'
Submodule 'src/deps/magnum-plugins' (https://github.com/mosra/magnum-plugins.git) registered for path 'src/deps/magnum-plugins'
Submodule 'src/deps/pybind11' (https://github.com/pybind/pybind11.git) registered for path 'src/deps/pybind11'
Submodule 'src/deps/rapidjson' (https://github.com/Tencent/rapidjson.git) registered for path 'src/deps/rapidjson'
Submodule 'src/deps/recastnavigation' (https://github.com/erikwijmans/recastnavigation.git) registered for path 'src/deps/recastnavigation'
Submodule 'src/deps/tinyobjloader' (https://github.com/syoyo/tinyobjloader.git) registered for path 'src/deps/tinyobjloader'
Cloning into '/home/anonymity/Repos/habitat-sim/docs/m.css'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/Sophus'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/assimp'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/corrade'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/eigen-git-mirror'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/glfw'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/glog'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/googletest'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/imgui'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/magnum'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/magnum-bindings'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/magnum-integration'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/magnum-plugins'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/pybind11'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/rapidjson'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/recastnavigation'...
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/tinyobjloader'...
Submodule path 'docs/m.css': checked out '01ea5ad52eb8e85b018027a5d540f5822ebfebba'
Submodule path 'src/deps/Sophus': checked out 'b475c0a81a197ff8ea6bdcf8df8a2f5a7b1da879'
Submodule path 'src/deps/assimp': checked out '8f0c6b04b2257a520aaab38421b2e090204b69df'
Submodule path 'src/deps/corrade': checked out '2eac8ab3f1192d92641e0c9f78a08ebb501f32a4'
Submodule 'toolchains' (https://github.com/mosra/toolchains.git) registered for path 'src/deps/corrade/toolchains'
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/corrade/toolchains'...
Submodule path 'src/deps/corrade/toolchains': checked out 'f694eee91faf46cc185616ae3668cdba00a3e028'
Submodule path 'src/deps/eigen-git-mirror': checked out 'd10e1f03d4ef456f78ff6b91d9078b1f13a5975c'
Submodule path 'src/deps/glfw': checked out '0a49ef0a00baa3ab520ddc452f0e3b1e099c5589'
Submodule path 'src/deps/glog': checked out '8d7a107d68c127f3f494bb7807b796c8c5a97a82'
Submodule path 'src/deps/googletest': checked out '00938b2b228f3b70d3d9e51f29a1505bdad43f1e'
Submodule path 'src/deps/imgui': checked out 'ca30ef4269150e71a950106c6cfa8f94898581da'
Submodule path 'src/deps/magnum': checked out '902e8053c7871853f20a8c1592095c94d7e870de'
Submodule 'toolchains' (https://github.com/mosra/toolchains.git) registered for path 'src/deps/magnum/toolchains'
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/magnum/toolchains'...
Submodule path 'src/deps/magnum/toolchains': checked out 'f694eee91faf46cc185616ae3668cdba00a3e028'
Submodule path 'src/deps/magnum-bindings': checked out 'a6114550343cd0a35fba9b1f0df23ba5fb244831'
Submodule path 'src/deps/magnum-integration': checked out 'cc90f56ab508bf26c35a377cb03de2f180085058'
Submodule 'toolchains' (https://github.com/mosra/toolchains.git) registered for path 'src/deps/magnum-integration/toolchains'
Cloning into '/home/anonymity/Repos/habitat-sim/src/deps/magnum-integration/toolchains'...
Submodule path 'src/deps/magnum-integration/toolchains': checked out '23cd3a402cb3004a35c1248f4e1c185c49d097da'
[... cut ...]
Unpacking libminizip1:amd64 (1.1-8build1) ...
Selecting previously unselected package minizip.
Preparing to unpack .../minizip_1.1-8build1_amd64.deb ...
Unpacking minizip (1.1-8build1) ...
Setting up libminizip1:amd64 (1.1-8build1) ...
Setting up minizip (1.1-8build1) ...
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
Processing triggers for libc-bin (2.27-3ubuntu1) ...
```
And then recloned the repo and tried to reinstall it by
```
python setup.py build_ext --parallel 2 install --headless
```
But still got the same error.

erikwijmans, 2020-06-25:
Looks like OpenGL is not being found, can you do `sudo apt install libgl1-mesa-dev`?

BryanBo-Cao, 2020-06-25:
I tried that,we can search ```libgl1-mesa-dev``` to find the log ithis page.

erikwijmans, 2020-06-25:
When it build successfully, where does it say it found OpenGL?  The line of code that is failing is older than habitat-sim itself (magnum finds OpenGL for us), so there isn't an obvious reason why the newer commit would change anything.

BryanBo-Cao, 2020-07-01:
I downloaded and rebuilt it on a fresh new ```Ubuntu 18.04``` successfully. The latest commit was 020041d75eaf3c70378a9ed0774b5c67b9d3ce99 tagged ```stable``` on May 27, 2020. It 'Built target MagnumGL` based on the installation log. 

My code seems to be working fine now.

closed 2020-07-01 by BryanBo-Cao