facebookresearch/habitat-sim #662: How to recompute glb files used by the simulator for Matterport dataset?

free-bit, 2020-06-26:
## ❓ Questions and Help
Hello,

I am searching for a way to recompute the files used by habitat-sim for the Matterport dataset. My observation is except for the house file, all files are generated by the framework from the original dataset somehow to perform the simulation. I think the semantic ply file of each scene is generated from the original ply file provided by Matterport dataset in house segmentations by using the create_mp3d_semantic_mesh option of datatool script. Once the glb file is available, navmesh can be either computed on the fly or precomputed by using the create_navmesh option of datatool script. I also looked for a way to reproduce glb files created for Matterport (https://github.com/facebookresearch/habitat-api/#scenes-datasets) but I couldn't find it. At first, I thought it might be possible by using assimp's export function to convert the ply file generated for segmentation to glb but the glb file that I generated in this way doesn't match with the one provided. The file generated in this way was significantly larger than the original glb file. I also converted one of the available glb files to ply with assimp library and I realized that number of meshes and vertices in this ply file (converted from the provided glb) are significantly lower than in the respective segmentation ply file. My question is, how exactly do you compute glb files? Which tool(s) are used, or which part of the framework is responsible for that and which files from original Matterport dataset are used for this? I searched for an answer in the source code and issues but I couldn't find anything.

Thank you for your time.

erikwijmans, 2020-06-26:
The glbs come from the matterport_mesh, which is in obj+mtl format in the mp3d dataset. The original version of the glbs are just from converting obj+mtl to glb (IIRC via assimp).  The more recent version have basis compressed textures, that was done with a custom script using utilities in Magnum, this step isn't needed to get habitat to load them however.

free-bit, 2020-06-27:
Thanks for the answer. At first, I tried to reproduce the glb file by converting obj+mtl provided in matterport_mesh via assimp export tool with -fglb2 -embtex flags. However, the generated glb file in this way doesn't exactly match with the provided one. The one that I generated with assimp (v5.0) has a smaller size. I also opened it with viewer tool to see if it is visually correct but realized that texture doesn't get mapped to the meshes correctly. Might there be an extra processing step related to coordinates/axes before the conversion? As far as I remember, habitat-sim internally uses a different convention for axes. I thought it might be related to that.

dhruvbatra, 2020-06-27:
CC: @msavva

msavva, 2020-06-28:
With assimp (on MacOS homebrew assimp stable 5.0.1) I experience a similar issue where output glb files have garbled textures.  This can be confirmed by viewing with other glTF viewers as well, such as https://magnum.graphics/showcase/player/ and https://gltf-viewer.donmccurdy.com/ .

Instead, using https://github.com/CesiumGS/obj2gltf and ` obj2gltf -i test.obj -o test.glb -b` everything looks fine in Habitat and in other viewers.

I recall using assimp to do the initial conversion a while back, so perhaps this is caused by a regression / change in behavior in assimp ~5.0 ...  Hopefully the above alternative conversion pipeline suffices for your use case.

free-bit, 2020-06-28:
With obj2gltf, I was able to reproduce the glb file without any problem. Thank you all for your help.

closed 2020-06-28 by free-bit

zxwang96, 2021-04-03:
> With obj2gltf, I was able to reproduce the glb file without any problem. Thank you all for your help.

From the first reply I found the glb comes from combining obj+mtl. I wonder does obj2gltf only work with obj or obj+mtl?