facebookresearch/habitat-sim #664: Using habitat in Pytorch multiprocessing thread.

pushkalkatara, 2020-06-27:


erikwijmans, 2020-06-27:
As said in the issue you linked, you need to use `forkserver` or `spawn`, there isn't a way around this as its a core limitation of EGL.  PyTorch CUDA memory sharing also requires (https://pytorch.org/docs/stable/multiprocessing.html#sharing-cuda-tensors).   Where is the `Context has been already set` error coming from?  I don't recall where the is in our code.

pushkalkatara, 2020-06-29:
I traced the error in the starting of the script while importing torch and setting the start method. Adding a flag `force=True` to the `torch.multiprocessing.set_start_method('forkserver')` solved the `Context has already been set` error. It was more of a Pytorch issue rather than Habitat, thanks for the help.

closed 2020-06-29 by pushkalkatara