facebookresearch/habitat-sim #665: [iccv2019-hacky-wip branch] Error in setting up WebGL Habitat-Sim

Ram81, 2020-06-29:
## 🐛 Bug

I followed the instruction on https://github.com/facebookresearch/habitat-sim#webgl to run the webgl demo from `iccv2019-hacky-wip` branch. However, in step 4., when I run: ./build_js.sh , I get the following error:

```
Scanning dependencies of target CorradeUtility
[ 25%] Building CXX object deps/magnum/src/Magnum/GL/CMakeFiles/MagnumGLObjects.dir/Implementation/TextureState.cpp.o
[ 26%] Building CXX object deps/corrade/src/Corrade/Utility/CMakeFiles/CorradeUtility.dir/Arguments.cpp.o
make[2]: *** No rule to make target `/Users/ram.ramrakhya/research/habitat-sim/build_corrade-rc/deps/corrade/src/Corrade/Utility/corrade-rc', needed by `deps/magnum/src/Magnum/DebugTools/resource_MagnumDebugTools_RESOURCES.cpp'.  Stop.
make[1]: *** [deps/magnum/src/Magnum/DebugTools/CMakeFiles/MagnumDebugToolsObjects.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 26%] Building CXX object deps/corrade/src/Corrade/Utility/CMakeFiles/CorradeUtility.dir/ConfigurationGroup.cpp.o
[ 26%] Building CXX object deps/corrade/src/Corrade/Utility/CMakeFiles/CorradeUtility.dir/Format.cpp.o
[ 26%] Building CXX object deps/magnum/src/Magnum/GL/CMakeFiles/MagnumGLObjects.dir/Implementation/driverSpecific.cpp.o
[ 26%] Building CXX object deps/magnum/src/Magnum/GL/CMakeFiles/MagnumGLObjects.dir/Implementation/maxTextureSize.cpp.o
[ 27%] Building CXX object deps/magnum/src/Magnum/GL/CMakeFiles/MagnumGLObjects.dir/PrimitiveQuery.cpp.o
[ 28%] Building CXX object deps/corrade/src/Corrade/Utility/CMakeFiles/CorradeUtility.dir/Resource.cpp.o
[ 28%] Building CXX object deps/corrade/src/Corrade/Utility/CMakeFiles/CorradeUtility.dir/String.cpp.o
[ 28%] Building CXX object deps/magnum/src/Magnum/GL/CMakeFiles/MagnumGLObjects.dir/TextureArray.cpp.o
[ 28%] Building CXX object deps/magnum/src/Magnum/GL/CMakeFiles/MagnumGLObjects.dir/TransformFeedback.cpp.o
[ 28%] Building CXX object deps/corrade/src/Corrade/Utility/CMakeFiles/CorradeUtility.dir/Unicode.cpp.o
[ 29%] Building CXX object deps/magnum/src/Magnum/GL/CMakeFiles/MagnumGLObjects.dir/Implementation/TransformFeedbackState.cpp.o
[ 29%] Building CXX object deps/magnum/src/Magnum/GL/CMakeFiles/MagnumGLObjects.dir/AbstractQuery.cpp.o
[ 29%] Building CXX object deps/magnum/src/Magnum/GL/CMakeFiles/MagnumGLObjects.dir/Implementation/QueryState.cpp.o
[ 29%] Linking CXX static library ../../../../../Release/lib/CorradeUtility.bc
[ 29%] Built target MagnumGLObjects
[ 29%] Built target CorradeUtility
make: *** [all] Error 2
```

The same setup works for master branch and I am able to use WebGL setup for master branch.

## Command
`./build_js.sh`

## System Info

System Info
ENVIRONMENT INFO:
Platform: Darwin-18.7.0-x86_64-i386-64bit
Machine: x86_64
Processor: i386
Libc version:
Mac version: 10.13.3
Python version: 3.6.4
Architecture: 64bit
Win version:
System OS: Darwin
Release: 18.7.0
Version: Darwin Kernel Version 18.7.0: Tue Aug 20 16:57:14 PDT 2019; root:xnu-4903.271.2~2/RELEASE_X86_64
Operational System: darwin
GCC version: Apple LLVM version 10.0.0
CMAKE version: cmake version 3.14.3
NVIDIA-SMI:
Pip packages versions:

Conda packages versions:
attrs 19.3.0 py_0 19.1.0
numba 0.48.0 pypi_0 pypi
numpy 1.18.1 pypi_0 pypi
numpy-quaternion 2019.12.11.22.25.52 pypi_0 pypi
pillow 7.0.0 py36h4655f20_0
scipy 1.4.1 py36h9fa6033_0 1.3.0
tqdm 4.43.0 pypi_0 pypi

erikwijmans, 2020-06-30:
@msbaines @apsdehal may be able to help.  However, as the "wip-hacky" in the branch name suggests, I doubt anyone was intending to support that code, so the knowledge needed to get that branch to compile may have been lost to the sands of time.

msbaines, 2020-06-30:
Do not use this branch. WebGL should be supported in master.

Ram81, 2020-07-02:
@erikwijmans @msbaines sure. I will try to setup the iccv demo code in the webGL setup from master branch.

closed 2020-07-02 by Ram81