facebookresearch/habitat-sim #667: Segmentation fault on headless build when running the example script

asolano, 2020-07-01:
## 🐛 Bug

The example script for non-interactive testing fails with a segmentation fault. 

## Command
`$ python examples/example.py --scene ./data/scene_datasets/habitat-test-scenes/skokloster-castle.glb`

## To Reproduce

After a standard installation with miniconda and a headless build:

```
$ python examples/example.py --scene ./data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
sim_cfg.physics_config_file = ./data/default.phys_scene_config.json
==== Initialized Sensor Spec: =====
Sensor uuid:  color_sensor
Sensor type:  SensorType.COLOR
Sensor position:  [0.  1.5 0. ]
===================================
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0701 15:05:44.320101 111003 Simulator.cpp:96] Loading navmesh from ./data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
I0701 15:05:44.320597 111003 Simulator.cpp:98] Loaded.
I0701 15:05:44.320873 111003 SceneGraph.h:92] Created DrawableGroup: 
Segmentation fault
```

## Expected behavior

The example script finishes normally and prints some stats.

## Additional context

The system is a cluster using the Modules package, without sudo privileges. The cmake system did not get the right compiler version on its own, so we had to add environment variables to the build command:

```
$ module load cuda/10.0/10.0.130
$ module load cudnn/7.4/7.4.2
$ module load gcc/7.4.0
$ module load cmake/3.11.4 
$ CC=`which gcc` CXX=`which g++` python setup.py install --with-cuda --headless
```

## System Info

```
$ python  habitat_sim/utils/collect_env.py 
ENVIRONMENT INFO:
Platform: Linux-3.10.0-862.el7.x86_64-x86_64-with-centos-7.5.1804-Core
Machine: x86_64
Processor: x86_64
Libc version: glibc 2.9
Mac version: 
Python version: 3.6.10
Architecture: 64bit 
Win version:    
System OS: Linux
Release: 3.10.0-862.el7.x86_64
Version: #1 SMP Fri Apr 20 16:44:24 UTC 2018
Operational System: linux
GCC version: gcc (GCC) 7.4.0
CMAKE version: cmake version 3.11.4
NVIDIA-SMI: Wed Jul  1 15:14:51 2020       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.33.01    Driver Version: 440.33.01    CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  Tesla V100-SXM2...  On   | 00000000:B1:00.0 Off |                    0 |
| N/A   36C    P0    43W / 300W |      0MiB / 16160MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
Pip packages versions:
matplotlib==3.2.2
numba==0.50.1
numpy==1.19.0
numpy-quaternion==2020.5.19.15.27.24
tqdm==4.47.0
Conda packages versions:
attrs                     19.3.0                   pypi_0    pypi  19.1.0
gitpython                 3.1.3                    pypi_0    pypi
matplotlib                3.2.2                    pypi_0    pypi
numba                     0.50.1                   pypi_0    pypi
numpy                     1.19.0                   pypi_0    pypi
numpy-quaternion          2020.5.19.15.27.24          pypi_0    pypi
pillow                    7.2.0                    pypi_0    pypi
scipy                     1.5.0                    pypi_0    pypi  1.3.0
tqdm                      4.47.0                   pypi_0    pypi

```

erikwijmans, 2020-07-01:
Can you rebuild in debug mode, i.e. `python setup.py install --headless --with-cuda --debug` and then do:

```
> gdb python
(gdb) r  examples/example.py --scene ./data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
```

then once it segfaults, do `bt` to get the backtrace.  That will let us know where the segfault is happening.

asolano, 2020-07-02:
Hi Erik, thank you for the quick reply.

The `--debug` option seems to be not supported (using the `stable` tag)
```
$ CC=`which gcc` CXX=`which g++` python setup.py install --with-cuda --headless --debug
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: option --debug not recognized
```

So I opened `setup.py` and found it there: 

https://github.com/facebookresearch/habitat-sim/blob/020041d75eaf3c70378a9ed0774b5c67b9d3ce99/setup.py#L223 

Hard-coding it to `Debug` (I'm sure there is a better way for it) and compiling again, the debug trace is:

```
Program received signal SIGSEGV, Segmentation fault.
0x0000000000000000 in ?? ()
Missing separate debuginfos, use: debuginfo-install glibc-2.17-222.el7.x86_64 libX11-1.6.5-1.el7.x86_64 libXau-1.0.8-2.1.el7.x86_64 libXdamage-1.1.4-4.1.el7.x86_64 libXext-1.3.3-3.el7.x86_64 libXfixes-5.0.3-1.el7.x86_64 libXxf86vm-1.1.4-1.el7.x86_64 libdrm-2.4.83-2.el7.x86_64 libffi-3.0.13-18.el7.x86_64 libselinux-2.5-12.el7.x86_64 libuuid-2.23.2-52.el7.x86_64 libwayland-client-1.14.0-2.el7.x86_64 libwayland-server-1.14.0-2.el7.x86_64 libxcb-1.12-1.el7.x86_64 libxshmfence-1.2-1.el7.x86_64 mesa-libEGL-17.2.3-8.20171019.el7.x86_64 mesa-libGL-17.2.3-8.20171019.el7.x86_64 mesa-libgbm-17.2.3-8.20171019.el7.x86_64 mesa-libglapi-17.2.3-8.20171019.el7.x86_64 pcre-8.32-17.el7.x86_64
(gdb) bt
#0  0x0000000000000000 in ?? ()
#1  0x00002aaacc316749 in esp::gfx::(anonymous namespace)::ESPEGLContext::ESPEGLContext (this=0x555556d2a700, device=0)
    at /home/acb11899xv/habitat-sim/src/esp/gfx/WindowlessContext.cpp:96
#2  0x00002aaacc317955 in std::make_unique<esp::gfx::(anonymous namespace)::ESPEGLContext, int&> (__args#0=@0x7fffffffb074: 0)
    at /bb/apps/gcc/7.4.0/include/c++/7.4.0/bits/unique_ptr.h:825
#3  0x00002aaacc3177af in esp::gfx::(anonymous namespace)::ESPEGLContext::create_unique<int&> (args#0=@0x7fffffffb074: 0)
    at /home/acb11899xv/habitat-sim/src/esp/gfx/WindowlessContext.cpp:192
#4  0x00002aaacc318358 in esp::gfx::WindowlessContext::Impl::Impl (this=0x555556a5d020, device=0)
    at /home/acb11899xv/habitat-sim/src/esp/gfx/WindowlessContext.cpp:230
#5  0x00002aaacc318494 in spimpl::make_unique_impl<esp::gfx::WindowlessContext::Impl, int&> (args#0=@0x7fffffffb0f4: 0)
    at /home/acb11899xv/habitat-sim/src/esp/core/spimpl.h:426
#6  0x00002aaacc31767e in esp::gfx::WindowlessContext::WindowlessContext (this=0x555556a5d000, device=0)
    at /home/acb11899xv/habitat-sim/src/esp/gfx/WindowlessContext.cpp:277
#7  0x00002aaacc350aec in std::make_unique<esp::gfx::WindowlessContext, int&> (__args#0=@0x555556771784: 0)
    at /bb/apps/gcc/7.4.0/include/c++/7.4.0/bits/unique_ptr.h:825
#8  0x00002aaacc34f7a8 in esp::gfx::WindowlessContext::create_unique<int&> (args#0=@0x555556771784: 0)
    at /home/acb11899xv/habitat-sim/src/esp/gfx/WindowlessContext.h:22
#9  0x00002aaacc34a163 in esp::sim::Simulator::reconfigure (this=0x555556771670, cfg=...) at /home/acb11899xv/habitat-sim/src/esp/sim/Simulator.cpp:135
#10 0x00002aaacc349213 in esp::sim::Simulator::Simulator (this=0x555556771670, cfg=...) at /home/acb11899xv/habitat-sim/src/esp/sim/Simulator.cpp:35
#11 0x00002aaacc1ed06c in pybind11::detail::initimpl::construct_or_initialize<esp::sim::Simulator, esp::sim::SimulatorConfiguration const&, 0> (args#0=...)
    at /home/acb11899xv/habitat-sim/src/deps/pybind11/include/pybind11/detail/init.h:61
#12 0x00002aaacc1e73b7 in void pybind11::detail::initimpl::constructor<esp::sim::SimulatorConfiguration const&>::execute<pybind11::class_<esp::sim::Simulator, std::shared_ptr<esp::sim::Simulator> >, , 0>(pybind11::class_<esp::sim::Simulator, std::shared_ptr<esp::sim::Simulator> >&)::{lambda(pybind11::detail::value_and_holder&, esp::sim::SimulatorConfiguration const&)#1}::operator()(pybind11::detail::value_and_holder&, esp::sim::SimulatorConfiguration const&) const (
    __closure=0x5555560c29d8, v_h=..., args#0=...) at /home/acb11899xv/habitat-sim/src/deps/pybind11/include/pybind11/detail/init.h:174
#13 0x00002aaacc2026a3 in pybind11::detail::argument_loader<pybind11::detail::value_and_holder&, esp::sim::SimulatorConfiguration const&>::call_impl<void, void pybind11::detail::initimpl::constructor<esp::sim::SimulatorConfiguration const&>::execute<pybind11::class_<esp::sim::Simulator, std::shared_ptr<esp::sim::Simulator> >, , 0>(pybind11::class_<esp::sim::Simulator, std::shared_ptr<esp::sim::Simulator> >&)::{lambda(pybind11::detail::value_and_holder&, esp::sim::SimulatorConfiguration const&)#1}&, 0ul, 1ul, pybind11::detail::void_type>(void pybind11::detail::initimpl::constructor<esp::sim::SimulatorConfiguration const&>::execute<pybind11::class_<esp::sim::Simulator, std::shared_ptr<esp::sim::Simulator> >, , 0>(pybind11::class_<esp::sim::Simulator, std::shared_ptr<esp::sim::Simulator> >&)::{lambda(p---Type <return> to continue, or q <return> to quit---
ybind11::detail::value_and_holder&, esp::sim::SimulatorConfiguration const&)#1}&, std::integer_sequence<unsigned long, 0ul, 1ul>, pybind11::detail::void_type&&)
    (this=0x7fffffffb9c0, f=...) at /home/acb11899xv/habitat-sim/src/deps/pybind11/include/pybind11/cast.h:1931
#14 0x00002aaacc2008db in _ZNO8pybind116detail15argument_loaderIJRNS0_16value_and_holderERKN3esp3sim22SimulatorConfigurationEEE4callIvNS0_9void_typeERZNS0_8initimpl11constructorIJS8_EE7executeINS_6class_INS5_9SimulatorEJSt10shared_ptrISH_EEEEJELi0EEEvRT_DpRKT [... cut ...]
#58 0x000055555576df63 in run_file (p_cf=0x7fffffffd11c, filename=0x5555558a8330 L"examples/example.py", fp=0x555555d38010)
    at /tmp/build/80754af9/python_1588903631989/work/Modules/main.c:340
#59 Py_Main () at /tmp/build/80754af9/python_1588903631989/work/Modules/main.c:811
#60 0x0000555555637f2e in main () at /tmp/build/80754af9/python_1588903631989/work/Programs/python.c:69
#61 0x00002aaaaaf0d445 in __libc_start_main () from /lib64/libc.so.6
#62 0x000055555571727f in _start () at ../sysdeps/x86_64/elf/start.S:103
(gdb) 
```

Hope it helps!

erikwijmans, 2020-07-02:
Ah sorry, that should be `python setup.py build_ext --debug install --headless --with-cuda`

erikwijmans, 2020-07-02:
Line 96 is `eglQueryDevicesEXT`, it seems like that function is a null pointer, which would imply a driver issue or an issue with glvnd.  @mosra Have you seen that before?

asolano, 2020-07-02:
Thanks, the updated command works better 👍 

Regarding the driver issue, it seems the installation lacks `libglvnd` so I guess that explains the null pointer.

```
$ yum list | grep libEGL
mesa-libEGL.x86_64                        17.2.3-8.20171019.el7        @dvd     
mesa-libEGL-devel.x86_64                  17.2.3-8.20171019.el7        @dvd     
mesa-libEGL.i686                          17.2.3-8.20171019.el7        dvd      
mesa-libEGL-devel.i686                    17.2.3-8.20171019.el7        dvd      

$ yum list | grep libglvnd
$
```

Since I don't have root access I guess my only options are compiling it locally or going for a Singularity container? (no custom Docker images allowed).

erikwijmans, 2020-07-02:
The Nvidia driver should provide that in most cases, look for `libGLdispatch`.

asolano, 2020-07-03:
Thanks for the suggestion, Erik. Unfortunately it's not there either, it seems the option for the GL libraries was not selected during the driver installation (the `--no-opengl-libs` command line flag).

Just in case I tried to compile `libglvnd` from source and install it to my `$HOME`, and modify the LD_LIBRARY_PATH variable to include it, the result of the sample is:

```
==== Initialized Sensor Spec: =====
Sensor uuid:  color_sensor
Sensor type:  SensorType.COLOR
Sensor position:  [0.  1.5 0. ]
===================================
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0702 15:00:34.318120 19023 Simulator.cpp:96] Loading navmesh from ./data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
I0702 15:00:34.320502 19023 Simulator.cpp:98] Loaded.
I0702 15:00:34.320621 19023 SceneGraph.h:92] Created DrawableGroup: 
F0702 15:00:34.320842 19023 WindowlessContext.cpp:99] Check failed: numDevices > 0 [EGL] No devices detected
*** Check failure stack trace: ***

Program received signal SIGABRT, Aborted.
0x00002aaaaaf21277 in raise () from /lib64/libc.so.6
Missing separate debuginfos, use: debuginfo-install glibc-2.17-222.el7.x86_64 libX11-1.6.5-1.el7.x86_64 libXau-1.0.8-2.1.el7.x86_64 libXext-1.3.3-3.el7.x86_64 libuuid-2.23.2-52.el7.x86_64 libxcb-1.12-1.el7.x86_64
(gdb) bt
#0  0x00002aaaaaf21277 in raise () from /lib64/libc.so.6
#1  0x00002aaaaaf22968 in abort () from /lib64/libc.so.6
#2  0x00002aaaccbfefd8 in google::logging_fail () at /home/acb11899xv/habitat-sim/src/deps/glog/src/logging.cc:1507
#3  0x00002aaaccbff002 in google::LogMessage::Fail () at /home/acb11899xv/habitat-sim/src/deps/glog/src/logging.cc:1520
#4  0x00002aaaccbfef5f in google::LogMessage::SendToLog (this=0x7fffffffae40) at /home/acb11899xv/habitat-sim/src/deps/glog/src/logging.cc:1477
#5  0x00002aaaccbfe8c2 in google::LogMessage::Flush (this=0x7fffffffae40) at /home/acb11899xv/habitat-sim/src/deps/glog/src/logging.cc:1346
#6  0x00002aaaccc01d26 in google::LogMessageFatal::~LogMessageFatal (this=0x7fffffffae40, __in_chrg=<optimized out>)
    at /home/acb11899xv/habitat-sim/src/deps/glog/src/logging.cc:2096
#7  0x00002aaacc3168ac in esp::gfx::(anonymous namespace)::ESPEGLContext::ESPEGLContext (this=0x555556d25ad0, device=0)
    at /home/acb11899xv/habitat-sim/src/esp/gfx/WindowlessContext.cpp:99
#8  0x00002aaacc317965 in std::make_unique<esp::gfx::(anonymous namespace)::ESPEGLContext, int&> (__args#0=@0x7fffffffb064: 0)
    at /bb/apps/gcc/7.4.0/include/c++/7.4.0/bits/unique_ptr.h:825
#9  0x00002aaacc3177bf in esp::gfx::(anonymous namespace)::ESPEGLContext::create_unique<int&> (args#0=@0x7fffffffb064: 0)
    at /home/acb11899xv/habitat-sim/src/esp/gfx/WindowlessContext.cpp:192
#10 0x00002aaacc318368 in esp::gfx::WindowlessContext::Impl::Impl (this=0x555556c91a50, device=0)
    at /home/acb11899xv/habitat-sim/src/esp/gfx/WindowlessContext.cpp:230
#11 0x00002aaacc3184a4 in spimpl::make_unique_impl<esp::gfx::WindowlessContext::Impl, int&> (args#0=@0x7fffffffb0e4: 0)
    at /home/acb11899xv/habitat-sim/src/esp/core/spimpl.h:426
#12 0x00002aaacc31768e in esp::gfx::WindowlessContext::WindowlessContext (this=0x555556c91a30, device=0)
---Type <return> to continue, or q <return> to quit---
```

Note the addition of the  "No devices detected" part. My guess is the vendor dispatch API is working but has nothing to dispatch calls to.

erikwijmans, 2020-07-03:
Yeah, unfortunately the Nvidia specific version of libEGL (libEGL_nvidia) is necessary for getting a headless context on Nvidia GPUs.  Depending on the driver install process, it should be possible to install these without requiring a node reboot.

mosra, 2020-07-03:
> Line 96 is `eglQueryDevicesEXT`, it seems like that function is a null pointer, which would imply a driver issue or an issue with glvnd. @mosra Have you seen that before?

There's no line `src/esp/gfx/WindowlessContext.cpp:96` nor any `ESPEGLContext` anymore, in current master. The [original code](https://github.com/facebookresearch/habitat-sim/blob/175ea235d9a12cc0a7bb6903c474c705b48438d4/src/esp/gfx/WindowlessContext.cpp#L96) doesn't seem to be checking for required extensions before, so *if* the target machine has Mesa before 19.4 or any other driver, such extension is probably not there at all which is why it's crashing. I think?

asolano, 2020-07-07:
I see, thanks for the confirmation. Unfortunately we don't have sudo access to this particular cluster so installing new packages is not an option.

I will test Singularity and see if I can get the libraries installed inside the container.

asolano, 2020-07-08:
Just a quick check, is it possible to not use EGL in habitat? I tried setting `MUJOCO_GL=glfw` as with `dm_control` but it did not take.

erikwijmans, 2020-07-08:
For true headless mode, no, you can only do it via EGL.  If you a machine with an attached display, you can build in non-headless mode and it will piggyback off the window server (in a similar way that `MUJOCO_GL=glfw` does) and run in windowless mode.

asolano, 2020-07-09:
Thanks for the quick reply, Erik. With `dm_control` we can use a VNC server to attach and run, so that option could work too.

Unfortunately we get a new error when removing the `--headless` flag:

```
[ 71%] Building CXX object deps/magnum/src/Magnum/MeshTools/CMakeFiles/MagnumMeshTools.dir/RemoveDuplicates.cpp.o
[ 71%] Building CXX object deps/magnum/src/Magnum/MeshTools/CMakeFiles/MagnumMeshTools.dir/Compile.cpp.o
[ 71%] Building CXX object deps/magnum/src/Magnum/MeshTools/CMakeFiles/MagnumMeshTools.dir/Interleave.cpp.o
/home/acb11899xv/habitat-sim/src/deps/magnum/src/Magnum/Platform/WindowlessGlxApplication.cpp: In constructor ‘Magnum::Platform::WindowlessGlxContext::WindowlessGlxContext(const Magnum::Platform::WindowlessGlxContext::Configuration&, Magnum::Platform::GLContext*)’:
/home/acb11899xv/habitat-sim/src/deps/magnum/src/Magnum/Platform/WindowlessGlxApplication.cpp:92:9: error: ‘GLX_CONTEXT_PROFILE_MASK_ARB’ was not declared in this scope
         GLX_CONTEXT_PROFILE_MASK_ARB, GLX_CONTEXT_CORE_PROFILE_BIT_ARB,
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/acb11899xv/habitat-sim/src/deps/magnum/src/Magnum/Platform/WindowlessGlxApplication.cpp:92:9: note: suggested alternative: ‘GL_CONTEXT_PROFILE_MASK’
         GLX_CONTEXT_PROFILE_MASK_ARB, GLX_CONTEXT_CORE_PROFILE_BIT_ARB,
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
         GL_CONTEXT_PROFILE_MASK
/home/acb11899xv/habitat-sim/src/deps/magnum/src/Magnum/Platform/WindowlessGlxApplication.cpp:92:39: error: ‘GLX_CONTEXT_CORE_PROFILE_BIT_ARB’ was not declared in this scope
         GLX_CONTEXT_PROFILE_MASK_ARB, GLX_CONTEXT_CORE_PROFILE_BIT_ARB,
                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/acb11899xv/habitat-sim/src/deps/magnum/src/Magnum/Platform/WindowlessGlxApplication.cpp:92:39: note: suggested alternative: ‘GL_CONTEXT_CORE_PROFILE_BIT’
         GLX_CONTEXT_PROFILE_MASK_ARB, GLX_CONTEXT_CORE_PROFILE_BIT_ARB,
                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                       GL_CONTEXT_CORE_PROFILE_BIT
gmake[2]: *** [deps/magnum/src/Magnum/Platform/CMakeFiles/MagnumWindowlessGlxApplication.dir/WindowlessGlxApplication.cpp.o] Error 1
gmake[1]: *** [deps/magnum/src/Magnum/Platform/CMakeFiles/MagnumWindowlessGlxApplication.dir/all] Error 2
```

Any idea what might be the cause? Searching past issues did not get any results.

mosra, 2020-07-09:
Interesting, I'm not aware of getting such compile error *ever*, even systems from almost a decade ago had those GLX defines present.

I could fix this on the engine side, but I suspect that if those defines are not present, then the driver wouldn't support `GLX_ARB_create_context_profile` either (which is assumed here). What's the package that owns `/usr/include/GL/glxext.h` and how old is it?

erikwijmans, 2020-07-09:
VNC is rough.  I would highly recommend bugging someone to install the Nvidia driver OpenGL libs (you shouldn't need to reboot to do that, so it should be fairly painless).  With headed mode, you will be constrained to just being able to target one GPU, which is sad.  We may be able to hack around to get you up and running, but true headless mode is a thing of beauty when working on a cluster, so it is likely worth your time to see if you can make that happen :)

asolano, 2020-07-10:
Thanks for joining in, Vladimir. It's a relatively new system AFAIK, not even close to a decade, so it is surprising. 

The GL extension headers are:
```
$ yum whatprovides  /usr/include/GL/glxext.h
Loaded plugins: etckeeper, fastestmirror
Loading mirror speeds from cached hostfile
mesa-libGL-devel-17.2.3-8.20171019.el7.i686 : Mesa libGL development package
Repo        : dvd
Matched from:
Filename    : /usr/include/GL/glxext.h

mesa-libGL-devel-17.2.3-8.20171019.el7.x86_64 : Mesa libGL development package
Repo        : dvd
Matched from:
Filename    : /usr/include/GL/glxext.h

mesa-libGL-devel-17.2.3-8.20171019.el7.x86_64 : Mesa libGL development package
Repo        : @dvd
Matched from:
Filename    : /usr/include/GL/glxext.h
```

So about three years? It is a standard HPC-oriented cluster but with a decent number of GPU nodes, but for some reason headless rendering does not seem to be a priority.

Yes, Erik, VNC would be a kind of compromise. Ideally we'd like to run habitat on this system headless so we can use all GPUs in each node (and if possible combine several nodes too) and have more people iterate faster on research ideas. Now that you confirmed it is actually possible the prospect is even more enticing, so I may use your comment to try to persuade the right people!

erikwijmans, 2020-07-10:
As a note on that, you will have to start a VNC per GPU you want to do rendering on -- i.e. the `gpu_device_id` in the config must be 0 when the simulator is not built in headless mode.

In terms of rallying people, MuJoCo will also use EGL to do headless rendering, so no one would need to use VNC for MuJoCo experiments either :)

asolano, 2020-07-15:
Good point, headless rendering is definitely the way to go and any cluster that has AI in mind should probably have the EGL libraries installed :)

After much fiddling we managed to get a Singularity image that mimics the exact OS and NVIDIA driver inside the container, and lo and behold it works. Both `example.py` and `benchmark.py` ran to completion on a single node without issue, so for the time being it is an acceptable solution (multi-gpu and multi-node are still pending).

(It is a bit of an anti-pattern so I still plan to try to convince the administrators to install the proper libraries at the host level.)

For reference, in case someone has a similar problem in the future, here is the recipe:

```
Bootstrap: docker
From: nvidia/cudagl:10.1-devel-centos7

%setup
    #Runs on host. The path to the image is $SINGULARITY_ROOTFS

%post
    # Post setup, runs inside the image

    # Install NVIDIA driver inside the container to get the EGL libraries
    # FIXME Version must match exactly the one at the host
    yum install -y epel-release
    yum-config-manager --add-repo https://developer.download.nvidia.com/compute/cuda/repos/rhel7/x86_64/
    yum install -y kernel-devel kernel-headers gcc
    yum install -y nvidia-driver-branch-440-440.33.01-1.el7
    
    # gcc7
    yum group install -y "Development Tools"
    yum install -y centos-release-scl
    yum install -y devtoolset-7-gcc-c++

    # Dependencies
    yum install -y \
        curl \
        wget \
        git \
        curl \
        vim

    # Conda
    wget -O miniconda.sh https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh
    bash miniconda.sh -b -p /opt/conda
    /opt/conda/bin/conda install -y numpy pyyaml scipy ipython mkl mkl-include
    /opt/conda/bin/conda clean -ya
    
    export PATH="/opt/conda/bin:$PATH"
    conda --version
    
    # Cmake 3.14
    wget https://github.com/Kitware/CMake/releases/download/v3.14.0/cmake-3.14.0-Linux-x86_64.sh
    mkdir /opt/cmake
    sh /cmake-3.14.0-Linux-x86_64.sh --prefix=/opt/cmake --skip-license
    ln -s /opt/cmake/bin/cmake /usr/local/bin/cmake
    cmake --version

    conda create -y -n habitat python=3.6 cmake=3.14.0
    
    # Enable GCC7, default version fails 
    scl enable devtoolset-7 bash

    export CUDACXX="/usr/local/cuda-10.1/bin/nvcc"
    git clone --branch stable https://github.com/facebookresearch/habitat-sim.git
    /bin/bash -c ". activate habitat; cd habitat-sim; pip install -r requirements.txt; python setup.py install --with-cuda --headless"
    git clone --branch stable https://github.com/facebookresearch/habitat-api.git
    /bin/bash -c ". activate habitat; cd habitat-api; pip install -e ."

%environment
    LANG=en_US.UTF-8
    LANGUAGE=en_US:en
    LC_ALL=en_US.UTF-8
    export LANG LANGUAGE LC_ALL
    export PATH=/opt/conda/bin:$PATH
    export GLOG_minloglevel=2
    export MAGNUM_LOG="quiet"

%runscript
    #Executes when the "singularity run" command is used
    #Useful when you want the container to run as an executable
    source /opt/conda/etc/profile.d/conda.sh
    conda activate habitat
    exec python $@ 
```

closed 2020-08-09 by dhruvbatra