facebookresearch/habitat-sim #681: GPU memory usage

knzo25, 2020-07-10:
I apologize if should be asking this anywhere else,
but I have some doubts about the GPU memory management of habitat-sim.

I have seen that although habitat-sim usually uses about 2.2GB of GPU for MP3D (RGBD 256x256),
when the available memory is not enough, somehow the processes "shrink" their memory usage, as low as 800MB or so. How does this happen? Can it be controlled/enabled by default?

My problem is that after initializing the environments, which use a lot of memory, I initialize RL agents in pytorch, which usually ends up with in out-of-memory errors. Knowing that the envs could use less memory makes me want to understand the phenomena as it would allow me to run more experiments in parallel.

Thank you !

Skylion007, 2020-07-10:
OpenGL will automatically page memory to system memory if needed (like CUDA Unified Memory).  The downside is that if you push all this memory to system memory performance will really, really suffer so we don't really provide anyways of controlling it.

If you care about reducing memory usage, make sure you are using our newest basis-fied MP3D meshes as they very memory efficient.

knzo25, 2020-07-11:
Thanks ! you're a life saver, I was looking the code for something more specific but has having no luck ahaha.

I downloaded the habitat-mp3d again (the new zip file is 16.1GB compared with the one 15.2GB I had). However, now each process takes 3273MB of GPU compared with the 2276 it previously used !
Is this the new meshes you mentioned? If possible, can you tell me how much memory it uses in your case?

Also (although a bit unrelated), do you know why the amount of  memory each simulator uses is not constant with respect to the amount of simulator instances? For example 2 processes = 500MB each. 4 processes = 1GB (each),  6 processes = 3273MB (each).

Thanks !

erikwijmans, 2020-07-11:
The amount of memory the simulator uses is a function of the scene loaded.  For MP3D especially, this can vary wildly (about 2 orders of magnitude).

closed 2020-07-22 by Skylion007