facebookresearch/habitat-sim #688: Bullet Physics Support for WebGL

Ram81, 2020-07-15:
## ❓ Questions and Help

I am trying to add support for actions like `addObject` and `removeObject` in the WebGL setup of habitat-sim. In order to get it working I need to add bindings for `PhysicsManager` in the current setup. But when I am trying to create a simple class binding for the `PhysicsManager` in `src/esp/binding_js/bindings_js.cpp`. I am getting the following error

```
In file included from /Users/ram.ramrakhya/research/habitat-sim/src/esp/bindings_js/bindings_js.cpp:5:
In file included from /Users/ram.ramrakhya/research/emsdk/fastcomp/emscripten/system/include/emscripten/bind.h:16:
In file included from /Users/ram.ramrakhya/research/emsdk/fastcomp/emscripten/system/include/libcxx/string:477:
In file included from /Users/ram.ramrakhya/research/emsdk/fastcomp/emscripten/system/include/libcxx/string_view:176:
In file included from /Users/ram.ramrakhya/research/emsdk/fastcomp/emscripten/system/include/libcxx/__string:56:
In file included from /Users/ram.ramrakhya/research/emsdk/fastcomp/emscripten/system/include/libcxx/algorithm:640:
/Users/ram.ramrakhya/research/emsdk/fastcomp/emscripten/system/include/libcxx/type_traits:1584:38: error:
      incomplete type 'esp::physics::PhysicsManager' used in type trait expression
    : public integral_constant<bool, __is_polymorphic(_Tp)> {};
                                     ^
/Users/ram.ramrakhya/research/emsdk/fastcomp/emscripten/system/include/emscripten/wire.h:67:51: note:
      in instantiation of template class 'std::__2::is_polymorphic<esp::physics::PhysicsManager>'
      requested here
                if(has_unbound_type_names || std::is_polymorphic<C>::value) {
                                                  ^
/Users/ram.ramrakhya/research/emsdk/fastcomp/emscripten/system/include/emscripten/wire.h:105:40: note:
      in instantiation of member function
      'emscripten::internal::LightTypeID<esp::physics::PhysicsManager>::get' requested here
                return LightTypeID<T>::get();
                                       ^
/Users/ram.ramrakhya/research/emsdk/fastcomp/emscripten/system/include/emscripten/bind.h:1136:36: note:
      in instantiation of member function
      'emscripten::internal::TypeID<esp::physics::PhysicsManager>::get' requested here
                TypeID<ClassType>::get(),
                                   ^
/Users/ram.ramrakhya/research/habitat-sim/src/esp/bindings_js/bindings_js.cpp:260:3: note: in
      instantiation of member function 'emscripten::class_<esp::physics::PhysicsManager,
      emscripten::internal::NoBaseClass>::class_' requested here
  em::class_<esp::physics::PhysicsManager>("PhysicsManager");
  ^
/Users/ram.ramrakhya/research/habitat-sim/src/esp/assets/ResourceManager.h:60:7: note: forward
      declaration of 'esp::physics::PhysicsManager'
class PhysicsManager;
      ^
In file included from /Users/ram.ramrakhya/research/habitat-sim/src/esp/bindings_js/bindings_js.cpp:5:
In file included from /Users/ram.ramrakhya/research/emsdk/fastcomp/emscripten/system/include/emscripten/bind.h:21:
In file included from /Users/ram.ramrakhya/research/emsdk/fastcomp/emscripten/system/include/emscripten/val.h:15:
/Users/ram.ramrakhya/research/emsdk/fastcomp/emscripten/system/include/emscripten/wire.h:87:66: error:
      no member named 'value' in 'std::__2::is_polymorphic<esp::physics::PhysicsManager>'
            if(has_unbound_type_names || std::is_polymorphic<C>::value) {
                                         ~~~~~~~~~~~~~~~~~~~~~~~~^
/Users/ram.ramrakhya/research/emsdk/fastcomp/emscripten/system/include/emscripten/bind.h:1036:20: note:
      in instantiation of function template specialization
      'emscripten::internal::getLightTypeID<esp::physics::PhysicsManager>' requested here
            return getLightTypeID(*ptr);
                   ^
/Users/ram.ramrakhya/research/emsdk/fastcomp/emscripten/system/include/emscripten/bind.h:1130:36: note:
      in instantiation of function template specialization
      'emscripten::internal::getActualType<esp::physics::PhysicsManager>' requested here
            auto _getActualType = &getActualType<ClassType>;
                                   ^
/Users/ram.ramrakhya/research/habitat-sim/src/esp/bindings_js/bindings_js.cpp:260:3: note: in
      instantiation of member function 'emscripten::class_<esp::physics::PhysicsManager,
      emscripten::internal::NoBaseClass>::class_' requested here
  em::class_<esp::physics::PhysicsManager>("PhysicsManager");
  ^
2 errors generated.
shared:ERROR: compiler frontend failed to generate LLVM bitcode, halting
make[2]: *** [esp/bindings_js/CMakeFiles/hsim_bindings.dir/bindings_js.cpp.o] Error 1
make[1]: *** [esp/bindings_js/CMakeFiles/hsim_bindings.dir/all] Error 2
make: *** [all] Error 2
```

For starters I have only tried creating a class without any functions. Can someone help me understand why it is not working?
```
em::class_<esp::physics::PhysicsManager>("PhysicsManager");
```
Here's a link to my local [branch](https://github.com/Ram81/habitat-sim/blob/webgl/src/esp/bindings_js/bindings_js.cpp)

jturner65, 2020-07-15:
Hello

PhysicsManager is not currently intended to be a user-consumed construct; AddObject and RemoveObject are handled currently through the Simulator object in habitat-sim's c++ and python paradigms.  Any javascript access to this functionality should consume bindings that process the Simulator class.  

See [this](https://github.com/facebookresearch/habitat-sim/blob/9eabc8a518ef5eaa6273a43b405c88988e38a81d/src/esp/bindings_js/bindings_js.cpp#L219) to see current Simulator javascript bindings, and see [simulator](https://github.com/facebookresearch/habitat-sim/blob/9eabc8a518ef5eaa6273a43b405c88988e38a81d/src/esp/sim/Simulator.h#L213) for the underlying c++ engine code that does what you are trying to do.  

I would also suggest checking out the pybind11 bindings for the same functionality, since they are similar in many ways to the javascript bindings, and currently offer more complete coverage of habitat-sim functionality.  You may be able to get some ideas on how to accomplish what you want by looking at them.  

For example, [SimBindings.cpp](https://github.com/facebookresearch/habitat-sim/blob/9eabc8a518ef5eaa6273a43b405c88988e38a81d/src/esp/bindings/SimBindings.cpp#L99) shows the Simulator method bindings used to add objects.  Also, the [python rigid object tutorial](https://github.com/facebookresearch/habitat-sim/blob/9eabc8a518ef5eaa6273a43b405c88988e38a81d/examples/tutorials/nb_python/rigid_object_tutorial.py#L198) demonstrates the process of adding rigid objects in python.  Any similar process in javascript would probably have to follow a similar pathway, and require similar bindings.

Ram81, 2020-07-15:
@jturner65 thanks for the prompt response. At first I tried using the same approach you mentioned and exposing the functions from Simulator through a js binding, you can see the code [here](https://github.com/Ram81/habitat-sim/blob/webgl/src/esp/bindings_js/bindings_js.cpp#L247). But it didn't work and when i started debugging I observed that `PhysicsManager` is not being instantiated. I am not sure how to move forward with this so I thought of instantiating both ResourceManger and PhysicsManager on js side by exposing bindings and using those objects for Simulator Physics initialization. 

If you have any other approach it would be helpful.

jturner65, 2020-07-15:
Hi
If PhysicsManager is only instantiated when the (server) code is compiled with bullet/physics support and then executed with appropriate tags.  I am not sure if the build_js.sh will even support the required tags, to be honest, now that i look over things.  

If i were going to try to get object/physics support in webgl this is where I would start - make sure the code can be compiled properly with bullet support (which would require modifications to build_js.sh I imagine), and the server is launched with appropriate flags as well.  

We currently have no plans to support physics in webgl, although we would gladly review PRs along these lines :)

Ram81, 2020-07-15:
Makes sense to me. I just went over the JS code once and I saw that SimulatorConfiguration is not using `enablePhysics` flag right now. Just to verify whether the bullet build is present or not I am trying to enablePhysics in WebGL simulator once to see if it works or not. In case it doesn't work I can work on adding the support for that and open up a PR for same.

jturner65, 2020-07-15:
Great! I for one would love to see progress on this.

Ram81, 2020-07-15:
@jturner65 I tested it and it seems WebGL is already being compiled with bullet. If we just set the enable physics configuration in `SimulatorConfiguration` the physics manager and resource manager is being initialized.

Ram81, 2020-07-15:
@jturner65 After debugging more about why added objects are not being rendered in WebGL I found out that `BulletPhysicsManager` is not there in WebGL. And it is using default `PhysicsManager` when ran with `enablePhysics`. I cross verified in the `build_js` folder and it doesn't have Bullet files in `esp/physics` package

Skylion007, 2020-07-15:
So you are going to need to make a few changes to the build_js.sh file to get it built with bullet.

Firstly, you need to add -DBUILD_WITH_BULLET=ON to cmake configuration lines.
You will probably also need to add some bullet libraries built with emscriptem. And example of how to do that can be found in [here](https://github.com/mosra/magnum-integration/blob/master/package/ci/travis-emscripten.sh).

@mosra Probably has the best idea of how to cleanly do it.

dhruvbatra, 2020-07-15:
Actually, @HelloN00bman has probably dealt with this most recently. Ben -- would you be able to answer Ram's question?

jturner65, 2020-07-15:
> Firstly, you need to add -DBUILD_WITH_BULLET=ON to cmake configuration lines.

This is what I meant by compiling the code with bullet support.  Without bullet, "physics" consists of kinematic motion support provided by PhysicsManager.

I would start with this tag set first.  You may not require anything else to get dynamics working properly.

jturner65, 2020-07-15:
> @jturner65 After debugging about why added objects are not being rendered in WebGL I found out that `BulletPhysicsManager` is not there in WebGL. And WebGL is using the default `PhysicsManager` when ran with `enablePhysics`. I cross verified in the `build_js` folder and it doesn't have Bullet files in `esp/physics` package

Just to reiterate, bullet being present or not would not have any bearing on whether objects are being rendered or not. With just PhysicsManager, an object (either a primitive-based or file-based object) can be created and put in the scene, and it will stay where it is put until it is explicitly moved.

Ram81, 2020-07-15:
Okay got it. I will debug why the objects are not being rendered with the PhysicsManager.

Skylion007, 2020-07-15:
@Ram81 I modified the build_js script to build bullet:
```
#!/usr/bin/env bash

# Propagate failures properly
set -e

git submodule update --init --recursive

mkdir -p build_corrade-rc
pushd build_corrade-rc
cmake ../src \
    -DBUILD_GUI_VIEWERS=OFF \
    -DBUILD_PYTHON_BINDINGS=OFF \
    -DBUILD_ASSIMP_SUPPORT=OFF \
    -DBUILD_DATATOOL=OFF \
    -DBUILD_PTEX_SUPPORT=OFF
cmake --build . --target corrade-rc
popd

mkdir -p build_js
cd build_js
echo "$EMSCRIPTEN"
cmake ../src \
    -DCORRADE_RC_EXECUTABLE=../build_corrade-rc/RelWithDebInfo/bin/corrade-rc \
    -DBUILD_GUI_VIEWERS=ON \
    -DBUILD_PYTHON_BINDINGS=OFF \
    -DBUILD_ASSIMP_SUPPORT=OFF \
    -DBUILD_DATATOOL=OFF \
    -DBUILD_PTEX_SUPPORT=OFF \
    -DCMAKE_BUILD_TYPE=Release \
    -DCMAKE_PREFIX_PATH="$EMSCRIPTEN" \
    -DCMAKE_TOOLCHAIN_FILE="../src/deps/magnum-integration/toolchains/generic/Emscripten-wasm.cmake" \
    -DCMAKE_FIND_ROOT_PATH=$(pwd) \
    -DCMAKE_INSTALL_PREFIX="." \
    -DBUILD_WITH_BULLET=ON \
    -DUSE_EMSCRIPTEN_PORTS_BULLET=ON \
    -DCMAKE_CXX_FLAGS="-s FORCE_FILESYSTEM=1 -s ALLOW_MEMORY_GROWTH=1" \
    -DCMAKE_EXE_LINKER_FLAGS="-s USE_WEBGL2=1  -s USE_BULLET=1"

cmake --build . -- -j 80
cmake --build . --target install -- -j 80

echo "Done building."
echo "Run:"
echo "python2 -m SimpleHTTPServer 8000"
echo "Or:"
echo "python3 -m http.server"
echo "Then open in browser:"
echo "http://0.0.0.0:8000/build_js/utils/viewer/viewer.html?scene=skokloster-castle.glb"
echo "Or:"
echo "http://0.0.0.0:8000/build_js/esp/bindings_js/bindings.html?scene=skokloster-castle.glb"
````

jturner65, 2020-07-15:
Are you sure they are not being rendered?  Perhaps they are being placed in the scene, but somewhere where they cannot be seen.

Ram81, 2020-07-15:
@Skylion007 thanks. I will test it once. 

@jturner65 I am trying to figure that out. the addObject actions runs successfully when triggered but I couldn't find the object in the scene. Once I confirm that I'll update here.

Skylion007, 2020-07-15:
@Ram81 Keep in mind that (0,0,0) is not necessarily in the middle of the object scene and you could be spawning the object outside of the room.

Ram81, 2020-07-19:
Closing this. Fixed in #690

closed 2020-07-19 by Ram81