facebookresearch/habitat-sim #689: Irreproducible RGBSensor Noise

Xiaoming-Zhao, 2020-07-16:
## ❓ Questions and Help

Hi, 

I am wondering whether we could make the RGB sensor noise deterministic. Currently, the RGB noise is totally random across different runs, which makes reproducibility from using Habitat-API almost impossible if users want to try noisy setting.

For example, if we use the [PointNav config file](https://github.com/facebookresearch/habitat-challenge/blob/master/configs/challenge_pointnav2020.local.rgbd.yaml) from this year's challenge as the following:
```python
env = habitat.Env(config=config)
obs = env.reset()
```
You will not get the same `obs['rgb']` every time you try the above snippet. I think the culprit is the `numba` parallel feature used here:
https://github.com/facebookresearch/habitat-sim/blob/9eabc8a518ef5eaa6273a43b405c88988e38a81d/habitat_sim/sensors/noise_models/gaussian_noise_model.py#L16

I am not sure to what extent it will hurt the speed if we disable it.

Do you have any suggestions? Thanks

Skylion007, 2020-07-16:
You have to set the np.seed within the @numba.jit compiled code section to change the seed used by JIT'd code. Numba actually uses a different, faster PRNG than vanilla numpy. 

See the code for setting the seed in [Habitat-API](https://github.com/facebookresearch/habitat-api/blob/e0807ed403902de78ee37b36766c2f06e5886fea/habitat/core/env.py#L276)

Xiaoming-Zhao, 2020-07-16:
Hi, thanks a lot for your reply. Actually, I have set the seed of the end. The minimal reproducible code is the following:
```python
config = habitat.get_config("/path/to/point_nav_habitat_challenge_2020.yaml")
env = habitat.Env(config=config)
env.seed(123)
obs = env.reset()
```

Meanwhile, I will have deterministic results if I set the following environment variables:
```bash
export NUMBA_NUM_THREADS=1
export OMP_NUM_THREADS=1
export MKL_NUM_THREADS=1
```

Therefore, I think the randomness comes from the `parallel` feature of `numba`, which makes sense since the behavior of parallelism could not be determined by setting the random seed.

Skylion007, 2020-07-17:
Yes, if you want to remove all the randomness of the JIT function, then you should set 'parallel=False' and 'fastmath=False,' but that will likely cause a significant performance penalty. The underlying PRNG is likely still faster than vanilla numpy's even without parallelism. Forcing all OpenMP and MKL code to be single threaded though will likely incur performance penalties.

Xiaoming-Zhao, 2020-07-17:
Thanks a lot for your reply. So for Habitat's current noisy settings, there is no elegant way to guarantee complete reproducibility, right?

closed 2020-07-20 by Xiaoming-Zhao