facebookresearch/habitat-sim #692: Question about Proximity Sensor

Xiaoming-Zhao, 2020-07-20:
## ❓ Questions and Help

Hi, I encounter an issue when using ProximitySensor in Habitat's PPO. However, I personally think it is more suitable to be proposed here. I continuously meet this error:
```bash
Math::Vector::projectedOntoNormalized(): line Vector(-nan, -nan, -nan) is not normalized
Traceback (most recent call last):
  File "/path/to/ddppo/algo/ddppo_trainer.py", line 549, in train
    running_episode_stats,
  File "/path/to/ppo/ppo_trainer.py", line 311, in _collect_rollout_step
    outputs = self.envs.step([a[0].item() for a in actions])
  File "/habitat-api/habitat/core/vector_env.py", line 389, in step
    return self.wait_step()
  File "/habitat-api/habitat/core/vector_env.py", line 376, in wait_step
    observations.append(read_fn())
  File "/path/to/python/lib/python3.6/multiprocessing/connection.py", line 250, in recv
    buf = self._recv_bytes()
  File "/path/to/python/lib/python3.6/multiprocessing/connection.py", line 407, in _recv_bytes
    buf = self._recv(4)
  File "/path/to/python/lib/python3.6/multiprocessing/connection.py", line 383, in _recv
    raise EOFError
EOFError
Exception ignored in: <bound method VectorEnv.__del__ of <habitat.core.vector_env.VectorEnv object at 0x7f5698eeb828>>
Traceback (most recent call last):
  File "/habitat-api/habitat/core/vector_env.py", line 518, in __del__
    self.close()
  File "/habitat-api/habitat/core/vector_env.py", line 397, in close
    read_fn()
  File "/path/to/python/lib/python3.6/multiprocessing/connection.py", line 250, in recv
    buf = self._recv_bytes()
  File "/path/to/python/lib/python3.6/multiprocessing/connection.py", line 407, in _recv_bytes
    buf = self._recv(4)
  File "/path/to/python/lib/python3.6/multiprocessing/connection.py", line 383, in _recv
    raise EOFError
EOFError:
```  
This traceback does not give detailed information about which function indeed causes the error of Magnum (I guess the `Math::Vector::projectedOntoNormalized()` comes from Magnum). Not sure whether the culprit are the following lines:
https://github.com/facebookresearch/habitat-sim/blob/92739ac92e0170601abf0e08d057a3ca148417e1/src/esp/nav/PathFinder.cpp#L1212-L1213

I would really appreciate it if you could give some suggestions! Thanks

erikwijmans, 2020-07-20:
The sensor and that error are likely unrelated as there isn't anything related to that sensor that would call `Math::Vector::projectedOntoNormalized()`.  Are there any errors or logging statements before that error?  The times I have seen that error something else goes wrong before hand as that error implies the agent's current state has a NaN somewhere in it.

Xiaoming-Zhao, 2020-07-20:
Hi, thanks a lot for your reply. I have checked the logging from ~10min before this error message's occurring but could not find something related to NaN. Can I know why there could be NaN in the agent's state? Is there a way to handle this NaN error outside the simulator? Currently, it seems like the simulator will automatically close the environment is NaN appears.

erikwijmans, 2020-07-21:
That generally happens when the navmesh is broken or the agent isn't close enough to a navigable region on the navmesh to get snapped to one.

It is something that shouldn't happen, so we don't have anything built to handle it.

Xiaoming-Zhao, 2020-07-31:
@erikwijmans Hi Erik, I encountered this problem again without ProximitySensor. The error message is the same and I just paste here some more logging info from Habitat-sim:
```bash
I0727 10:05:49.871567 27518 simulator.py:146] Loaded navmesh /path/to/dataset/Gibson/gibson/Monson.navmesh
I0727 10:05:49.872212 27518 simulator.py:158] Recomputing navmesh for agent's height 0.88 and radius 0.18.
I0727 10:05:49.891613 27518 PathFinder.cpp:375] Building navmesh with 275x112 cells
I0727 10:05:50.016330 27518 PathFinder.cpp:643] Created navmesh with 137 vertices 61 polygons
I0727 10:05:50.016373 27518 Simulator.cpp:609] reconstruct navmesh successful
Math::Vector::projectedOntoNormalized(): line Vector(-nan, -nan, -nan) is not normalized
Traceback (most recent call last):
  File "/path/to/rl/ddppo/algo/ddppo_trainer.py", line 557, in train
    running_episode_stats,
  File "/path/to/rl/ppo/ppo_trainer.py", line 311, in _collect_rollout_step
    outputs = self.envs.step([a[0].item() for a in actions])
  File "/path/to/habitat-api/habitat/core/vector_env.py", line 389, in step
    return self.wait_step()
  File "/path/to/habitat-api/habitat/core/vector_env.py", line 376, in wait_step
    observations.append(read_fn())
  File "/path/to/python/envs/lib/python3.6/multiprocessing/connection.py", line 250, in recv
    buf = self._recv_bytes()
  File "/path/to/python/envs/lib/python3.6/multiprocessing/connection.py", line 407, in _recv_bytes
    buf = self._recv(4)
  File "/path/to/python/envs/lib/python3.6/multiprocessing/connection.py", line 383, in _recv
    raise EOFError
EOFError
Exception ignored in: <bound method VectorEnv.__del__ of <habitat.core.vector_env.VectorEnv object at 0x7f1f1134c860>>
Traceback (most recent call last):
  File "/path/to/habitat-api/habitat/core/vector_env.py", line 518, in __del__
    self.close()
  File "/path/to/habitat-api/habitat/core/vector_env.py", line 397, in close
    read_fn()
  File "/path/to/python/envs/lib/python3.6/multiprocessing/connection.py", line 250, in recv
    buf = self._recv_bytes()
  File "/path/to/python/envs/lib/python3.6/multiprocessing/connection.py", line 407, in _recv_bytes
    buf = self._recv(4)
  File "/path/to/python/envs/lib/python3.6/multiprocessing/connection.py", line 383, in _recv
    raise EOFError
EOFError:
```

Is it possible that some episodes in Gibson's v2 dataset's training split are not suitable for the Agent's configuration of Habitat-Challenge 2020, where the agent is wider? Thanks

erikwijmans, 2020-07-31:
No, the Gibson v2 dataset was generated explicitly for that agent configuration.  What commit of habitat-api are you on?  Can you try being on one after https://github.com/facebookresearch/habitat-api/pull/399 (or removing that same code in your copy)?  That check could cause this issue in very rare circumstances.

Xiaoming-Zhao, 2021-05-24:
@erikwijmans Hi Erik, I continuously encountered error of `Math::Vector::projectedOntoNormalized(): line Vector(-nan, -nan, -nan) is not normalized` when I try to train DDPPO etc. It does not happen that often but it will happen when I train long enough.

I cannot find a good way to reproduce that until recently. Maybe the following steps can help identify the issue, which utilizes [OccAnt](https://github.com/facebookresearch/OccupancyAnticipation)'s data preprocessing script:

1. download OccAnt code.
2. download the dataset, which I think is just normal Gibson v2
```bash
# Download gibson task dataset
wget https://dl.fbaipublicfiles.com/OccupancyAnticipation/gibson.zip
unzip gibson.zip && rm gibson.zip
``` 
3. We only need the scene of `Ballou`, so remove all the other scenes's `*.json.gz` file under `gibson/v2/train/content`
4. run their preprocess script as described [here](https://github.com/facebookresearch/OccupancyAnticipation#generating-environment-layouts):
```bash
SPLIT="train"
cd $OCCANT_ROOT_DIR
python occant_baselines/generate_topdown_maps/generate_occant_gt_maps.py \
    --config-path occant_baselines/generate_topdown_maps/configs/occant_gt/gibson_${SPLIT}.yaml \
    --save-dir data/datasets/exploration/gibson/v2/${SPLIT}/occant_gt_maps \
    --global-map-size 961
```

I think you will encounter the error I mentioned above at around episode ID 21127. Sorry for not providing a minimal reproducible example since the error does not always appear. The running time for the above script is around 5min so I hope this is not that cumbersome. Thanks a lot in advance.

erikwijmans, 2021-05-24:
https://github.com/facebookresearch/OccupancyAnticipation/blob/master/occant_baselines/generate_topdown_maps/configs/occant_gt/gibson_train.yaml looks like it's for the v1 agent spec (the default) and the instructions say to use the v1 episodes.  Did you also change the agent config?  The v2 episodes are not compatible with the v1 agent

Xiaoming-Zhao, 2021-05-24:
Yes, I modified the config to change it to v2. [This](https://gist.github.com/Xiaoming-Zhao/3ba5fc04eaf6935c154941d9f6aedbc4) is my config.

erikwijmans, 2021-05-24:
```
ExplorationEpisode(episode_id='21128', scene_id='data/scene_datasets/gibson/Ballou.glb', start_position=[inf, inf, inf], start_rotation=[0, 0.10224925722765092, 0, -0.9947588096601073], info=None, _shortest_path_cache=None, start_room=None)
```

There's an episode with an inf's for the starting position.  I also check the episodes from the habitat-lab github page and those have an inf in the same spot.  I checked all the episodes in that zip and removed the ones with infs (there were two more).  They were all train episodes so they can be safely removed.

Xiaoming-Zhao, 2021-05-24:
Thanks a lot for checking and glad it issue has been identified!

closed 2021-05-24 by Xiaoming-Zhao