facebookresearch/habitat-sim #703: accessing build/viewer

VladislavHrosinkov, 2020-07-27:
## ❓ Questions and Help
How does one access build/viewer code? It does not seem to be on GitHub.

Skylion007, 2020-07-27:
Just pass -DBUILD_GUI_VIEWERS=ON to CMake.

VladislavHrosinkov, 2020-07-27:
Forgive me, but how is CMake involved? I know it it a dependency, but I have installed it without specifically in invoking it.

erikwijmans, 2020-07-27:
> How does one access build/viewer **code**

Is this what you are looking for: https://github.com/facebookresearch/habitat-sim/blob/master/src/utils/viewer/viewer.cpp?

closed 2020-08-09 by dhruvbatra

KenaHemnani, 2020-11-18:
How to build this viewer as it is already not there in build folder?

aclegg3, 2020-11-18:
You'll need to build habitat-sim for a system with display as described here: https://github.com/facebookresearch/habitat-sim/blob/master/BUILD_FROM_SOURCE.md (i.e., do not include the --headless argument).

KenaHemnani, 2020-11-19:
Hii @aclegg3 thank you for response , I installed habitat-sim using the same .But, still got error as :
**./build/viewer no such file or directory exists.**

aclegg3, 2020-11-19:
@KenaHemnani then we'll need more info from your build before we can move forward. You should see an alias to viewer in the habitat-sim/build directory and a habitat-sim/utils/viewer/ directory containing the viewer application.

KenaHemnani, 2020-11-20:
Hii @aclegg3 I reinstalled habitat-sim using conda . The headless version is working fine , but with display I am not able to visualize even with VNC , Yes I do have habitat-viewer inside habitat-sim/utils/viewer/ directory .So, I run following command
 **./habitat-viewer ~/DIR/habitat-lab/data/scene_datasets/hitat-test-scenes/skokloster-castle.glb**
It shows following `error:
[image]

Is there any other way to visualize the performance of agents as how it is navigating inside 3D habitat environment? locally on small dataset.
`