facebookresearch/habitat-sim #709: Is there a way to import the full Replica Dataset (with high-resolution textures)?

MeisterNewman, 2020-07-29:
## ❓ Questions and Help
When I generate images with the Replica dataset (using the subset of the dataset formatted specifically for use with AI Habitat), they seem to use lower fidelity/resolution textures and maps as compared to the sample images presented for the dataset (an image I captured is attached to demonstrate the difference). Is this due to an importing error on my end, and if not, how should I go about getting the full-resolution models into AI Habitat?

Thanks for the help!
[image]

erikwijmans, 2020-08-01:
The higher quality rendering comes from the PTex meshes.  Those can be loaded into habitat by pointing it to the `mesh.ply` file for a given scene.  You will also need to downloading the sorted faces from here: https://github.com/facebookresearch/habitat-sim/pull/221#issuecomment-533719083

@bigbike Did we ever talk to the Replica folks about including the sorted faces by default?

closed 2020-08-03 by MeisterNewman

emilia-szymanska, 2024-05-08:
Hi @erikwijmans! I have a question related to this issue - is there an option to use the default python viewer to view these high quality meshes similarly to `python examples/viewer.py --scene path/to/mesh.ply`? I have only needed the viewer.py so far (slightly modified) since I do not need much functionalities for my task (only manually walking around and saving the currently observed frame), I have the sorted faces downloaded, and I just would like to save the more realistic frames now. Is there a way?