facebookresearch/habitat-sim #711: How to extract single color per object in semantic segmentation

qmaruf, 2020-07-31:
## ❓ Questions and Help
I am following this example https://aihabitat.org/docs/habitat-api/habitat-sim-demo.html to extract semantic segmentation data from Replica dataset. Here is an output.

[image]

I am wondering why the `chair`/`table` instance has multiple color code? Is it changing because of the angular distance? How can I get a unique color code per object?

erikwijmans, 2020-07-31:
That doesn't look correct, can you please provide the mesh name and agent position?

qmaruf, 2020-08-02:
I was using `mesh_preseg_semantic.ply` instead of `mesh_semantic.ply`. The output seems Ok now.

[image]

What is the difference between these two files?
Here is the notebook for reference. https://colab.research.google.com/drive/14t0oOuDTKgH-awUohwoJZ79nus96BmRS?usp=sharing

erikwijmans, 2020-08-02:
Ah, that makes sense.  Preseg is a planar decomposition that creates the segments that are joined together to create the instances (there is also sometimes a little bit of manual fine-tuning).

closed 2020-08-03 by qmaruf