facebookresearch/habitat-sim #719: Panoramic image

JockTokk, 2020-08-05:
## ❓ Questions and Help
Hello!
How to extract a panoramic image from the simulator? Should I take some kind of sensor like a 360 camera or take 6 shots in 6 directions and stitch them together?

Thanks for your attention.

Skylion007, 2020-08-05:
Unfortunately, the current pinhole camera model has issues at 180/360 degrees FOV. We really do need to implement proper omnidirectional sensors. As a cludge, you can just make a 270 FOV for camera that will cover a wide area. 

I have also heard of people stitching together prospective views to achieve the same effect. 

PR: welcome on proper 180/360 camera support though!

JockTokk, 2020-08-06:
@ Skylion007, can you please tell me how can I do 270 FOV?

Skylion007, 2020-08-07:
You can specify the FOV as part of the sensor config.

JockTokk, 2020-08-10:
@Skylion007 thanks for you help!

closed 2020-09-14 by JockTokk

ronjamoller, 2020-10-05:
Hi, sorry for barging in after this issue is closed - I am aiming for a similar thing - I want my robot to have "surround view" and experimenting with the hfov has not been very successful for me - the distortion is too impractical for high fov values and I would rather combine 4 times 90 degrees. I would be very interested in the stitching idea, could you elaborate on that ? 
I have tried to register a second rgb camera externally as a sensor but I am running into issues with the config types - looking at the implementation of the `class HabitatSimRGBSensor(RGBSensor):` in `habitat_lab/habitat/sims/habitat_simulator/habitat_simulator.py`  it seems like the config type used for the external sensor ( `habitat.Config()` registration init is of a different type than the one that is used for registering the measure in `habitat_lab/habitat/sims/habitat_simulator/habitat_simulator.py` or i am using it incorrectly. 
I have tried to set the necessary  config values (height, fov etc)  internally but they are immutable so that does not work, giving the correct config (the one that containes hfov, width and height etc.)  instead of `habitat.Config()` during the `config.TASK.SENSORS.append` step also does not work. If possible I would like to avoid editing `/habitat-lab/habitat/config/default.py` or pretty much anything in the habitat core but it seems like I might have to, especially given the immutable nature of the config entries and i have to set the euler angles of the rotated camera somehow.
To make matters more complicated, I would later like to access the render camera `_sim.get_active_semantic_scene_graph().get_default_render_camera()` and I am unsure if that makes a difference in terms of how "deep" I have to get into the simulator itself.
I would be very thankful for any ideas !

Skylion007, 2020-10-05:
@ronjamoller Cool, let me know if you need anything else

ronjamoller, 2020-10-05:
Hi, thank you for offering - im actually running into the same issue from above - when i call `config.SIMULATOR.RGB_SENSOR.ORIENTATION = orient[0]` i get the `CfgNode is immutable` is immutable error. I just pulled the newest master and that didnt fix it. I did look at the issue where matsuren posted the demo script and i also copied the obs_transformers.py and the changes in /default.py in the hope that it would be ok to stay on the master. But im guessing i have to pull the  `batched_cube2eq`, correct ? I was hesitating because blindly pulling the newest stuff has ended badly in the past. Do you think pulling that branch is the only solution or is there a simple fix for the immutability issue (in case it is the only issue).
thanks for your help !

mathfac, 2020-10-05:
Just call: ‘config.defrost()’ before the config change.

ronjamoller, 2020-10-06:
Thank you - looking back at the 2 code examples it seems i just didnt copy it properly, sorry. The code runs properly for me now,  do you by any chance know how to access the projection and camera matrix for the different orientations ? I normally access it via `get_active_semantic_scene_graph().get_default_render_camera()` - I guess since the cameras only differ in orientation I can just transform the respective matrices but I wanted to check if there is a "standard" way to access different cameras.
thank you !

Skylion007, 2020-10-06:
CC: @bigbike @mosra

ronjamoller, 2020-10-13:
Update: In case anyone else wants to do something like this: as mentioned the projection matrix is unchanged and the camera matrix is very easy to transform for 90 degree rotations - only the x and y components of the rotation element and the translation element have to be swapped/inverted, its relatively straightforward to implement and if the hfov is set to 90 degrees as well there is no overlap and to get the (x,y) image coordinates for any 3d point, one just has to iterate over the camera orientations and check which one returns a valid entry, then add the respective x offset.
thanks :)