facebookresearch/habitat-sim #737: ground truth semantic map.

Singh-sid930, 2020-08-14:
## ❓ Questions and Help
Is there a way to extract the entire scene's semantic map. Or the only way is to stitch the episodic maps together. 
For example, this is a episodic semantic map. Buy I would like to have the entire scene semantic map of which this is a part. Is this possible. ? 
Or a new feature is needed for this?
[image]

mathfac, 2020-08-26:
To clarify, are you interested in 3D semantic map or TopDown semantic map?
Based on your example possibly you can use SceneGraph interface to extract [SemanticMesh](https://github.com/facebookresearch/habitat-sim/blob/e18766c92bd57d0c15d791d956299f60cb51b2d7/src/esp/assets/ResourceManager.h#L163) and build map similar to what you called episodic map.

Raminmoslemi, 2021-06-08:
@mathfac for a scene, how can I extract the ground truth top view map for the whole environment?

mukulkhanna, 2022-02-09:
@mathfac This seems to be a duplicate of #467; can therefore be closed?