facebookresearch/habitat-sim #742: ResourceManager.cpp:483] Check failed: Cr::Utility::Directory::exists(physicsFilename); Aborted (core dumped)

zkytony, 2020-08-18:
## 🐛 Bug

I am following the tutorial "rigid_object_tutorial.py". It was strange to me that when I `cd` into the `habitat-sim/examples/tutorials` directory, and run 
```
python rigid_object_tutorial.py
```
I get the error:
```
F0818 11:19:36.033696  6052 ResourceManager.cpp:483] Check failed: Cr::Utility::Directory::exists(physicsFilename) 
*** Check failure stack trace: ***
Aborted (core dumped)
```
However, when I am outside of the `examples/tutorials` directory and run it (as suggested on the Doc),
```
python path/to/habitat-sim/examples/tutorials/rigid_object_tutorial.py
```
I don't get that error.

Why would this happen? I wouldn't expect the program's output would be different because of the path it is being run from.

## Command
`# command that you run`

## To Reproduce

Steps to reproduce the behavior:

1. `cd` into the `habitat-sim/examples/tutorials` directory
2. `python rigid_object_tutorial.py` (Get the error)
3. python path/to/habitat-sim/examples/tutorials/rigid_object_tutorial.py (no error)

## Expected behavior

No error in both cases.

## System Info

```
ENVIRONMENT INFO:
Platform: Linux-4.15.0-112-generic-x86_64-with-debian-stretch-sid
Machine: x86_64
Processor: x86_64
Libc version: glibc 2.17
Mac version: 
Python version: 3.7.4
Architecture: 64bit ELF
Win version:    
System OS: Linux
Release: 4.15.0-112-generic
Version: #113~16.04.1-Ubuntu SMP Fri Jul 10 04:37:08 UTC 2020
Operational System: linux
GCC version: gcc (Ubuntu 9.3.0-10ubuntu2~16.04) 9.3.0
CMAKE version: cmake version 3.18.0
NVIDIA-SMI: Tue Aug 18 11:24:52 2020       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 450.36.06    Driver Version: 450.36.06    CUDA Version: 11.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce GTX 1070    Off  | 00000000:01:00.0  On |                  N/A |
| 28%   40C    P8     7W / 151W |    524MiB /  8117MiB |      8%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
Pip packages versions:

Conda packages versions:
```

zkytony, 2020-08-18:
I see. There is a configuration parameter, in `examples/settings`
```
"physics_config_file": "./data/default.phys_scene_config.json",
```
It's set to be relative to the root directory of `habitat-sim` repository.

Skylion007, 2020-08-18:
Yeah, we may want to update that via our gitpython hack for finding the root directory of Habitat-Sim that we use in the colab tutorials.

Skylion007, 2020-08-18:
BTW, we haven't advertised it yet because we haven't transferred the text from the RST, but that particular tutorial is available in colab: https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/master/examples/tutorials/colabs/rigid_object_tutorial.ipynb

zkytony, 2020-08-18:
Cool, thanks for the colab link. It's fine now when I set my own `physics_config_file` parameter!

closed 2020-08-18 by zkytony