facebookresearch/habitat-sim #743: CORS error in browser demo

pringithub, 2020-08-18:
## 🐛 Bug

Trying to open the browser demo. On latest Chrome version `Version 84.0.4147.125 (Official Build) (64-bit)` on Ubuntu 18.04.

Getting a CORS request error

[image]

Skylion007, 2020-08-18:
Odd... can't reproduce on OSX Chrome:
Version 84.0.4147.125 (Official Build) (64-bit)

Skylion007, 2020-08-18:
@pringithub Make sure the demo is open in HTTPS. I think our CORS scripts support http://

pringithub, 2020-08-18:
Didn't work with either.

Skylion007, 2020-08-18:
https://github.com/facebookresearch/habitat-lab/issues/416

Skylion007, 2020-08-18:
@pringithub This is a known issue that happens when the webpage is opened as HTTP instead of HTTPS as the error log indicates.

Skylion007, 2020-08-28:
We changed some settings on our website to enforce strict HTTPS. Reopen if the issue persists.

closed 2020-08-28 by Skylion007