facebookresearch/habitat-sim #762: Installation failure on RHEL based OS

Lucas2012, 2020-08-26:
## ❓ Questions and Help

Hi,

I've been trying to install habitat-sim but failed on multiple attempts with different possible combinations. The compilation process is successful. But when importing habitat_sim in python, it directly gives a "segmentation fault error" and exits. I'm using RHEL based OS and noticed that the instructions are for Ubuntu.

*Here's what I did and the error message I got:

   gcc 8.3.1

   git clone --branch stable https://github.com/facebookresearch/habitat-sim.git
   cd habitat-sim

   conda create -n habitat python=3.6 cmake=3.14.0
   source activate habitat
   pip install -r requirements.txt

   python setup.py install --headless --with-cuda

   cd ../
   python
   import habitat_sim

   >> gives error "[1]    25711 segmentation fault  python"

* My python 3.6 pip freeze output:
$ pip freeze
absl-py==0.10.0
asn1crypto==0.24.0
astor==0.8.1
attrs==20.1.0
certifi==2018.4.16
cffi==1.11.5
chardet==3.0.4
conda==4.5.4
cryptography==2.2.2
cycler==0.10.0
gast==0.4.0
gitdb==4.0.5
GitPython==3.1.7
google-pasta==0.2.0
grpcio==1.31.0
h5py==2.10.0
habitat-sim==0.1.5
idna==2.6
importlib-metadata==1.7.0
Keras-Applications==1.0.8
Keras-Preprocessing==1.1.2
kiwisolver==1.2.0
leveldb==0.201
llvmlite==0.32.1
magnum==0.0.0
Markdown==3.2.2
matplotlib==3.3.1
numba==0.49.1
numpy==1.19.1
numpy-quaternion==2020.7.30.15.57.32
Pillow==7.2.0
protobuf==3.13.0
pycosat==0.6.3
pycparser==2.18
pyOpenSSL==18.0.0
pyparsing==2.4.7
PySocks==1.6.8
python-dateutil==2.8.1
requests==2.18.4
ruamel-yaml==0.15.37
scipy==1.5.2
six==1.11.0
smmap==3.0.4
tensorboard==1.14.0
tensorflow==1.14.0
tensorflow-estimator==1.14.0
termcolor==1.1.0
tqdm==4.48.2
urllib3==1.22
Werkzeug==1.0.1
wrapt==1.12.1
zipp==3.1.0

Do you have any guess on the issue and any possible solutions?
Thanks.

Skylion007, 2020-08-26:
Run it with `python -X faulthandler ...` to get a stack trace of where the segfault is occuring and post the results here.

Skylion007, 2020-08-26:
Also we do have a conda-nightly build you could use instead. See the `examples/colab_utils/install.sh` to see how to install it.

Lucas2012, 2020-08-27:
> Run it with `python -X faulthandler ...` when you import habitat_sim to get a stack trace of where the segfault is occuring and post the results here.

Thank you. This is the output:

Fatal Python error: Segmentation fault                                                                                          

Current thread 0x00007f3af1ded740 (most recent call first):
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 922 in create_module
  File "<frozen importlib._bootstrap>", line 571 in module_from_spec
  File "<frozen importlib._bootstrap>", line 658 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 955 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 971 in _find_and_load
  File "/path-to-softwares/softwares/anaconda3_habitat/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.1.5-py3.6-linux-x86_64.egg/habitat_sim/bindings/__init__.py", line 29 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 678 in exec_module
  File "<frozen importlib._bootstrap>", line 665 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 955 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 971 in _find_and_load
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1023 in _handle_fromlist
  File "/path-to-softwares/softwares/anaconda3_habitat/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.1.5-py3.6-linux-x86_64.egg/habitat_sim/errors.py", line 4 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 678 in exec_module
  File "<frozen importlib._bootstrap>", line 665 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 955 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 971 in _find_and_load
  File "/path-to-softwares/softwares/anaconda3_habitat/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.1.5-py3.6-linux-x86_64.egg/habitat_sim/agent/agent.py", line 13 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 678 in exec_module
  File "<frozen importlib._bootstrap>", line 665 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 955 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 971 in _find_and_load
  File "/path-to-softwares/softwares/anaconda3_habitat/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.1.5-py3.6-linux-x86_64.egg/habitat_sim/agent/__init__.py", line 7 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 678 in exec_module
  File "<frozen importlib._bootstrap>", line 665 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 955 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 971 in _find_and_load
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1023 in _handle_fromlist
  File "/path-to-softwares/softwares/anaconda3_habitat/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.1.5-py3.6-linux-x86_64.egg/habitat_sim/__init__.py", line 14 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 678 in exec_module
  File "<frozen importlib._bootstrap>", line 665 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 955 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 971 in _find_and_load
  File "<stdin>", line 1 in <module>
[1]    45861 segmentation fault  python -X faulthandler

Skylion007, 2020-08-27:
Hmm, and this same issue happens when using the CONDA_NIGHTLY build? 
Also try building from master and see if the issue still occurs.

Lucas2012, 2020-08-30:
> Hmm, and this same issue happens when using the CONDA_NIGHTLY build?
> Also try building from master and see if the issue still occurs.
> If that doesn't work, try doing it without the `--with-cuda` flag and see if it still occurs.
> 
> We have had this issue occur before when the libraries is built against different libraries than the ones used at runtime.

Thank you. Removing --with-cuda didn't help. But the nightly version works fine -- don't know what was the reason of previous issue though.

I used the following command:

conda install -y --prefix /path/to/env/habitat -c aihabitat-nightly -c conda-forge habitat-sim headless withbullet python=3.6

BTW is there any way to install the with-cuda version? If run without cuda, how big will be the drop?

-- Thanks.

Skylion007, 2020-09-19:
Shouldn't be that big of a drop in performance unless you are using the GPU noise models. However, even that will cause a performance regression in terms of VRAM due to each simulator