facebookresearch/habitat-sim #780: Adding realistic depth noise

ankurhanda, 2020-09-05:
## 🚀 Feature

We have been using out depth noise models to benchmark SLAM system as well as doing sim-to-real perception with neural networks. They are more realistic and you may find them useful too 

Here is the code https://github.com/ankurhanda/simkinect and it has examples on how to use them. 

## Motivation

Better noise models will aid sim-to-real transfer. 

## Pitch

The python code I wrote above is very slow and may need pybind11 with CUDA kernels. I may not get to it soon so I thought I may point it out to you the codebase and see if you can incorporate that in habitat-sim.

## Alternatives

## Additional context

This is what it looks like. Left is the clean image and right is the image with depth noise.

[image]

Skylion007, 2020-09-07:
If you can get it to use NUMBA, that might make might sufficiently fast, only way to know for sure is to try it though.

ankurhanda, 2020-09-07:
Yes, so as said I am not finding time but the depth noise model is better than the one you currently have in habitat. I have been using it for the past 6 years in most of my papers. But I'd let you make a decision if you want to add that in habitat. I just wanted to point it out. 

I may go back to doing it but won't be any time soon (not in the next 6 months). 

Thank you.