facebookresearch/habitat-sim #787: IMU sensor support ?

youzhenxing, 2020-09-11:
## 🚀 Feature

## Motivation

## Pitch

## Alternatives

## Additional context

youzhenxing, 2020-09-11:
Is there any plans for imu sensor type support? I meas IMU is also an very common sensor type for robot integration.

Skylion007, 2020-09-12:
How would you envision this IMU sensor? How would deviate from knowing the ground truth translation/rotation from the start point? We do have similar sensors GPS+Compass in our Habitat-Lab repo.

youzhenxing, 2020-09-15:
> How would you envision this IMU sensor? How would deviate from knowing the ground truth translation/rotation from the start point? We do have similar sensors GPS+Compass in our Habitat-Lab repo.

One of the scenarios is to use habitat-sim to do visual internal fusion based localization. There are still differences and modeling errors  between using differential of GPS+Compass data  to simulate IMU data and the direct IMU model characteristics

dhruvbatra, 2020-09-15:
> There are still differences and modeling errors between using differential of GPS+Compass data to simulate IMU data and the direct IMU model characteristics

Agreed, but are there realistic IMU noise models available online? Because if not, then differential of GPS+Compass is the noise-free IMU measurement.

closed 2020-10-08 by Skylion007

ltstein, 2021-09-28:
Could a similar approach to other robotics simulators be used where there is a bias and noise is added to the perfect data, such as in [Gazebo](http://gazebosim.org/tutorials?tut=sensor_noise#IMUnoise), especially with the recent addition of physics? However, it might be out of scope for habitat sim and  may be a better use case for the ROS-X-Habitat package so you can use another simulator such as Ignition Gazebo for certain portions and then train the higher level agent using the Habitat sim.